RIOT-OS/RIOT #8664: pkg: doxygen documentation is not fully processed

aabadie, 2018-02-28:
#### Description

While looking for Semtech-loramac API documentation on https://doc.riot-os.org, I noticed that the public API is not referenced (see http://doc.riot-os.org/group__pkg__semtech-loramac.html)

And this problem is more general because it's also the case with openthread, lwip, nordic_softdevice_ble, etc

After looking at the [doxygen configuration file](https://github.com/RIOT-OS/RIOT/blob/master/doc/doxygen/riot.doxyfile), I noticed that all pkg directories are skipped by doxygen (see [here](https://github.com/RIOT-OS/RIOT/blob/master/doc/doxygen/riot.doxyfile#L830))

#### Steps to reproduce the issue

Run the following command:

```
make doc
```

Here is the list of packages with the problem (because they expose a public API, compatible with RIOT, on top the package API):

- [ ] emb6
- [ ] lwip
- [ ] nordic_softdevice_ble
- [ ] openthread
- [ ] semtech-loramac

I'll come up with PRs fixing some of them.

#### Expected results

API documentation of packages are parsed and appear in the generated html documentation

#### Actual results

API documentation of packages are not parsed and doesn't appear in the generated html documentation

#### Versions

```
Installed toolchain versions
----------------------------
          native gcc: gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
          msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
             avr-gcc: missing/error
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               clang: clang version 4.0.1-6 (tags/RELEASE_401/final)
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: "2.1.0"
            avr-libc: missing/error (missing/error)
            cppcheck: Cppcheck 1.80
          coccinelle: spatch version 1.0.4 with Python support and with PCRE support
                 git: git version 2.14.1
```

aabadie, 2018-02-28:
I won't work on lwip because there are too many changes required (mainly missing things) and I don't feel I have enough expertise to work on this one. For emb6, the changes should be quite simple.

miri64, 2018-02-28:
As a user of lwIP with RIOT, maybe @vincent-d can also provide some help here?

gebart, 2018-02-28:
I am not convinced this is a good idea. The RIOT API docs is already quite large, there is a risk that mixing in all pkg documentation in the same place will make reading the documentation confusing with different library APIs mixed in the same documentation index.

vincent-d, 2018-02-28:
> As a user of lwIP with RIOT, maybe @vincent-d can also provide some help here?

I don't use lwIP, sorry

aabadie, 2018-02-28:
> there is a risk that mixing in all pkg documentation in the same place will make reading the documentation confusing with different library APIs mixed in the same documentation index

In fact, the documentation should only be related to the adaption layer added to make it fit with RIOT internal API (mainly network APIs). It won't process the package APIs themselves.

aabadie, 2018-02-28:
And there's nothing that checks the doxygen documentation is complete or valid (like Murdock does), so if someone wants to known what a function is about, he may find non existent or invalid information in the code (see changes in related PRs).

gebart, 2018-02-28:
Ah ok, thanks for clarifying that. I thought it was going to add the entire documentation for all pkgs. If it is only the glue layer then I am all for this.

miri64, 2018-03-01:
>> As a user of lwIP with RIOT, maybe @vincent-d can also provide some help here?
>
> I don't use lwIP, sorry

Or was it @lebrush? :-/ Sorry, my memory is out of wack the last few days ....

aabadie, 2019-08-11:
This is still a valid issue that we should fix.

miri64, 2019-10-09:
Fixed the doc of `emb6` in #12389 only to deprecate it.

miri64, 2019-10-09:
And with #12403 `lwip` should be fixed.

PR #12403 (2019-10-09): lwip: improve and fix documentation

### Contribution description
This fixes and improves the documentation of the lwIP porting layer to make it build again. While I was reading lwIP's documentation on porting, I noticed that some defines were not required anymore, which I removed, rather than documented in a piggy-backed commit.

### Testing procedure
```
make doc
```

passes and shows lwIP now under Packages (and Networking).

lwIP applications should still build with the removed macros.

### Issues/PRs references
Fixes #8664
closed 2019-10-09 by aabadie