RIOT-OS/RIOT #8730: unittests/spiffs: Unit tests fail on device without real MTD

gebart, 2018-03-02:
#### Description

Running tests-spiffs on samr21-xpro and frdm-kw41z results in test failures on vfs_mount (-ENODEV).

My first guess is that this is because of the vfs_format addition in #8315 

The boards which do not have an MTD device on board are using a minimal RAM backed MTD emulation defined inside the test application to be able to test the functionality.

#### Steps to reproduce the issue

```
cd tests/unittests
make BOARD=samr21-xpro flash term
```

#### Expected results

tests pass

#### Actual results

```
2018-03-02 14:20:10,911 - INFO # .
2018-03-02 14:20:10,920 - INFO # spiffs_tests.tests_spiffs_format (tests/unittests/tests-spiffs/tests-spiffs.c 164) exp 0 was -19
2018-03-02 14:20:10,921 - INFO # .
2018-03-02 14:20:10,930 - INFO # spiffs_tests.tests_spiffs_mount_umount (tests/unittests/tests-spiffs/tests-spiffs.c 178) exp 0 was -22
2018-03-02 14:20:10,931 - INFO # .
2018-03-02 14:20:10,940 - INFO # spiffs_tests.tests_spiffs_open_close (tests/unittests/tests-spiffs/tests-spiffs.c 188) res >= 0
2018-03-02 14:20:10,941 - INFO # .
2018-03-02 14:20:10,949 - INFO # spiffs_tests.tests_spiffs_write (tests/unittests/tests-spiffs/tests-spiffs.c 201) fd >= 0
2018-03-02 14:20:10,951 - INFO # .
2018-03-02 14:20:10,959 - INFO # spiffs_tests.tests_spiffs_unlink (tests/unittests/tests-spiffs/tests-spiffs.c 233) fd >= 0
2018-03-02 14:20:10,960 - INFO # .
2018-03-02 14:20:10,968 - INFO # spiffs_tests.tests_spiffs_readdir (tests/unittests/tests-spiffs/tests-spiffs.c 253) fd0 >= 0
2018-03-02 14:20:10,970 - INFO # .
2018-03-02 14:20:10,978 - INFO # spiffs_tests.tests_spiffs_rename (tests/unittests/tests-spiffs/tests-spiffs.c 317) fd >= 0
2018-03-02 14:20:10,979 - INFO # .
2018-03-02 14:20:10,988 - INFO # spiffs_tests.tests_spiffs_statvfs (tests/unittests/tests-spiffs/tests-spiffs.c 362) exp 0 was -2
2018-03-02 14:20:10,988 - INFO # 
2018-03-02 14:20:10,989 - INFO # run 8 failures 8
```

#### Versions

```
Installed toolchain versions
----------------------------
          native gcc: gcc (Gentoo 7.3.0 p1.0) 7.3.0
          msp430-gcc: missing/error
             avr-gcc: avr-gcc (Gentoo 7.3.0 p1.0) 7.3.0
   arm-none-eabi-gcc: arm-none-eabi-gcc (Gentoo 7.3.0 p1.0) 7.3.0
    mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               clang: clang version 5.0.1 (tags/RELEASE_501/final)
arm-none-eabi-newlib: "3.0.0"
 mips-mti-elf-newlib: "2.1.0"
            avr-libc: "2.0.0" ("20150208")
            cppcheck: Cppcheck 1.81
          coccinelle: spatch version 1.0.6 compiled with OCaml version 4.05.0
                 git: git version 2.16.2
```

gebart, 2018-03-05:
I assume the bug fix for this will be similar to #8607

vincent-d, 2018-03-05:
Sorry, I overlooked this.

> I assume the bug fix for this will be similar to #8607

Yup, I think so. I can provide a patch if you want.

gebart, 2018-03-05:
If you have the time. I can review, but I likely can't test until tomorrow

vincent-d, 2018-03-05:
see #8741

PR #8741 (2018-04-10): unittests/tests-spiffs: fix dummy mtd (as in #8607)
### Contribution description

This is the same fix as #8607 for spiffs.

### Issues/PRs references

Fixes #8730, see also #8607
closed 2018-04-10 by kaspar030