RIOT-OS/RIOT #8737: dist: header guard checker too nit-picky

miri64, 2018-03-05:
In #7881 I found myself stumped when both CI servers still complained about the header guards, though at least to me they looked alright. Turns out the header guard script expects there to be an *exact* number of spaces (1) between the `#endif` and the comment. I think this is very nit-picky and especially annoying since the script doesn't show (at least on the CI servers) on its own what the actual problem is.

My proposed fix would be to either just output the problem so a contributor is able to react quickly or even better make the number of whitespaces variable when applicable.

bergzand, 2018-03-05:
Would adjusting the `headercheck.py` script to output something like:

```
expected: "#endif /* HASHES_SHA3_H */" 
found     "#endif  /* HASHES_SHA3_H */"
```

solve this issue?

mtausig, 2018-03-05:
Yes. This I way I could fix the issues myself instead of waiting for someone else to do some extra work to tell me what I need to do.

smlng, 2018-03-05:
mhm, the script in `dist/tools/headerguards/headerguards.py` already does that, i.e., it prints a diff when executed locally, but it seems that `static-tests` suppresses the output.

PR #8738 (2018-03-05): tools: show diff for headerguards check on failure

### Contribution description

show output for headerguards check on failure

### Issues/PRs references

fixes #8737
commit 32cacd17 (2018-03-05): tools: show diff for headerguards check on failure

    fixes #8737
closed 2018-03-05 by miri64