RIOT-OS/RIOT #8750: pkg/semtech-loramac: semtech_loramac_send function stucked after device wakeup

dylad, 2018-03-07:
#### Description
I'm currently using the semtech-loramac pkg on a custom board (SAML21J17B / SX1276)
I'm able to send/receive data to/from TTN.
I put my device in standby sleep mode with `pm_set(1)`, device is wake up with the RTC at a given time, I'm trying to send data again but I'm stucked within the semtech_loramac_send function (more precisily at the `msg_receive(&msg);` line of this function.
On the gateway side I can see that I received the second transmission after device wake up.
Note: this issue doesn't occur when I don't use pm_set(1).
@aabadie did you have the same behavior using a nucleo board ?

#### Steps to reproduce the issue
Put the device to sleep between two transmissions.

#### Expected results
Device send data after the RTC wake up and then go back to sleep.

#### Actual results
App get stucked.

#### Versions
Current master

aabadie, 2018-03-07:
Are you sleeping between the TX and RX windows ?

dylad, 2018-03-07:
No I'm waiting for the timeout
My app looks like :  
```
//Init stuff
while(1) {
    //do stuff
    switch(semtech_loramac_send(xxxx)) {
        case 
        ...
    }
    // do stuff
    pm_set(1)
}
```

dylad, 2018-03-07:
How can I achieve this ? I don't use shell for this app.

dylad, 2018-03-07:
I just discovered `thread_getpid` funtion :)
It returns '2' before sleep and after wakeup.

aabadie, 2018-03-07:
Can you check if one of these lines [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac.c#L161) or [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac.c#L238) in the code is called ?

aabadie, 2018-03-07:
And if yes what is the value of `semtech_loramac_handler_pid` ?

dylad, 2018-03-07:
before sleep the code is called and the value of semtech_loramac_handler_pid is 2 (same as the thread_getpid return from main)
after sleep the code is not called.

aabadie, 2018-03-07:
> after sleep the code is not called

So that explains why the send function is stuck. Now one has to find why it's not called... Maybe enable debug output [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac.c#L44), and see what are the debug statements called.

dylad, 2018-03-07:
Here's the log : 
```
[semtech-loramac] loramac cmd                                                                       
[semtech-loramac] send frame a                                                                      
[semtech-loramac] MCPS_CONFIRMED                                                                    
[semtech-loramac] MCPS request OK                                                                   
[semtech-loramac] MAC timer timeout                                                                 
[semtech-loramac] Transmission completed                                                            
[semtech-loramac] MAC timer timeout                                                                 
[semtech-loramac] MAC timer timeout                                                                 
[semtech-loramac] MAC timer timeout                                                                 
[semtech-loramac] RX timer timeout                                                                  
[semtech-loramac] MAC timer timeout                                                                 
[semtech-loramac] MCPS confirm event                                                                
[semtech-loramac] MCPS confirm event OK                                                             
[semtech-loramac] MCPS confirm event CONFIRMED                                                      
[semtech-loramac] MCPS indication event                                                             
[semtech-loramac] MCPS indication Unconfirmed                                                                                                                                    
TX done 1                                                                                           
go to sleep
wakeup                                                                                                                                                                  
[semtech-loramac] loramac cmd                                                                       
[semtech-loramac] send frame a                                                                      
[semtech-loramac] MCPS_CONFIRMED
```

dylad, 2018-03-07:
Within the LoRaMac.c file, I'm stuck [here](https://github.com/Lora-net/LoRaMac-node/blob/f42be67be402a40b3586724800771bfe13fb18e6/src/mac/LoRaMac.c#L3303)
Looks like the frame is sent but there is no DIO interrupts after... I'll investigate.

aabadie, 2018-03-07:
> Looks like the frame is sent but there is no DIO interrupts after

So maybe an issue with interrupt configuration lost after wake-up ?

dylad, 2018-03-07:
I think I've found the culprit ! Looks like the timer is involved. If I allow my timer driver to run in standby mode, i'm getting the desired behavior otherwise it seems we're stuck somewhere (because of a missing timeout ?)
I'm wondering if we can reproduce this issue with others CPU.

aabadie, 2018-03-07:
Internally the mac timeout are managed by the xtimer that itself uses the timer peripheral driver. So it could be that the timer is not reinitialized/powered on correctly after wake up ?

> I'm wondering if we can reproduce this issue with others CPU.

This sounds hardware dependent (at driver level) so it's difficult to say.
I can test with stm32 based boards.

dylad, 2018-03-07:
> I can test with stm32 based boards.

Sounds good to me :)

aabadie, 2018-03-09:
> I can test with stm32 based boards.

I gave a try to pm + rtc + loramac on a STM32L0 based board (thanks to #7329) and it works with some adaption in the semtech-loramac package that may not be related to your problem (My application is sending messages from another thread than the main thread).

Regarding your issue, maybe you need to add calls to `pm_block`/`pm_unblock` functions. That's what I did.

I only tested the stop mode because using standby mode would require to restart the join procedure at each wake up (not 100% sure btw).

Let us know of you made some progress on your side.

dylad, 2018-03-09:
Thanks for testing @aabadie  !
Regarding my issue I'm getting more and more convicted this is related to a sillicon bug within SAML21.
In standby mode, GCLK main clock (processor clock) is not shutdown properly. GCLK main clock is taken from GCLK0 which is used for the saml21 timer driver (and almost all other peripherals)
When I go to standby mode all my peripherals must stop so GCLK0 must be stop at the same time. But the sillicon bug doesn't really shutdown GCLK0. I had to used the workaround provided by Atmel (switch GCLK0 input to the internal 16 MHz from OSCULP32K just before going to sleep (OSCULP32K consumption is super low compare to the 16 MHz internal oscillator.

Right now, I must force the timer driver to run in standby mode with (GCLK0 supplied by OSCULP32K or OSC16M) otherwise, timeouts never occur after wakeup. (Even if I called xtimer_init or timer_init when my device wakes up).
I somehow solved my issue but I'm not happy with the changes yet. (timer driver should go to sleep like all others peripherals and properly work after wakeup but that's not the case).

dylad, 2018-03-09:
> I only tested the stop mode because using standby mode would require to restart the join procedure at each wake up (not 100% sure btw).

Why do you need to restart the join procedure ? RAM is not in retention state in standby mode for STM32 ?

aabadie, 2018-03-09:
> Why do you need to restart the join procedure ? RAM is not in retention state in standby mode for STM32 ?

Apparently not, the application is doing a full reset apparently (I'm not a PM expert at all)

dylad, 2018-03-09:
@aabadie Should we close this since you cannot reproduce this issue on other arch ? I'll provide a fix if something is wrong with SAML21.

closed 2018-03-09 by dylad