RIOT-OS/RIOT #8752: mrf24j40: does not link for examples/default

miri64, 2018-03-07:
#### Description
When using this driver with a board that does not provide an on-board network device (e.g. `stm32f4discovery` or `nucleo-f302`) with `examples/default`, the application will fail to link due to the GNRC module missing. I think this is some dependency issue.

#### Steps to reproduce the issue
Build the `default` example with a board in question and `mrf24j40`:

```bash
BOARD=stm32f4discovery USEMODULE=mrf24j40 make -C examples/default/
```

#### Expected results
Linking step should succeed.

#### Actual results
Linking fails:

```
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default'
Building application "default" for "stm32f4discovery" with MCU "stm32f4".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/stm32f4discovery
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32f4
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32f4/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/mrf24j40
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/netdev_ieee802154
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/saul
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init/saul
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/div
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/fmt
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/isrpipe
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/luid
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/net/link_layer/ieee802154
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/newlib_syscalls_default
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/phydat
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/pm_layered
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/ps
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/saul_reg
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/shell
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/shell/commands
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/tsrb
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/uart_stdio
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/xtimer
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/bin/stm32f4discovery/application_default.a(main.o): In function `main':
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/main.c:42: undefined reference to `gnrc_netreg_register'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/main.c:48: undefined reference to `gnrc_pktdump_pid'
collect2: error: ld returned 1 exit status
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/../../Makefile.include:328: recipe for target 'link' failed
make: *** [link] Error 1
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default'
```

#### Versions
Operating system: Ubuntu 17.10
Build environment:
```
Installed toolchain versions
----------------------------
          native gcc: gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
          msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
             avr-gcc: missing/error
   arm-none-eabi-gcc: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
    mips-mti-elf-gcc: missing/error
dist/tools/ci/print_toolchain_versions.sh: 92: dist/tools/ci/print_toolchain_versions.sh: clang: not found
arm-none-eabi-newlib: "2.4.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: missing/error (missing/error)
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.14.1
```

bergzand, 2018-03-07:
@miri64 I can also reproduce this issue with:
`BOARD=stm32f4discovery USEMODULE=at86rf233 make -C  examples/default/`
and with
`BOARD=stm32f4discovery USEMODULE=cc2440 make -C  examples/default/`
so this is not limited to the mrf24j40, but a valid issue nonetheless

aabadie, 2018-04-15:
Briefly looked at the 'default' Makefile and gnrc related modules are only loaded if the board is listed in the `BOARD_PROVIDES_NETIF` variable.
So I would say that this is not a bug but a feature.
Could you try the same build command but with your board added to this list ? Maybe using `BOARD_PROVIDES_NETIF=stm32f4discovery BOARD=stm32f4discovery USEMODULE= cc2420 make` would work ?

miri64, 2018-04-15:
> So I would say that this is not a bug but a feature.

I think there is still a dependency issue if you get a linking error due to this. Even if the board is not in `BOARD_PROVIDES_NETIF`, the worst thing that should happen is that it isn't controllable, since GNRC is missing. So this *is* a bug.

But yes, besides that, we might either make the list configurable or, maybe much more convenient have it like PROVIDES_FEATURE with the `periph` interfaces, that every network device driver just provides the feature.

aabadie, 2018-04-15:
AFAIU, the `default` application mainly relies on the hardware provided by the boards. So if a device is not provided by default by a board (and also configured for this particular board), it's not guaranteed to work. Here, you are trying to build an external radio plugged on a board that doesn't provide/configure it.
Your `PROVIDES_FEATURE` proposition sound interesting though. But this won't solve the fact that a driver loaded at build time from command line will work. Is it worth doing this ?

kYc0o, 2018-07-20:
Jut to confirm that this still is an issue and quite annoying btw... @miri64 any ideas so far? What it has been discussed here is still valid?

miri64, 2018-07-20:
New ideas (apart from the one proposed in the discussion), no. I still think however that this is a configuration bug / flaw of the application and/or our dependency system.

aabadie, 2019-10-16:
I think this PR can be fixed by #11676.

fjmolinas, 2021-05-06:
This no longer fails for me:

```
BUILD_IN_DOCKER=1 BOARD=stm32f4discovery USEMODULE=mrf24j40 make -C examples/default/
Launching build container using image "riot/riotbuild:latest".
docker run --rm --tty --user $(id -u) -v '/usr/share/zoneinfo/Europe/Paris:/etc/localtime:ro' -v '/home/francisco/workspace/RIOT:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'      -e 'BOARD=stm32f4discovery'  -w '/data/riotbuild/riotbase/examples/default/' 'riot/riotbuild:latest' make      
Building application "default" for "stm32f4discovery" with MCU "stm32".

"make" -C /data/riotbuild/riotbase/boards/stm32f4discovery
"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/cpu/stm32
"make" -C /data/riotbuild/riotbase/cpu/cortexm_common
"make" -C /data/riotbuild/riotbase/cpu/cortexm_common/periph
"make" -C /data/riotbuild/riotbase/cpu/stm32/periph
"make" -C /data/riotbuild/riotbase/cpu/stm32/stmclk
"make" -C /data/riotbuild/riotbase/cpu/stm32/vectors
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/drivers/saul
"make" -C /data/riotbuild/riotbase/drivers/saul/init_devs
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/fmt
"make" -C /data/riotbuild/riotbase/sys/isrpipe
"make" -C /data/riotbuild/riotbase/sys/malloc_thread_safe
"make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default
"make" -C /data/riotbuild/riotbase/sys/phydat
"make" -C /data/riotbuild/riotbase/sys/pm_layered
"make" -C /data/riotbuild/riotbase/sys/ps
"make" -C /data/riotbuild/riotbase/sys/saul_reg
"make" -C /data/riotbuild/riotbase/sys/shell
"make" -C /data/riotbuild/riotbase/sys/shell/commands
"make" -C /data/riotbuild/riotbase/sys/stdio_uart
"make" -C /data/riotbuild/riotbase/sys/tsrb
   text	   data	    bss	    dec	    hex	filename
  20880	    132	   2484	  23496	   5bc8	/data/riotbuild/riotbase/examples/default/bin/stm32f4discovery/default.elf
```

@miri64 can you confirm and this can be closed?

miri64, 2021-05-06:
> @miri64 can you confirm and this can be closed?

Confirmed

closed 2021-05-06 by miri64