RIOT-OS/RIOT #8754: CC101x Initialization kernel panic

Citrullin, 2018-03-08:
Hi, I currently checked the following pull request: https://github.com/RIOT-OS/RIOT/pull/8437

I wrote the following code
```
#include <stdio.h>

#include "radio.h"
#include "tlsf-malloc.h"
#include "msg.h"
#include "shell.h"
#include "ccn-lite-riot.h"
#include "net/gnrc/netif.h"
#include "net/gnrc/pktdump.h"

#include "cc110x.h"
#include "cc110x_params.h"
#include "gnrc_netif_cc110x.h"
#include "cc110x-netdev.h"

/* main thread's message queue */
#define MAIN_QUEUE_SIZE     (2)
static msg_t _main_msg_queue[MAIN_QUEUE_SIZE];

/* 10kB buffer for the heap should be enough for everyone */
#define TLSF_BUFFER     (5600 / sizeof(uint32_t))
static uint32_t _tlsf_heap[TLSF_BUFFER];

int main(void)
{
    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    tlsf_create_with_pool(_tlsf_heap, sizeof(_tlsf_heap));
    msg_init_queue(_main_msg_queue, MAIN_QUEUE_SIZE);

    puts("Basic CCN-Lite example");

    ccnl_core_init();

    ccnl_start();

    char stack[50];
    netdev_cc110x_t* netdev_cc110x = (netdev_cc110x_t*) malloc(sizeof(netdev_cc110x_t));
    netdev_cc110x_setup(netdev_cc110x, cc110x_params);
    gnrc_netif_t *netif = gnrc_netif_cc110x_create(stack, 50, 'X', "main", &(netdev_cc110x->netdev));

    if (((netif = gnrc_netif_iter(NULL)) == NULL) ||
        (ccnl_open_netif(netif->pid, GNRC_NETTYPE_CCN) < 0)) {
        puts("Error registering at network interface!");
        return -1;
    }

#ifdef MODULE_NETIF
    gnrc_netreg_entry_t dump = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                                          gnrc_pktdump_pid);
    gnrc_netreg_register(GNRC_NETTYPE_CCN_CHUNK, &dump);
#endif

    return 0;
}
```
I don't think that especially this part is correct:
```
gnrc_netif_cc110x_create(stack, 50, 'X', "main", &(netdev_cc110x->netdev));
```
I can compile it and the MCU even starts. But it results in a kernel panic:
```
cc110x: initialized with address=138 and channel=0
0x8007c27
*** RIOT kernel panic:
FAILED ASSERTION.

	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
	  - | isr_stack            | -        - |   - |    512 (  132) | 0x20000000 | 0x200001c8
	  1 | idle                 | pending  Q |  15 |    256 (  132) | 0x20000450 | 0x200004cc 
	  2 | main                 | running  Q |   7 |   1536 (  896) | 0x20000550 | 0x200008dc 
	  3 | pktdump              | bl rx    _ |   6 |   1536 (  236) | 0x200043e0 | 0x200048f4 
	  4 | ccnl                 | bl rx    _ |   6 |   1536 (  176) | 0x20002504 | 0x20002a54 
	    | SUM                  |            |     |   5376 ( 1572)

Heap usage:
	Memory @ 0x20001880 is used, size: 68 (block: 0x20001878)
	Memory @ 0x200018c8 is used, size: 192 (block: 0x200018c0)
	Memory @ 0x2000198c is free, size: 2136 (block: 0x20001984)

*** halted.
```
I don't understand this kernel panic message. What assertion failed?

aabadie, 2018-03-08:
> What assertion failed?

Prefix your build command with `CFLAGS=-DDEBUG_ASSERT_VERBOSE`, you'll see where the assertion failed.

Citrullin, 2018-03-08:
@aabadie Thanks.
```
sys/net/gnrc/netif/gnrc_netif.c:68 => 0x8000ea7
```

I need later take a look into it. Maybe it already find it by myself.

miri64, 2018-03-08:
@Citrullin unless you explicitly deactivated the `auto_init` module all initialization you are doing is already done for you by that module (so I suspect `thread_create()` to return an error because the thread is initialized twice wit the same stack). Also the stack you are allocating for the network interface is scarily small (`sizeof(thread_t)` [48 on 32-bit platforms] is the minimum size for a stack and that just leaves place for the TCB). I would propose you to not disable `auto_init` and keep all the network device initialization stuff out of your `main()`:

```C
#include <stdio.h>

#include "radio.h"
#include "tlsf-malloc.h"
#include "msg.h"
#include "shell.h"
#include "ccn-lite-riot.h"
#include "net/gnrc/netif.h"
#include "net/gnrc/pktdump.h"

/* main thread's message queue */
#define MAIN_QUEUE_SIZE     (2)
static msg_t _main_msg_queue[MAIN_QUEUE_SIZE];

/* 10kB buffer for the heap should be enough for everyone */
#define TLSF_BUFFER     (5600 / sizeof(uint32_t))
static uint32_t _tlsf_heap[TLSF_BUFFER];

int main(void)
{
    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    tlsf_create_with_pool(_tlsf_heap, sizeof(_tlsf_heap));
    msg_init_queue(_main_msg_queue, MAIN_QUEUE_SIZE);

    puts("Basic CCN-Lite example");

    ccnl_core_init();

    ccnl_start();

    if (((netif = gnrc_netif_iter(NULL)) == NULL) ||
        (ccnl_open_netif(netif->pid, GNRC_NETTYPE_CCN) < 0)) {
        puts("Error registering at network interface!");
        return -1;
    }

#ifdef MODULE_NETIF
    gnrc_netreg_entry_t dump = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                                          gnrc_pktdump_pid);
    gnrc_netreg_register(GNRC_NETTYPE_CCN_CHUNK, &dump);
#endif

    return 0;
}
```

miri64, 2018-03-08:
BTW there is no module `netif`, so the `#ifdef MODULE_NETIF` is always excluded. What you are looking for is `MODULE_GNRC_PKTDUMP`, I think.

maribu, 2019-01-22:
@Citrullin: Did @miri64 reply solve your problem?

Citrullin, 2019-01-23:
@maribu Since you are replacing the driver and this works fine, I would just say: Close this ticket. Not interesting for me anymore. But as I remember this issue isn't solved. I wasn't able to use the CC1101x with the old driver. Probably just because of the complicated drive or I just didn't understand RIOT at this point very well.

maribu, 2019-01-23:
Then we can close this issue once the new driver is ready and merged :-)

closed 2019-08-20 by PeterKietzmann