RIOT-OS/RIOT #8755: [wiki]: Broken MIPS toolchain links

jia200x, 2018-03-08:
#### Description

I just noticed all links pointing to MIPS toolchain are broken, as seen [in the wiki entry](https://github.com/RIOT-OS/RIOT/wiki/Family%3A-MIPS). They are pointing to the root site, so it seems the site changed the URL format.
#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

aabadie, 2018-10-19:
Would it make sense to use the following link ?
https://codescape.mips.com/components/toolchain/2017.10-08/downloads.html

gebart, 2018-10-19:
For the latest version, go to https://www.mips.com/develop/tools/codescape-mips-sdk/ and select "Codescape GNU Toolchain" (which is the same link that @aabadie posted above)

aabadie, 2018-10-19:
Thanks @gebart. I edited the wiki page with new instructions. If it's ok, we can close this issue.

gebart, 2018-10-19:
Looks good to me, closing

closed 2018-10-19 by gebart