RIOT-OS/RIOT #8761: io1_xplained: crashes with SAUL

miri64, 2018-03-08:
#### Description
I'm able to crash a `samr21-xpro` with a `io1_xplained` attached, when in `examples/saul`.

#### Steps to reproduce the issue
I built `examples/saul` with the `io1_explained` module for `samr21-xpro` and flash it:

```bash
BOARD=samr21-xpro USEMODULE=io1_xplained make -C examples/saul/ flash term
```

When typing in `saul` I already get weird output

```
> saul
2018-03-08 18:02:31,967 - INFO #  saul
2018-03-08 18:02:31,968 - INFO # ID	Class		Name
2018-03-08 18:02:31,971 - INFO # #0	ACT_SWITCH	LED(orange)
2018-03-08 18:02:31,973 - INFO # #1	SENSE_BTN	Button(SW0)
2018-03-08 18:02:31,977 - INFO # #2	SENSE_TEMP	Temperature (IO1 Xplained)
2018-03-08 18:02:31,980 - INFO # #3	CLASS_UNKNOWN	Temperature (IO1 Xplained)
2018-03-08 18:02:31,984 - INFO # #4	CLASS_UNKNOWN	LED (IO1 Xplained)
2018-03-08 18:02:31,987 - INFO # #5	CLASS_UNKNOWN	GPIO1 (IO1 Xplained)
```

The temperature sensor is shown twice and the last three are listed as `CLASS_UNKNOWN`, when I address one of the `CLASS_UNKNOWN` devices (either writing or reading) the node crashes. E.g.:

```
> saul read 2
2018-03-08 18:04:47,408 - INFO #  saul read 2
2018-03-08 18:04:47,415 - INFO # Reading from #2 (Temperature (IO1 Xplained)|SENSE_TEMP)
2018-03-08 18:04:47,416 - INFO # Data:	[0] 25.62°C
> saul read 3
2018-03-08 18:04:50,912 - INFO #  saul read 3
2018-03-08 18:04:50,912 - INFO # 
2018-03-08 18:04:50,915 - INFO # Context before hardfault:
2018-03-08 18:04:50,916 - INFO #    r0: 0x20000224
2018-03-08 18:04:50,918 - INFO #    r1: 0x20000af0
2018-03-08 18:04:50,920 - INFO #    r2: 0x00000003
2018-03-08 18:04:50,921 - INFO #    r3: 0x20000200
2018-03-08 18:04:50,923 - INFO #   r12: 0xffffffff
2018-03-08 18:04:50,925 - INFO #    lr: 0x000018b7
2018-03-08 18:04:50,926 - INFO #    pc: 0x20000200
2018-03-08 18:04:50,928 - INFO #   psr: 0x20000000
2018-03-08 18:04:50,928 - INFO # 
2018-03-08 18:04:50,928 - INFO # Misc
2018-03-08 18:04:50,930 - INFO # EXC_RET: 0xfffffffd
2018-03-08 18:04:50,934 - INFO # Attempting to reconstruct state for debugging...
2018-03-08 18:04:50,935 - INFO # In GDB:
2018-03-08 18:04:50,937 - INFO #   set $pc=0x20000200
2018-03-08 18:04:50,938 - INFO #   frame 0
2018-03-08 18:04:50,938 - INFO #   bt
2018-03-08 18:04:50,939 - INFO # 
2018-03-08 18:04:50,942 - INFO # ISR stack overflowed by at least 8 bytes.
2018-03-08 18:04:56,943 - INFO # main(): This is RIOT! (Version: 2018.04-devel-617-g19cf2-sarajevo)
2018-03-08 18:04:56,945 - INFO # Welcome to RIOT!
2018-03-08 18:04:56,945 - INFO # 
2018-03-08 18:04:56,950 - INFO # Type `help` for help, type `saul` to see all SAUL devices
2018-03-08 18:04:56,950 - INFO # 
> saul write 4 0
2018-03-08 18:05:05,241 - INFO #  saul write 4 0
2018-03-08 18:05:05,245 - INFO # Writing to device #4 - LED (IO1 Xplained)
2018-03-08 18:05:05,246 - INFO # Data:	[0] 0
2018-03-08 18:05:05,247 - INFO # 
2018-03-08 18:05:05,249 - INFO # Context before hardfault:
2018-03-08 18:05:05,251 - INFO #    r0: 0x20000ae0
2018-03-08 18:05:05,253 - INFO #    r1: 0x20000ae4
2018-03-08 18:05:05,254 - INFO #    r2: 0x77777777
2018-03-08 18:05:05,256 - INFO #    r3: 0x40001108
2018-03-08 18:05:05,257 - INFO #   r12: 0x00000003
2018-03-08 18:05:05,259 - INFO #    lr: 0x00000c19
2018-03-08 18:05:05,261 - INFO #    pc: 0x000002a0
2018-03-08 18:05:05,262 - INFO #   psr: 0x21000000
2018-03-08 18:05:05,262 - INFO # 
2018-03-08 18:05:05,263 - INFO # Misc
2018-03-08 18:05:05,265 - INFO # EXC_RET: 0xfffffffd
2018-03-08 18:05:05,269 - INFO # Attempting to reconstruct state for debugging...
2018-03-08 18:05:05,270 - INFO # In GDB:
2018-03-08 18:05:05,271 - INFO #   set $pc=0x2a0
2018-03-08 18:05:05,272 - INFO #   frame 0
2018-03-08 18:05:05,273 - INFO #   bt
2018-03-08 18:05:05,273 - INFO # 
2018-03-08 18:05:05,276 - INFO # ISR stack overflowed by at least 8 bytes.
```

Some of the resolvable addresses seen above resolved (lines refer to state in of master as of writing this issue [19cf25c7fb675dd11b4ba5f09df83b46d9be]):

* `0x20000224`: drivers/io1_xplained/include/io1_xplained_params.h:72
* `0x000018b7`: sys/saul_reg/saul_reg.c:121
* `0x00000c19`: core/thread.c:222
* `0x000002a0`: cpu/cortexm_common/thread_arch.c:139

#### Expected results
Node should not crash but in the worst case output an error message

#### Actual results
Node crashes

#### Versions
Operating system: Ubuntu 16.04.4
Build environment: 
```
Installed compiler toolchains 
-----------------------------
           native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
    arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
              avr-gcc: avr-gcc (GCC) 4.9.2
     mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
           msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
                clang: clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)

Installed compiler libs
-----------------------
 arm-none-eabi-newlib: "2.5.0"
  mips-mti-elf-newlib: "2.1.0"
             avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                cmake: cmake version 3.5.1
             cppcheck: Cppcheck 1.72
                  git: git version 2.7.4
           coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

aabadie, 2018-03-09:
I could reproduce the issue and fixed it with #8762

aabadie, 2018-03-09:
There was a position mismatch in saul entries register, an invalid comparison in the assert and the gpio driver was wrong.

PR #8762 (2018-03-09): sys/auto_init: fix crash with io1_xplained driver

### Contribution description

This PR fixes the saul auto initialization of the Atmel IO1 Xplained extension driver.

### Issues/PRs references

Fixes #8761
closed 2018-03-09 by miri64