RIOT-OS/RIOT #8764: 6lowpan : loss of the default router

LalunePierr0t, 2018-03-09:
Hello,

#### Description
I m using RIOT OS ( RIOT-2017.04 ) on 2 border routers and several nodes with 6lowpan.
From the node, if I loose my router ( non response on ICMP ping or COAP request for example), I want my node discover the second router as quick as possible. 
In other words, I would say my goal is to find any router available as quick as possible, if I lost the current on.

#### Steps to reproduce the issue
HW prerequisites :

- 2  routers, switched on

- 1 node node switched on

Steps :
1. on the the node discover a router 
2. ping the discovered router
2. switch off the discovered router

#### Expected results
The second router is discovered immediately.

#### Actual results
The second router is NOT discovered. 
If I reboot the node it is discovered immediately 

#### Versions

RIOT :  RIOT-2017.04

```
/eisoxRIOT/dist/tools/ci$ ./print_toolchain_versions.sh 
Installed toolchain versions
----------------------------
          native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
          msp430-gcc: missing/error
             avr-gcc: missing/error
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: missing/error
./print_toolchain_versions.sh: 81: ./print_toolchain_versions.sh: clang: not found
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: missing/error (missing/error)
            cppcheck: missing
          coccinelle: missing
```

miri64, 2018-03-09:
Can you try with `CFLAGS=-DGNRC_IPV6_NIB_DEFAULT_ROUTER_NUMOF=2`, please?

miri64, 2018-03-09:
(at the node)

miri64, 2018-03-09:
(Otherwise there is no space for a second router in the network information base)

bergzand, 2018-03-09:
@miri64 `GNRC_IPV6_NIB_DEFAULT_ROUTER_NUMOF` requires the 2018.01 release (or at least the netif/nib overhaul) right? The issue states that 2017.04 is used for the setup.

miri64, 2018-03-09:
Ah... then @LalunePierr0t should go to the most current release, because the ND on 2017.04 is known to be broken.

miri64, 2018-03-09:
(and next to impossible to fix, that's why we re-modeled the Neighbor Discovery in 2018.01)

PeterKietzmann, 2019-02-21:
Pin @LalunePierr0t any update on this?

miri64, 2019-08-27:
Judging from my comments, this should be fixed and there is no news from @LalunePierr0t other than OP. So I think it is safe to close this issue. Reopen if you disagree.

closed 2019-08-27 by miri64