RIOT-OS/RIOT #8767: gcoap and c++

llueder, 2018-03-11:
Hi all!

I am quite new to RIOT and had troubles buildung a gcoap application with c++.

When trying to compile the gcoap example (on current master) with c++, I got compiler errors regarding atomic types, because `rmutex.h` includes `stdatomic.h`, when c++ needs `<atomic>`. After fixing this using [1], no copy assignment operator for the `rmutex_t` struct could be generated (the atomic int member is to blame), so I wrote a dummy for that.

Next I had to adapt gcoap.h. Again, different atomic headers had to be chosen for C and C++.
After that, a few issues in nanocoap.h had to be fixed:
* a struct member `uint8_t data[]` is not allowed in ISO-C++
* a function argument called `class`

After that I was able to build. (it crashed in the first coap request, though. I blame my dummy cpy-assignment, which was just written to be compilable.)

My code with the modified example and (halfway) fixes is on https://github.com/llueder/RIOT.
I tested native and pba-d-01-kw2x.

[1] https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60932

gebart, 2018-03-11:
I don't think you are doing anything out of the ordinary regarding C++. These are valid issues for gcoap in my opinion.
- Re: stdatomic.h: the gcoap_state_t struct looks like it is only needed from inside the gcoap implementation and could be moved to either gcoap.c or a new implementation use only header.
- `class` as a variable name is definitely a mistake and should be renamed.
- the flexible array member in nanocoap needs investigating how to solve for C++. It is valid C99, but not valid C++

kb2ma, 2018-03-29:
Thanks for pointing out the unnecessary external, @gebart! Created #8854 to address this.

kb2ma, 2018-03-31:
#8854 merged, so first issue resolved.

aabadie, 2018-10-19:
I opened #10199 to tackle one of the issues reported.

Regarding `uint8_t data[], would it be valid to use `uint8_t *data` instead ?

llueder, 2018-10-21:
Not as a direct replacement, because until now `data` is the address of the first array entry in the `coap_hdr_t` struct. With `data` defined as a pointer, the actual data would be used as an address.

Hence, the use of `data` has to be changed to `&data` as well, since the C shortcut doesn't work anymore.

But the zero-sized array does not contribute to the struct's size. A `uint8_t*` or `uint8_t` or anything else does.
I'm not familiar with the CoAP code, could this lead to new/other problems, e.g. when the data is accessed when there is no data? Does this new behavior change this already invalid case and can it happen in the first place?

Is here the right place for this discussion or would a PR (with my proposed [fix](https://github.com/llueder/RIOT/commit/2dc4e5fc7de2df8d2a690aa3bfa0bcf3a87f8bd9)) be a better place?

kb2ma, 2018-10-24:
@llueder, @aabadie: thanks for moving this forward. My comments here are as someone who has not dealt with C / C++ coexistence issues in the past.

I am not the author of coap_hdr_t, but I expect @kaspar030 wrote it that way to provide a zero-cost reference to the rest of the packet (PDU) data. So I think reworking the member generically so it takes up space will be a tough sell.

Yes, a CoAP message may be "Empty", and consist of only a 4-byte header. So, there may not be any data.

The problem really is an incompatibility with C++, and the data member is accessed in the code only in a couple of places. I wonder if the best solution is some sort of #ifdef on the struct to isolate the handling for C++. Maybe define it as a uint8_t* only in that case, and similarly #ifdef the handling code as required.

I think putting this work in a PR will make the problem more concrete for everyone's review. Hopefully we can find consensus on the least painful solution. :-)

llueder, 2018-10-25:
I opened #10248 with an attempt to tackle this problem with an inline function which mimicks the old `data` behavior.

llueder, 2018-11-05:
Now that #10248 was merged my problem in the CoAP-area are fixed, thanks a lot for your help!
I still encounter one more problem outside the scope of CoAP and will open a new issue for that when I find time for it.
--closed--

closed 2018-11-05 by llueder