RIOT-OS/RIOT #8778: sys/random: random_uint32_range is only uniform if range is a power of 2

gebart, 2018-03-14:
#### Description

It does not say so explicitly in the documentation, but most users would assume that this function will return a uniform distributed random number in the interval [a,b).

```c
static inline uint32_t random_uint32_range(uint32_t a, uint32_t b)
{
    return (random_uint32() % (b - a)) + a;
}
```

However, the assumption is only true if `b - a` is a power of two, because of the discrete values (integer) involved. If `b - a` is _not_ a power of two, then the output will be weighted towards the lower end of the range, depending on the distance between `b - a` and the closest power of two less than `b - a`.

#### Steps to reproduce the issue

```c
uint32_t val = random_uint32_range(0, 0xC0000000);
```

#### Expected results

`val` above should have a uniform distribution

#### Actual results

The distribution of `val` above will have more weight in the first quartile.

#### Versions

Current RIOT master branch

gebart, 2018-03-14:
I will fix and add a unit test when I find some time

kaspar030, 2018-04-10:
how did you catch this? :) any news?

gebart, 2018-04-11:
Caught it on paper :) I did some calculations because the expression in the code looked non-uniform. This is really only an issue when the reduced range of numbers is large, otherwise the difference is pretty much impossible to detect from samples.

kaspar030, 2018-04-16:
I'll untag for the release, as I don't think we'd block the release for this.

tcschmidt, 2018-06-26:
Fixed with #9410.

closed 2018-06-26 by tcschmidt