RIOT-OS/RIOT #8826: stm32: Add support for QSPI

maxvankessel, 2018-03-23:
Currently I'm working on a project which has support for serial data flash through a quad spi interface.
The project has a STM32F4 controller, with serial nor flash.

_STM32L476DISCOVERY is provided with an onboard micron serial flash device._

I've seen no implementation for quad spi in the peripheral folders. After digging deeper the mtd_spi_nor driver only support one method of transfer and this is fixed to the conventional spi. 

_Note: The quad spi is quite similar to the conventional spi (in terms of clock polarity and clock phase), and the flash driver is the same. Clock speed is a bit higher, and data lines become bi-directional._

I was wondering if someone has any knowledge of an existing branch which does support this interface?

Or

Give some pointers in how to alter/extend the current driver to support multiple interface (the preferred way, because I couldn't find a "design" document).

Thanks in advance.

vincent-d, 2018-03-23:
From what I know, qspi and spi interfaces are quite different. I took a look some time ago and thought it would be better to add a new periph interface for quadspi peripherals. Then a mtd wrapper would be easy to write.

The main difference between spi and qspi are that the former is full-duplex while the latter is half-duplex. A typical qspi transaction is: instruction -> address -> alternate bytes -> dummy cycles -> data (read or write), which is quite different from a typical spi transaction where you only read and write n bytes at the same time.

closed 2019-09-10 by stale[bot]

benpicco, 2020-04-28:
I'm not sure if a new periph interface is needed - it should suffice if the QSPI driver implements the MTD API.