RIOT-OS/RIOT #8833: Missed files ccn-lite-riot.h and ccnl-pkt-ndntlv.h

redradist, 2018-03-25:
I have faced with the following issue:
./RIOT-OS/sys/shell/commands/sc_ccnl.c:24:10: fatal error: ccn-lite-riot.h: No such file or directory
./RIOT-OS/sys/shell/commands/sc_ccnl.c:24:10: fatal error: ccnl-pkt-ndntlv.h: No such file or directory

#### Description
Cannot find ccn-lite-riot.h: No such file or directory
Cannot find ccnl-pkt-ndntlv.h: No such file or directory

#### Steps to reproduce the issue
Download ROIT-OS 2018-01 and compile example with shell

#### Expected results
Files to be found

#### Actual results
Files not found

#### Versions
Ununtu

cgundogan, 2018-03-26:
@redradist thank you for this report. Could you however give the master branch of RIOT a shot? I cannot reproduce this issue neither on the master branch nor on the 2018-01 branch. Could you give a pointer to your application's Makefile?

redradist, 2018-03-26:
@cgundogan I have just switched to tag "2018.01" (commit e4fb634e3f18067331fccb968559dff35d989697) and there are not files **ccn-lite-riot.h** and **ccnl-pkt-ndntlv.h**

About Makefile it is not easy cause I have made a build of RIOT based on CMake system.
If it would be useful I can provide it

cgundogan, 2018-04-04:
Yup, a link to your customized build system would greatly help in pinpointing the problem. Maybe it's just a missing `INCLUDE` or a wrong dependency in the Cmake file?

redradist, 2018-04-14:
@cgundogan It is not a missing include file. I cannot find manually "ccn-lite-riot.h" and "ccnl-pkt-ndntlv.h"

This error happens during compiling ./RIOT-OS/sys/shell/commands/sc_ccnl.c
In this file included:

    #include "ccn-lite-riot.h"
    #include "ccnl-pkt-ndntlv.h"

And those files are absent at the file hierarcy

cgundogan, 2018-04-15:
>>#include "ccn-lite-riot.h"
#include "ccnl-pkt-ndntlv.h"

>And those files are absent at the file hierarcy

@redradist Those files are not in the code base of RIOT, but exist [here](https://github.com/cn-uofbasel/ccn-lite/blob/master/src/ccnl-riot/include/ccn-lite-riot.h) and [here](https://github.com/cn-uofbasel/ccn-lite/blob/master/src/ccnl-pkt/include/ccnl-pkt-ndntlv.h). CCN-lite is an external package from the perspective of RIOT, i.e., CCN-lite is cloned during the build process. That's probably what's missing in your CMake based build system?

See [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/ccn-lite/Makefile#L21) and [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/pkg.mk) for how the package is cloned in RIOT.

cgundogan, 2018-04-28:
@redradist were you able to resolve your issue? Can we close this?

cgundogan, 2018-06-18:
@redradist I am closing this because I think the issue is resolved. Feel free to re-open if you think otherwise.

closed 2018-06-18 by cgundogan

redradist, 2018-06-20:
@cgundogan Yes, an issue is closed it is my mistake