RIOT-OS/RIOT #8850: Race condition RIOT and Radvd

Josar, 2018-03-28:
In my setup i have an Jiminy-atmega256rfr2 and a RPI3 with openlabs transceiver.

On the RPI i have WPAN tools and radvd as native IPv6 6LBR.
On the Jiminy i use the gnrc_networking example.

I just constantly ping the node from the rpi.
After some time there starts a race condition between the two. 
It seem like constantly RS from RIOT and RA from RPI, NS from RIOT and NA from RPI. 
This can be reproduced it just needs some time.

Any tips on how to debug? 
Which Module to set to debug and where the culprit might be hidden?

[image]

Best regards
Josua

miri64, 2018-03-29:
I need some more infos here (but in general: yes, setting the module's `ENABLE_DEBUG` is sometimes the best bet):

1. What exactly is the race condition? At the moment it looks more like the timer is somehow broken on the jiminy for me. But I'm of even that I'm not sure, since you cut of the time in your PCAP dump ;-).
2. What is the radio you are using? From what I can see, the Jiminy comes with an on-board radio, but I'm unsure if we support that yet.
3. Do you use `gnrc_networking` as is?

Best regards,
Martine

Josar, 2018-03-29:
1. The race condition is that they two just continue with rs ra, ns na and start rs na ...
As fast as they can transmit packages.

This just loops until msg_que is full. Then some interrupt are allegedly lost.

2. I implement at86rfr2 support into at86rfr2xx which is currently under review.

3. Yes, as it is. Only added assert verbose.

Where in the net implementation is the decision made to send an rs and accept ra? So I can enable the debug massages there?

miri64, 2018-03-29:
> Where in the net implementation is the decision made to send an rs and accept ra? So I can enable the debug massages there?

Just start with `sys/net/gnrc/network_layer/ipv6/nib/nib.c` (the central pieces of the NDP protocol implementation are here), if that doesn't give you enough information, add `sys/net/gnrc/network_layer/ipv6/nib/_nib_arsm.c` (this is the implementation of the **a**ddress **r**esolution **s**tate **m**achine [ARSM for short]; neighbor solicitations sending is triggered from here), `sys/net/gnrc/network_layer/ipv6/nib/_nib-router.c` (router advertisement sending is triggered from here) or `sys/net/gnrc/network_layer/ndp/gnrc_ndp.c` (the actual message construction of all NDP messages and subsequent sending is happening here, but no context why this happens is provided).

Josar, 2018-03-29:
It starts after ifconfig.

There are a lot `Handle timer event`.
```
on address: fd00:1:1:1:dedc:20b7:3c21:8b3e
     - Flags: RS-
nib: Handle packet (icmpv6->type = 134)
nib: Received valid router 
nib: Initialize interface 7
nib: add address based on fe80::/64 automatically to interface 7
main(): This is RIOT! (Version: 2018.04-devel-722-g05e14-josua-VBox-jiminy_master)
RIOT network stack example application
All up, running the shell now
> nib: Handle packet (icmpv6->type = 134)
nib: Received valid router advertisement:
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fe80::dedc:20b7:3c21:8b3e
     - Cur Hop Limit: 255
     - Flags: --
     - Router Lifetime: 1800s
     - Reachable Time: 0ms
     - Retrans Timer: 0ms
nib: received valid Prefix Information option:
     - Prefix: fd00:1:1:1::/64
     - Flags: -A
     - Valid lifetime: 86400
     - Preferred lifetime: 14400
nib: add address based on fd00:1:1:1::/64 automatically to interface 7
nib: Handle timer event (ctx = 0x499d, type = 0x4fc3, now = 29ms)
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-link or in NC, start address resolution
nib: resolve address fe80::1ac0:ffee:d:ecaf%7 from neighbor cache
nib: Handle timer event (ctx = 0x49b9, type = 0x4fcf, now = 43ms)
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-link or in NC, start address resolution
nib: resolve address fe80::1ac0:ffee:d:ecaf%7 from neighbor cache
nib: Handle packet (icmpv6->type = 136)
nib: Received valid neighbor advertisement:
     - Target address: fe80::1ac0:ffee:d:ecaf
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fd00:1:1:1:dedc:20b7:3c21:8b3e
     - Flags: RS-
nib: Handle packet (icmpv6->type = 136)
nib: Received valid neighbor advertisement:
     - Target address: fe80::1ac0:ffee:d:ecaf
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fd00:1:1:1:dedc:20b7:3c21:8b3e
     - Flags: RS-
nib: Handle timer event (ctx = 0x49b9, type = 0x4fcf, now = 0ms)
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-link or in NC, start address resolution
nib: resolve address fe80::1ac0:ffee:d:ecaf%7 from neighbor cache
nib: Handle packet (icmpv6->type = 135)
nib: Received valid neighbor solicitation:
     - Target address: fe80::dedc:20b7:3c21:8b3e
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fe80::dedc:20b7:3c21:8b3e
nib: Storing SL2AO for later handling
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-link or in NC, start address resolution
nib: resolve address fe80::1ac0:ffee:d:ecaf%7 from neighbor cache
nib: Handle packet (icmpv6->type = 136)
nib: Received valid neighbor advertisement:
     - Target address: fe80::1ac0:ffee:d:ecaf
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fd00:1:1:1:dedc:20b7:3c21:8b3e
     - Flags: RS-
nib: Handle packet (icmpv6->type = 135)
nib: Received valid neighbor solicitation:
     - Target address: fd00:1:1:1:dedc:20b7:3c21:8b3e
[... 280 lines cut ...]
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-link or in NC, start address resolution
nib: resolve address fe80::1ac0:ffee:d:ecaf%7 from neighbor cache
nib: Handle timer event (ctx = 0x499d, type = 0x4fc3, now = 57ms)
nib: Handle timer event (ctx = 0x6e48, type = 0x4fc0, now = 14ms)
nib: Handle timer event (ctx = 0x499d, type = 0x4fc3, now = 31ms)
nib: Handle timer event (ctx = 0x6e48, type = 0x4fc7, now = 48ms)
nib: Handle packet (icmpv6->type = 134)
nib: Received valid router advertisement:
     - Source address: fe80::1ac0:ffee:d:ecaf
     - Destination address: fe80::dedc:20b7:3c21:8b3e
     - Cur Hop Limit: 255
     - Flags: --
     - Router Lifetime: 1800s
     - Reachable Time: 0ms
     - Retrans Timer: 0ms
nib: received valid Prefix Information option:
     - Prefix: fd00:1:1:1::/64
     - Flags: -A
     - Valid lifetime: 86400
     - Preferred lifetime: 14400
nib: add address based on fd00:1:1:1::/64 automatically to interface 7
nib: get next hop link-layer address of fe80::1ac0:ffee:d:ecaf%7
nib: fe80::1ac0:ffee:d:ecaf is on-l
```

Josar, 2018-03-29:
When adding debug massages only to `_nib_arsm.c` the race conditions stops after some time.
Then starts again. Stops ...

Its between `nib neigh`.
In wireshark i see a lot of RS RA, NS NA.

```

main(): This is RIOT! (Version: 2018.04-devel-722-g05e14-josua-VBox-jiminy_master)
RIOT network stack example application
All up, running the shell now
> nib: Creating NCE for (ipv6 = fe80::1ac0:ffee:d:ecaf, iface = 7, nud_state = STALE)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib neigh
fe80::1ac0:ffee:d:ecaf dev #7 lladdr 18:c0:ff:ee:00:0d:ec:af router REACHABLE GC
> 
> nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout reachability
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Retransmit neighbor solicitation
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Retransmit neighbor solicitation
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Retransmit neighbor solicitation
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Retransmit neighbor solicitation
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Retransmit neighbor solicitation
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Retransmit neighbor solicitation
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
[... 599 lines cut ...]
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Retransmit neighbor solicitation
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Retransmit neighbor solicitation
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Timeout DELAY state
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Retransmit neighbor solicitation
nib: Probing nib: unicast to fe80::1ac0:ffee:d:ecaf (retrans. timer = 1000ms)
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: fe80::1ac0:ffee:d:ecaf%7 is a router
nib: Set fe80::1ac0:ffee:d:ecaf%7 to REACHABLE for 21720ms
nib: Timeout reachability
nib neigh
fe80::1ac0:ffee:d:ecaf dev #7 lladdr 18:c0:ff:ee:00:0d:ec:af router STALE GC
> 
> 
```

miri64, 2018-03-29:
This looks exactly like I suspected: there seems to be something wrong with `evtimer` or `xtimer` on the Jiminy board. Basically, the timers are firing way to fast. Are you sure the timers are working correctly on that board?

Josar, 2018-03-29:
First thank you, second i will investigated.

xtimer_usleep with pin toogle is in expected range.
´´´
main(): This is RIOT! (Version: 2018.04-devel-722-g05e14-josua-VBox-jiminy_master)
Running test 5 times with 3 distinct sleep times
Please hit any key and then ENTER to continue
Slept for 10272 us (expected: 10000 us)
Slept for 50224 us (expected: 50000 us)
Slept for 100224 us (expected: 100000 us)
Slept for 10224 us (expected: 10000 us)
Slept for 50224 us (expected: 50000 us)
Slept for 100216 us (expected: 100000 us)
Slept for 10224 us (expected: 10000 us)
Slept for 50224 us (expected: 50000 us)
Slept for 100224 us (expected: 100000 us)
Slept for 10224 us (expected: 10000 us)
Slept for 50224 us (expected: 50000 us)
Slept for 100224 us (expected: 100000 us)
Slept for 10224 us (expected: 10000 us)
Slept for 50224 us (expected: 50000 us)
Slept for 100224 us (expected: 100000 us)
Test ran for 956016 us
´´´

evtimer_msg 
```
is RIOT! (Version: 2018.04-devel-722-g05e14-josua-VBox-jiminy_master)
Testing generic evtimer (start time = 21 ms
main(): This is RIOT! (Version: 2018.04-devel-722-g05e14-josua-VBox-jiminy_master)
Testing generic evtimer (start time = 21 ms)
Are the reception times of all 4 msgs close to the supposed values?
If yes, the tests were successful
At   5306 ms received msg 0: "supposed to be 659"
At   7794 ms received msg 1: "supposed to be 1000"
At  12026 ms received msg 2: "supposed to be 1500"
At  31906 ms received msg 3: "supposed to be 3954"
```

Any ideas why this drifffts ?

gebart, 2018-03-29:
Try running the benchmark in #8531 on your board. I found some pretty bad bugs in the Kinetis timer implementation using that application.

Josar, 2018-03-29:
The output suggests that there is something wrong, i have to get a better understanding of this test to be able to interpret where the error might come from.

```
Statistical benchmark for timers
Running timer test with seed 123 using Tiny Mersenne Twister PRNG.
TEST_MIN = 16
TEST_MAX = 128
TEST_MIN_REL = 0
TEST_MAX_REL = 112
TEST_NUM = 113
log2(TEST_NUM - 1) = 6
state vector elements per variant = 7
number of variants = 8
sizeof(state) = 32 bytes
state vector total memory usage = 1792 bytes
TIM_TEST_DEV = 1, TIM_TEST_FREQ = 1000000, TIM_TEST_CHAN = 0
TIM_REF_DEV  = 0, TIM_REF_FREQ  = 1000000
USE_REFERENCE = 1
TEST_PRINT_INTERVAL_TICKS = 30000000
Expected error variance due to truncation in tick conversion: 0
Calibrating spin delay...
spin_max = 64
Estimating benchmark overhead...
overhead_target = 49 (s2 = 0)
overhead_read = 50 (s2 = 0)
exp_mask = 00000007
max interval = 255
------------- BEGIN STATISTICS --------------
Limits: mean: [-10, 10], variance: [0, 16]
Target error (actual trigger time - expected trigger time), in reference timer ticks
positive: timer is late, negative: timer is early
=== timer_set running ===
   interval    count       sum       sum_sq    min   max  mean  variance
  16 -   17:     464     -5698          226    -13   -11   -12      0  <=== SIC!
  18 -   19:     481     -5862          257    -13   -11   -12      0  <=== SIC!
  20 -   23:     460     -5652          201    -13   -11   -12      0  <=== SIC!
  24 -   31:     429     -4866          965    -13    -8   -11      2  <=== SIC!
  32 -   47:     450      2196        19503    -13    14     4     43  <=== SIC!
  48 -   79:     400      5155       112878    -13    36    12    282  <=== SIC!
  80 -  143:     351     -1117        35250    -13    17    -3    100  <=== SIC!
      TOTAL     3035    -15844       424125    -13    36    -5    139  <=== SIC!
=== timer_set resched ===
   interval    count       sum       sum_sq    min   max  mean  variance
  16 -   17:     469     -5316       198335    -13   433   -11    423  <=== SIC!
  18 -   19:     443     -5414          246    -13   -11   -12      0  <=== SIC!
  20 -   23:     469     -5744          198    -13   -11   -12      0  <=== SIC!
  24 -   31:     432     -5489          311    -13   -11   -12      0  <=== SIC!
  32 -   47:     448      2347        15634    -13    14     5     34  <=== SIC!
  48 -   79:     417      5045       124478    -13    36    12    299  <=== SIC!
  80 -  143:     309     -1061        32753    -13    17    -3    106  <=== SIC!
      TOTAL     2987    -15632       627493    -13   433    -5    210  <=== SIC!
=== timer_set stopped ===
   interval    count       sum       sum_sq    min   max  mean  variance
  16 -   17:     402     -8573          131    -22   -21   -21      0  <=== SIC!
  18 -   19:     442     -9475          246    -22   -20   -21      0  <=== SIC!
  20 -   23:     450       796        15201    -22     5     1     33  <=== SIC!
  24 -   31:     485     -1381         3008     -7     1    -2      6
  32 -   47:     464     -6808         9275    -22    -7   -14     20  <=== SIC!
  48 -   79:     437      4673        47789    -22    27    10    109  <=== SIC!
  80 -  143:     338     -6189         7717    -22    -5   -18     22  <=== SIC!
      TOTAL     3018    -26957       503721    -22    27    -8    166  <=== SIC!
=== timer_set resched, stopped ===
   interval    count       sum       sum_sq    min   max  mean  variance
  16 -   17:     437     -9300          121    -22   -21   -21      0  <=== SIC!
  18 -   19:     449     -9610          255    -22   -20   -21      0  <=== SIC!
  20 -   23:     436       499        19612    -22     5     1     45  <=== SIC!
  24 -   31:     450     -1222         2641     -7     1    -2      5
  32 -   47:     452     -6598         9455    -22    -7   -14     20  <=== SIC!
  48 -   79:     451      4701        44589    -22    27    10     99  <=== SIC!
  80 -  143:     379     -7070         8074    -22    -5   -18     21  <=== SIC!
[... 34 lines cut ...]
  18 -   19:     479     -4662          477    -11    -9    -9      0
  20 -   23:     401     -5152          843    -15   -11   -12      2  <=== SIC!
  24 -   31:     478     -8838         2140    -22   -15   -18      4  <=== SIC!
  32 -   47:     468      8419        34431    -22    27    17     73  <=== SIC!
  48 -   79:     425     -2080        35089    -21    11    -4     82  <=== SIC!
  80 -  143:     368     -7789          185    -22   -20   -21      0  <=== SIC!
      TOTAL     3048    -23434       510766    -22    27    -7    167  <=== SIC!
=== timer_read statistics ===
Limits: mean: [-10, 10], variance: [0, 16]
timer_read error (TUT time elapsed - expected TUT interval), in timer under test ticks
positive: timer target handling is slow, negative: timer_read is dropping ticks
function              count       sum       sum_sq    min   max  mean  variance
 timer_set            11857   -144083   4291044816    -32 65479   -12 361930  <=== SIC!
  running              2964    -45619       419716    -23    26   -15    141  <=== SIC!
  resched              2946    -45280       633554    -23   422   -15    215  <=== SIC!
  stopped              2988      6589   4288663899    -32 65479     2 1435776  <=== SIC!
  resched, stopped     2959    -59773       484520    -32    16   -20    163  <=== SIC!

 timer_set_absolute   12209   -130394   4285654310    -32 65441   -10 351052  <=== SIC!
  running              3029     24330   4282816541    -32 65441     8 1414404  <=== SIC!
  resched              3030    -39908       366095    -32    16   -13    120  <=== SIC!
  stopped              3133    -58644       493332    -32    16   -18    157  <=== SIC!
  resched, stopped     3017    -56172       502150    -32    16   -18    166  <=== SIC!
-------------- END STATISTICS ---------------
```

miri64, 2018-03-29:
I guess this is definitely a timer problem and not a network problem then ;-)

Josar, 2018-04-02:
After #8858 first test run promising. Long run test has to follow.

miri64, 2018-04-13:
With #8858 being merged, this should be fixed.

closed 2018-04-13 by miri64