RIOT-OS/RIOT #8851: test/periph_i2c: cannot use init_master command for nrf52

dylad, 2018-03-29:
#### Description
Cannot use `init_master` command from `test/periph_i2c` on NRF52 based board.

#### Steps to reproduce the issue
call `init_master 0 1` (0 is first I2C bus available and 1 stand for `I2C_SPEED_NORMAL` so 100kbits/s)

#### Expected results
I2C is properly initialized.

#### Actual results
```
init_master 0 1
Error: Init: Unsupported speed value
```

#### Versions
current RIOT master

Current I2C implementation from NRF52 supports `I2C_NORMAL_SPEED` and `I2C_FAST_SPEED`.
From what I understand, nrf52 defines its own i2c_speed_t enum
```
typedef enum {
    I2C_SPEED_LOW       = 0xff,                         /**< not supported */
    I2C_SPEED_NORMAL    = TWIM_FREQUENCY_FREQUENCY_K100,    /**< 100kbit/s */
    I2C_SPEED_FAST      = TWIM_FREQUENCY_FREQUENCY_K400,    /**< 400kbit/s */
    I2C_SPEED_FAST_PLUS = 0xfe,                         /**< not supported */
    I2C_SPEED_HIGH      = 0xfd,                         /**< not supported */
} i2c_speed_t;
```
When we use ` test/periph_i2c`, this app expects 1 for `I2C_SPEED_NORMAL`, 2 for` I2C_SPEED_FAST` etc.
But since nrf52 has its own enum with custom value in order to ease the init in the driver I guess, this test app failed.

As far as I can see, we can either change the behaviour of the `test/periph_i2c` or rewrite the driver.

Additional notes :
This issue may also exists for efm32  and nrf51 family since they define their own `i2c_speed_t`

dylad, 2018-03-29:
@basilfx Could you confirm this issue also exists on EFM32 please ?

basilfx, 2018-03-29:
This would also affect EFM32, since it maps the enum values to speed values.

dylad, 2018-03-29:
Thanks @basilfx ! I've update the tittle.
What's the best workaround for now ? modify the driver or the test app ?

basilfx, 2018-03-29:
I've created #8852 to fix this issue. 

Since you may override `i2c_speed_t`, I went for modifying the test application.

PR #8852 (2018-04-03): tests: periph_i2c: map speed to i2c_speed_t
### Contribution description

This PR maps the given speed value to `i2c_speed_t`, so it initializes I2C master with a proper value, since the specs allows one to [override](https://github.com/RIOT-OS/RIOT/blob/master/drivers/include/periph/i2c.h#L114) `i2c_speed_t`.

Tested on SLSTK3401A.

### Issues/PRs references

Should fix #8851 for NRF52 and EFM32 (they override `i2c_speed_t`).
closed 2018-04-03 by dylad