RIOT-OS/RIOT #8859: cc1101 work instability

hahchenchen, 2018-04-01:
Hi,
i use cc1101 on nucleo144-f429, when i use ccn-lite-relay to communicate with another board, sometimes i can receive the content, but sometimes i can't, i think maybe cc1101 is not working reliably.
Is there any solution or advice for that? 

Thank you.

maribu, 2018-05-25:
Hi,
can you please check if it works with [my cc110x-dev branch]? I have fixed some timing issues. Sadly, on the MSB IoT board there is still an SPI issue (or at least I believe its SPI related). But the fixes solve at least some of the reasons for instability on my boards.

Kind regards,
Marian

maribu, 2018-05-29:
@hahchenchen: I think I had the same issue.

Please check PR https://github.com/RIOT-OS/RIOT/pull/9230 or PR https://github.com/RIOT-OS/RIOT/pull/9231 (both solve the same issue in different ways, so either should work if you have the same problem I had).

PeterKietzmann, 2018-05-29:
@hahchenchen please confirm so we can close this issue as duplicate.

PeterKietzmann, 2019-02-21:
There is #10340 which works quite stable even with the IP stack on top. Furthermore we leave #8754 open until the rewrite of the driver is merged. Now that @hahchenchen didn't respond anymore and #8754 reported defect of the radio device earlier, I'm closing this issue. @hahchenchen please give a sign if you disagree.

closed 2019-02-21 by PeterKietzmann