RIOT-OS/RIOT #8876: spiffs: compile problems on new platform xtensa-lx106 (ESP8266)

gschorcht, 2018-04-04:
I ran into compilation problems when I tried to use ```spiffs``` with my MTD flash device implementation in my RIOT port for the ESP8266 (xtensa-lx106). I got a lot of the following error message, one for each standard type.

```
.../newlib-xtensa/xtensa-lx106-elf/include/stdatomic.h:187:9: error: function definition declared 'typedef'
  typedef _Atomic(_Bool)   atomic_bool; 
```

Reason is propably that the xtensa version of gcc is not fully C11 standard compliant.

```xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6) 4.8.5```

The newlib version is 3.0.0.

I solved the problem for the moment, by adding a new variable ```SPIFFS_STD_OPTION``` in my ```Makefile.include``` and changing the the ```Makefile``` in ```spiffs``` as following:
```
# Compilation of spiffs fails in <stdatomic.h> on some platforms when
# option -std=c11 is set. Use SPIFFS_STD_FLAG to overwrite this option.
ifneq ($(SPIFFS_STD_OPTION), )
    CFLAGS += $(SPIFFS_STD_OPTION)
else
    CFLAGS += -std=c11
endif
```
It would be great if this could be changed also in the master. Otherwise, I could provide it with my upcoming pull request for my RIOT port. Maybe, there is a better way to solve this problem, I had no better idea.

vincent-d, 2018-04-11:
@gschorcht can you test #8924 and check if it fixes your issue?

gschorcht, 2018-04-11:
@vincent-d it seems to work :-)

PR #8924 (2018-04-12): pkg/spiffs: remove c11 dependency and clean-up makefile
### Contribution description

A C11 dependency was added to mask a warning about unnamed struct/union.
It can be solved by disabling `-Wpedantic` for the package, to avoid pulling c11 for compiler which do not support it.
I also cleaned up the makefiles.

### Issues/PRs references

Should fix #8876
closed 2018-04-12 by kaspar030