RIOT-OS/RIOT #8901: tests: OS X testrunner loses target output, many tests fail

kaspar030, 2018-04-09:
#### Description

On OS X, the the terminal doesn't get target output from before the testrunner start delay.

#### Steps to reproduce the issue

For samr21-xpro:

1. set ```TESTRUNNER_START_DELAY``` to something high (e.g., 10)
2. flash e.g., ps_schedstatistics
3. run "BOARD=samr21-xpro make tests"

#### Expected results

After ten seconds, all output of RIOT including "main(): This is RIOT!" appears on screen, test completes successfully.

#### Actual results

Some output is missing, causing the test to fail because of timeout.
Using picocom as terminal seems to work around the problem, probably because it starts way faster.

#### Versions

Seen on @kYc0o's Macbook.

kaspar030, 2018-04-09:
I don't use OSX. Maybe someone who does can investigate?
I assume there's a difference in buffering between Linux and OSX.

@cladmi You've introduced the delay. Any ideas?

kYc0o, 2018-04-09:
To complete the description, here my setup:

```console
Installed compiler toolchains 
-----------------------------
           native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
    arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
              avr-gcc: avr-gcc (GCC) 7.3.0
     mips-mti-elf-gcc: missing
           msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
                clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)

Installed compiler libs
-----------------------
 arm-none-eabi-newlib: "2.5.0"
  mips-mti-elf-newlib: missing
             avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                cmake: cmake version 3.11.0
             cppcheck: missing
              doxygen: missing
               flake8: missing
                  git: git version 2.17.0
           coccinelle: missing
```

cladmi, 2018-04-09:
This is the same problem that @smlng had: https://github.com/RIOT-OS/RIOT/issues/8211

I tried something on a local branch to add a 1s delay before calling main, but it required adding `xtimer` dependency… and I just forgot about it.

And after talking to @kYc0o, it looks like it is another problem than just timing, so I will check with him directly.

kaspar030, 2018-04-09:
Closing as duplicate of #8211.