RIOT-OS/RIOT #8910: tests: Unittests don't build

emmanuelsearch, 2018-04-10:
#### Description
Unittests don't build in current master.
Quick bisect we did with @miri64 points to #8844 as introducing this problem.

#### Steps to reproduce the issue
```
cd tests/unittests
make
```

 or

```
make tests-<module>
```

#### Expected results
It should build.

#### Actual results
It does not build. Sample error:

```
make: *** No rule to make target '/home/vagrant/RIOT/tests/unittests/bin/samr21-xpro/tests-tweetnacl.a', needed by '/home/vagrant/RIOT/tests/unittests/bin/samr21-xpro/tests_unittests.elf'.  Stop.

```

#### Versions
Issue detected on Linux and OS X

bergzand, 2018-04-10:
Confirmed, I noticed the problem here too

cladmi, 2018-04-10:
I was writing an issue about it, and know why murdock tests run also still pass.

I have a fix for it that I will PR.

emmanuelsearch, 2018-04-10:
@cladmi great.

cladmi, 2018-04-10:
They build if you run `make clean all` instead of just `make clean; make all`, that's why murdock could not detect it.
The BUG is hidden because of this line when calling `make clean all`: https://github.com/RIOT-OS/RIOT/blob/2018.01/Makefile.include#L350

Normally it should have failed with other examples, but because of another bug, (BASELIBS values are added after these lines), there was not problem. I will write a proper issue for the real problems.

PR #8911 (2018-04-17): Makefile.include: fix missing target for libraries
Unittests add libraries in 'BASELIBS' which do not have any rules to be built as
they are built by 'application.inc.mk' and the DIRS variable.
So make complains about missing target for the unittests archives.

The fix tells these files are generated when building '$(APPLICATION_MODULE).a'.

The bug was introduced by #8844

Fixes #8910
commit 94214cdc (2018-04-17): Makefile.include: fix missing target for libraries

Unittests add libraries in 'BASELIBS' which do not have any rules to be built as
they are built by 'application.inc.mk', packages and the DIRS variable.
So make complains about missing target for the unittests archives.

The fix tells these files are generated when building '$(APPLICATION_MODULE).a'.

The bug was introduced by #8844

Fixes #8910
closed 2018-04-17 by kaspar030