RIOT-OS/RIOT #8922: Native CAN device does not build when VFS is used

vincent-d, 2018-04-11:
#### Description

Cannot build `candev_linux` when `vfs` is used.
Bisecting returned 93a521c50132e24558e42bc57dcf9c94051f07e3 as the culprit (introduced in #8652)

#### Steps to reproduce the issue

```
cd tests/conn_can
USEMODULE=vfs make BOARD=native
```

#### Expected results
Build OK

#### Actual results
```
/home/vincent/Devel/git/RIOT/cpu/native/can/candev_linux.c: In function ‘_init’:
/home/vincent/Devel/git/RIOT/cpu/native/can/candev_linux.c:165:29: error: ‘PF_CAN’ undeclared (first use in this function); did you mean ‘AF_MAX’?
     dev->sock = real_socket(PF_CAN, SOCK_RAW, CAN_RAW);
                             ^~~~~~
                             AF_MAX
/home/vincent/Devel/git/RIOT/cpu/native/can/candev_linux.c:165:29: note: each undeclared identifier is reported only once for each function it appears in
/home/vincent/Devel/git/RIOT/cpu/native/can/candev_linux.c:198:23: error: ‘AF_CAN’ undeclared (first use in this function); did you mean ‘PF_CAN’?
     addr.can_family = AF_CAN;
                       ^~~~~~
                       PF_CAN
```

#### Versions

```
Installed compiler toolchains 
-----------------------------
           native gcc: gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
    arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
              avr-gcc: missing
     mips-mti-elf-gcc: missing
           msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
                clang: clang version 4.0.1-6 (tags/RELEASE_401/final)
```

cladmi, 2018-04-11:
Indeed, our posix headers do not define the `PF_` macros.

I will verify if defining them would be safe with a handwritten value or if https://github.com/RIOT-OS/RIOT/pull/8652 should be reverted.

cladmi, 2018-04-12:
I missed one thing with posix in the given PR, what is defined in the standard is only symbols not the values.

So even right now, if we look at the values of `AF_INET6`, they are not compatible:
* RIOT: 4
* Linux: 10
* MAC: 30

So using our posix headers in interaction with Linux is problematic. So I will revert the PR.

cladmi, 2018-04-12:
With the reverted, the compilation works:

 `USEMODULE=vfs make -C tests/conn_can BOARD=native`

PR #8940 (2018-04-12): REVERT native: remove non required NATIVEINCLUDES
### Contribution description

This revert #8652 PR because in fact, values of posix `AF_*` symbols, and maybe others are not portable. It will fix #8922.

When writing https://github.com/RIOT-OS/RIOT/pull/8652 PR, I was missing that what is defined in the posix standard is only symbols not values.

So even right now, if we look at the values of AF_INET6, they are not compatible:

  *  RIOT: 4
  *  Linux: 10
  *  MAC: 30

So using our posix headers in interaction with Linux is problematic.

### Further steps

* Add a documentation about it
* Re- [... cut]
closed 2018-04-12 by kYc0o