RIOT-OS/RIOT #8941: tests: split unittests?

kaspar030, 2018-04-12:
#### Description

Currently, RIOT's unittests test application consists of many tests for different parts of the system, e.g., tests-core or tests-relic.

By default all of them (that are compatible with the target board) are built into one big binary, but they can also be built seperately.

There are several issues with this;

1. the big binary is too large for almost all boards, thus the default binary will not even be linked
2. there's no automatic way of treating the individual tests as distinct applications (like our other tests), thus together with 1., without further scripting, CI cannot run the tests
3. there's a lot of hacky stuff going on in the unittests Makefile
4. the tests interact in unforseen ways. e.g., tests-tweetnacl doubles the stacksize, if compiled in, possibly hiding stack issues in other tests. Or, modules included by one tests are not missed by another that uses it unless tests are compiled seperately

The one advantage of having a big binary is just that there's only one binary.

I propose splitting the unittests, turning each of ```tests-*``` into it's own application.

What do you think?

bergzand, 2018-04-12:
What was the original reason for building it as one large application?

With the current problems I agree with the proposed idea of splitting every unittest into a single test. My next question then is if there is a difference between a regular hardware-independent test (using embunit assertion) and a unittest application.

aabadie, 2018-04-12:
> What do you think?

given your arguments, +1

aabadie, 2018-04-12:
what I see as a drawback is that it might significantly increase the number of jobs for a single CI run. Since all sub unit tests will be built for all platforms that support them.
Do you have an idea of that overhead ?

kaspar030, 2018-04-12:
> Do you have an idea of that overhead ?

54 tests, 116 boards -> ~6000 extra builds, not substracting the ones that wouldn't be built anyways. That is indeed significant.

miri64, 2018-04-12:
@kaspar030 

> What do you think?

+1, but the extra builds need to be addressed somehow. Maybe not split them up by suites (after all you can have multiple suites in one unittest) but by topic? E.g. put all non-GNRC network related code into one application, put non-core data structures in one application, put all storage related tests in one application, etc. The question is: Does this scale in the future or will we need to split them up again in the future (see also my next answer).

@bergzand 

> What was the original reason for building it as one large application?

Originally to have most of RIOT compiled in one place and to only have one application required for testing under native, but back then the code base was much smaller (e.g. GNRC or pkg didn't even exist at that point), and the implications of having (emb)unittests + threading were not that clear.

> My next question then is if there is a difference between a regular hardware-independent test (using embunit assertion) and a unittest application.

I don't understand that question.

bergzand, 2018-04-12:
> > My next question then is if there is a difference between a regular hardware-independent test (using embunit assertion) and a unittest application.
> 
> I don't understand that question.

If all unittests are split up in their own test application, is there an architectural difference between something like tests/gnrc_ipv6_nib and one of the unittests? From what I see they both use embunit assertions.

kaspar030, 2018-04-12:
No difference.

miri64, 2018-04-12:
> If all unittests are split up in their own test application, is there an architectural difference between something like tests/gnrc_ipv6_nib and one of the unittests? From what I see they both use embunit assertions.

That depends... For the `tests/unittests` application the real backend is sometimes will-fully left-out, so a the `set_up` routine can reset/redo stuff independent from the backend (e.g. with `gnrc_ipv6_nib` some of the NDP stuff is deactivated / doesn't really make sense in a real world scenario, but for the tests).

cladmi, 2018-07-11:
During a chat with @smlng when I was in Hamburg, we discussed that tests for packages could be moved to the package directory. An idea I like and that keep going toward self contained packages.

smlng, 2018-07-11:
and we could also (think about) restructure  the tests according to the main/root folders, i.e. `tests/core/`, `tests/drivers` ... `tests/sys` and consequently have the unittests within/below the respective subfolder.

gebart, 2018-07-12:
All great ideas in my opinion. Does anyone have the time to do a proof of concept PR?

cladmi, 2018-07-17:
I will do something for moving packages tests to their directory after the release.

smlng, 2018-10-18:
see #10187, too

kaspar030, 2018-11-26:
I'm closing this, as the monolithic unittests application is slowly getting split.

closed 2018-11-26 by kaspar030