RIOT-OS/RIOT #8973: doc: CoAP-RD modules appear on top level

miri64, 2018-04-18:
#### Description
All CoAP-RD sub-modules currently appear at the top-level of the module list:

[image]

This is due to them being assigned to a group `net_rdcli` (via `@ingroup`) that does not exist.

#### Steps to reproduce the issue
Go to http://doc.riot-os.org/modules.html and have a look ;-).

#### Expected results
The CoAP-RD modules should be in there own sub-group under `Networking`

#### Actual results
They appear on the top-level

#### Versions
—

cladmi, 2018-04-19:
Should be fixed when the doc is rebuilt.

bergzand, 2018-04-30:
Doc is rebuild, CoAP RD doesn't appear in the top level anymore

closed 2018-04-30 by bergzand