RIOT-OS/RIOT #8981: examples: gcoap: stack overflow when NON times out

miri64, 2018-04-19:
#### Description
When I send a NON GET with the gcoap example the `coap` thread overflows if there is no request.

#### Steps to reproduce the issue
1. Flash gcoap example to board (I used `samr21-xpro`, did not check for other types of boards)
2. Check current stack usage:

```
> ps 
2018-04-19 11:26:19,614 - INFO #  ps
2018-04-19 11:26:19,622 - INFO # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2018-04-19 11:26:19,630 - INFO # 	  - | isr_stack            | -        - |   - |    512 (  148) | 0x20000000 | 0x200001b8
2018-04-19 11:26:19,638 - INFO # 	  1 | idle                 | pending  Q |  15 |    256 (  156) | 0x200006c0 | 0x20000724 
2018-04-19 11:26:19,646 - INFO # 	  2 | main                 | running  Q |   7 |   1536 ( 1040) | 0x200007c0 | 0x20000c04 
2018-04-19 11:26:19,655 - INFO # 	  3 | 6lo                  | bl rx    _ |   3 |   1024 (  460) | 0x200036a0 | 0x2000396c 
2018-04-19 11:26:19,663 - INFO # 	  4 | ipv6                 | bl rx    _ |   4 |   1024 (  432) | 0x20001764 | 0x20001a2c 
2018-04-19 11:26:19,671 - INFO # 	  5 | udp                  | bl rx    _ |   5 |   1024 (  272) | 0x20003e6c | 0x2000415c 
2018-04-19 11:26:19,680 - INFO # 	  6 | coap                 | bl mbox  _ |   6 |   1024 (  776) | 0x200012c4 | 0x200013bc 
2018-04-19 11:26:19,688 - INFO # 	  7 | at86rf2xx            | bl rx    _ |   2 |   1024 (  504) | 0x20000e84 | 0x20001174 
2018-04-19 11:26:19,694 - INFO # 	    | SUM                  |            |     |   7424 ( 3788)
```

Send CoAP NON to fake destination:

```
> coap get affe::1 5683 /test
2018-04-19 11:32:47,258 - INFO #  coap get affe::1 5683 /test
2018-04-19 11:32:47,262 - INFO # gcoap_cli: sending msg ID 22606, 11 bytes
> 2018-04-19 11:32:52,271 - INFO #  gcoap: timeout for msg ID 22606
```

Check stack usage again:

```
ps
2018-04-19 11:32:55,406 - INFO # ps
2018-04-19 11:32:55,414 - INFO # 	pid | name                 | state    Q | pri | stack  ( used) | base addr  | current     
2018-04-19 11:32:55,422 - INFO # 	  - | isr_stack            | -        - |   - |    512 (  164) | 0x20000000 | 0x200001b8
2018-04-19 11:32:55,430 - INFO # 	  1 | idle                 | pending  Q |  15 |    256 (  156) | 0x200006c0 | 0x20000724 
2018-04-19 11:32:55,439 - INFO # 	  2 | main                 | running  Q |   7 |   1536 ( 1052) | 0x200007c0 | 0x20000c04 
2018-04-19 11:32:55,447 - INFO # 	  3 | 6lo                  | bl rx    _ |   3 |   1024 (  460) | 0x200036a0 | 0x2000396c 
2018-04-19 11:32:55,455 - INFO # 	  4 | ipv6                 | bl rx    _ |   4 |   1024 (  432) | 0x20001764 | 0x20001a2c 
2018-04-19 11:32:55,463 - INFO # 	  5 | udp                  | bl rx    _ |   5 |   1024 (  328) | 0x20003e6c | 0x2000415c 
2018-04-19 11:32:55,472 - INFO # 	  6 | coap                 | bl mbox  _ |   6 |   1024 ( 1024) | 0x200012c4 | 0x200013bc 
2018-04-19 11:32:55,480 - INFO # 	  7 | �D                   | bl rx    _ |   2 |    279 ( -241) | 0x20000e84 | 0x20001174 
2018-04-19 11:32:55,486 - INFO # 	    | SUM                  |            |     |   6679 ( 3375)
```

#### Expected results
Stack should not overflow.

#### Actual results
Stack overflows and overwrites TCB of following thread (might cause crash later on).

#### Versions
This was done on 689333ff04f102e1fd07c74eeda6fdc7f0b574b6 of RIOT.

miri64, 2018-04-19:
(I'm not sure if raising the stack size is the go-to solution here though it is obviously the simplest fix. Rather we should investigate why `gcoap` is using so much stack space in the first place (776 without even sending anything? Not even IPv6 and 6Lo which are quite complex as well are this wasteful).

kb2ma, 2018-04-19:
Thanks for the report @miri64. I have not seen this issue. I am short on time at the moment, but I expect the added size in coap_pkt_t from #8772 has triggered this issue. Note this extra size is specific to gcoap, which still uses the old mechanism for storing options. I am working on adapting gcoap as described in #8831.

kb2ma, 2018-04-23:
I confirmed that the stack issue surfaced with #8772. Prior to this PR, stack size was 724 initially, and 996 (out of 1024 available) after sending a message.

I did test gcoap when I reviewed #8772. I can only guess that I tested mostly on native.

PR #8998 (2018-04-23): net/gcoap: increase stack size by sizeof(coap_pkt_t)
### Contribution description

Apparently the latest nanocoap rework had side effects on gcoap's stack usage, and not of the good kind.

This PR adds ```sizeof(coap_pkt_t)``` to the gcoap stacksize. Meant as quickfix for the release, better solution is on its way.

### Issues/PRs references

Fixes #8981.
PR #9000 (2018-04-23): net/gcoap: increase stack size by sizeof(coap_pkt_t) [BACKPORT]
### Contribution description

(Backport of #8998.)

Apparently the latest nanocoap rework had side effects on gcoap's stack usage, and not of the good kind.

This PR adds sizeof(coap_pkt_t) to the gcoap stacksize. Meant as quickfix for the release, better solution is on its way.
Issues/PRs references

Fixes #8981.
closed 2018-04-23 by kaspar030