RIOT-OS/RIOT #8984: RFC: Set `PSEUDOMODULES` in packages `Makefile.include`

cladmi, 2018-04-19:
#### Description

I was the one suggesting @bergzand to set `PSEUDOMODULES` in his `libcose` package `Makefile.include` while developing so he would not need to modify the other RIOT files.

https://github.com/RIOT-OS/RIOT/blob/6c39d2d621d15cd6f252545c791bbc381b1b48bf/pkg/libcose/Makefile.include#L11-L12

I also thought it could be the occasion to ask if it is something we may want to do for packages as they already provide `Makefile.include` and `Makefile.dep`

As it has been merged, it is even more important to ask the question.

I plan to do a PR to also allow having a `Makefile.include` and `Makefile.dep` for `EXTERNAL_MODULE_DIRS`. So these directories would need to do this also.

#### Question

For packages, do we want to set the `PSEUDOMODULES` variables in the package `Makefile.include` instead of the centralized file ?

I would like to have your feedback and arguments if you are against.

#### Next steps

Depending on the result, either, change the libcose package, or add documentation saying that it's possible and wanted, a comment at the bottom of `pseudomodules.inc.mk` and update packages accordingly.

cladmi, 2018-04-19:
It would also be the way of doing it for external modules with: https://github.com/RIOT-OS/RIOT/pull/8987

cladmi, 2018-04-23:
I added a PR that proposes to add this in the documentation https://github.com/RIOT-OS/RIOT/pull/9003

jia200x, 2018-04-24:
> For packages, do we want to set the PSEUDOMODULES variables in the package Makefile.include instead of the centralized file ?

The idea sounds nice to me!

cladmi, 2018-07-05:
It is now supported and documented.

closed 2018-07-05 by cladmi