RIOT-OS/RIOT #8999: Can't find source code of LoRaWAN

KRolander, 2018-04-23:
Dear jia200x,

I have read that you  already created the implementation of LoRaWAN into RIOT (you presided that it's not a full implementation ), I didn't find the source code of LoRaWAN implementation into RIOT on GitHub (for example the netdev2 repertory doesn't exist in drivers/include/net/), could you help me please where can I find the source code ? Do you have any sample or test programs for trying LoRaWAN on RIOT ? 

Best Regards,
KRolander

miri64, 2018-04-23:
Hi @KRolander,

welcome to the RIOT community. Please use our [developer mailing list](http://doc.riot-os.org/#contribute-to-riot) for this kind of question (or write @jia200x directly). This issue tracker is mainly supposed to track bugs, feature requests and the like.

Thanks and best regards,
Martine

closed 2018-04-23 by miri64

KRolander, 2018-04-23:
Hi,

Could you help me please how can I write to @jia200x directly, because his
mail is unpublished ?

Thanks and Best Regards
KRolander

jia200x, 2018-04-23:
Hi @KRolander 

It seems the private messaging of Github is not available anymore.
Concerning the LoRaWAN implementation, with @fjmolinas we tried to do a custom LoRaWAN implementation based on the Semtech LoRaMAC code. It's available [here](https://github.com/RIOT-OS/RIOT/pull/6645) and working up to that date, but it's outdated to the latest LoRaWAN standard.

It was not merged because we opted to use the LoRaMAC code as a pkg (with [some fixes](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/patches/0001-adapt-for-RIOT.patch)). If you want to try LoRaWAN on RIOT, I strongly suggest you to use this one. There's an example application under tests/pkg_semtech_loramac.

Best regards,
José

KRolander, 2018-04-23:
Hi José,

Thank you so much for your quick answer, I'll try to make the patch, I hope
it will work...
Thank you again!!!

Best Regards,
Roland