RIOT-OS/RIOT #9001: examples/gnrc_minimal: don't work anymore for atmega

kYc0o, 2018-04-23:
#### Description

While doing some tests for the release, I noticed that the gnrc_minimal example
doesn't work anymore in AVR platforms. I tested for the 2017.10 release and
it works. I tried to bisect and fell into the netif2 commits, and git bisect cannot
help there anymore. It is broken since 2018.01.

#### Steps to reproduce the issue

Compile and test gnrc_minimal for any atmega platform. I'm using right now the
xbee and will test with other radios. The only I have currently is the mrf24j40.

#### Expected results

The example prints the address and you can ping it.

#### Actual results

The address is printed but you cannot ping it. I also put a `while (1) {}` which is
needed to avoid the node to kill all processes, as stated in #6526 .

#### Versions

Installed compiler toolchains 
-----------------------------
             native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 7.3.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
   riscv-none-embed-gcc: missing
                  clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.0
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.0
             coccinelle: missing

I have tagged it for the release just to not forget to put it as a known issue.

miri64, 2018-04-23:
Is it possible that the other threads starve because of your `while (1);` loop? Try put a `xtimer_sleep();` in there.

miri64, 2018-04-23:
If not, it would be great if you could provide us with some more information where approximately the packet got lost (i.e. Can you sniff it? Does the receive interrupt fire? Do you have the same problem with `default` or the network device's `test` application?).

ZetaR60, 2018-04-27:
@kYc0o Did you test with #8904 ? The context swap code was changed a few months before the gnrc_minimal example stopped working.

kYc0o, 2018-04-27:
Yes I did, but it's not due to that. I just discovered it doesn't work either on arduino-zero, thus I suspect it comes from gnrc code. I'll investigate deeper.

kYc0o, 2018-05-09:
Small update on this:

After several tests in several platforms to discard problems with GNRC and/or radio drivers, I have the following:

  - The `while (1) {}` is not necessary anymore thanks to #8094 and @ZetaR60 .
  - gnrc_minimal works "more or less" good, though it doesn't allow to be pinged at multicast groups.
  - The XBee radio works properly.
  - The example is still broken for atmega, at least with XBee. However, if I activate `ENABLE_DEBUG    (1)` on gnrc_ipv6.c it works from the second ping after a reset (the first never works), then it behaves as expected:

```console
2018-05-08 18:48:23,004 - INFO #  ping6 fe80::213:a200:40a9:f8d4
2018-05-08 18:48:24,006 - INFO # ping timeout
2018-05-08 18:48:25,018 - INFO # 12 bytes from fe80::213:a200:40a9:f8d4: id=89 seq=1 hop limit=64 time = 0.724 ms
2018-05-08 18:48:26,030 - INFO # 12 bytes from fe80::213:a200:40a9:f8d4: id=89 seq=2 hop limit=64 time = 0.731 ms
2018-05-08 18:48:26,035 - INFO # --- fe80::213:a200:40a9:f8d4 ping statistics ---
2018-05-08 18:48:26,041 - INFO # 3 packets transmitted, 2 received, 34% packet loss, time 3.0618378 s
2018-05-08 18:48:26,044 - INFO # rtt min/avg/max = 0.724/0.485/0.731 ms
> ping6 fe80::213:a200:40a9:f8d4
2018-05-08 18:48:28,683 - INFO #  ping6 fe80::213:a200:40a9:f8d4
2018-05-08 18:48:29,556 - INFO # 12 bytes from fe80::213:a200:40a9:f8d4: id=90 seq=1 hop limit=64 time = 865.371 ms
2018-05-08 18:48:31,435 - INFO # 12 bytes from fe80::213:a200:40a9:f8d4: id=90 seq=2 hop limit=64 time = 864.262 ms
2018-05-08 18:48:33,316 - INFO # 12 bytes from fe80::213:a200:40a9:f8d4: id=90 seq=3 hop limit=64 time = 865.949 ms
2018-05-08 18:48:33,320 - INFO # --- fe80::213:a200:40a9:f8d4 ping statistics ---
2018-05-08 18:48:33,327 - INFO # 3 packets transmitted, 3 received, 0% packet loss, time 4.06618312 s
2018-05-08 18:48:33,331 - INFO # rtt min/avg/max = 864.262/865.194/865.949 ms
```

Thus I suspect something is still wrong in atmega for timers (xtimer config) or the scheduler, to be investigated.

ZetaR60, 2018-05-09:
@kYc0o Can you narrow the problem down by gradually commenting out changes made by ENABLE_DEBUG? I am looking in particular at _event_loop and the msg_receive as a likely candidate (but also note that debug changes the stack size).

kYc0o, 2018-05-09:
>  (but also note that debug changes the stack size).

Maybe that's the key, I didn't think about it. Thanks!

kYc0o, 2018-05-09:
That was it! Stacks are very in the limit for networking applications on AVR. I'll submit a PR to increase them for "big" (more than 2K RAM) platforms. Thanks again @ZetaR60 !

cladmi, 2018-07-10:
@kYc0o This issue is still tagged on milestone 2018.04, please move it to 2018.07 or 2018.10.

kYc0o, 2018-07-10:
Done. I mark it for this release though it depends a lot on the I2C work (to allow some PRs to be merged which fix this problem).

cladmi, 2018-08-11:
This looks related to https://github.com/RIOT-OS/RIOT/issues/9619 so "only" an issue that the avr toolchain/build system currently does not detect the overflow correctly.

kYc0o, 2018-08-11:
> This looks related to #9619 so "only" an issue that the avr toolchain/build system currently does not detect the overflow correctly.

Nope, this example builds and flashes correctly on arduino-mega2560 with XBee for instance. The issue is already detected and it needs just a part of #9130 to be fixed.

ZetaR60, 2018-11-04:
@kYc0o Does the memory increase from #9866 fix this issue?

kYc0o, 2018-11-04:
Yes, it is solved now, closing.

PR #9130 (2019-02-08): atmega: refactor cpu/board code and build/flash variables

### Contribution description
This PR proposes a refactor of the atmega based boards to match the code reuse done in other architectures. In summary:

  - Added atmega.inc.mk for toolchain variables
  - Added avrdude.inc.mk for flashing variables
  - Move cpu.c and startup.c to cpu/atmega_common
  - Share cpu_conf.h to unify the stack memory definitions
  - Cleanup all unnecessary code

I hope I factored out as much as possible, if you find something else that can be reused please let me know.

### Issues/PRs references

Fixes #9001
closed 2018-11-04 by kYc0o