RIOT-OS/RIOT #9036: stm32_common: can't link on IoT-LAB frontend

miri64, 2018-04-26:
#### Description
It is not possible to link a RIOT application for `stm32_common` boards (this includes `iotlab-m3` and `iotlab-a8-m3`) on the SSH frontend server for the IoT-LAB testbed in Grenoble.

#### Steps to reproduce the issue
Go to the SSH frontend server

```bash
ssh <username>@grenoble.iot-lab.info
```

Clone RIOT repository, if necessary. Change into the RIOT repository and try to build an application, e.g. `hello-world`

```bash
make -C examples/hello-world
```

#### Expected results
The application should link.

#### Actual results
Link target fails:

```
stm32_common.ld:25: nonconstant expression for length
collect2: error: ld returned 1 exit status
/senslab/users/lenders/A8/riot/RIOT/examples/hello-world/../../Makefile.include:331: recipe for target 'link' failed
make: *** [link] Error 1
```

#### Versions
On the frontend server a Debian Jessie is running:

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.9 (jessie)
Release:	8.9
Codename:	jessie
$ dist/tools/ci/print_toolchain_versions.sh 
Installed compiler toolchains 
-----------------------------
           native gcc: gcc (Debian 4.9.2-10) 4.9.2
    arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]
              avr-gcc: missing
     mips-mti-elf-gcc: missing
           msp430-gcc: msp430-gcc (GCC) 3.2.3
                clang: missing

Installed compiler libs
-----------------------
 arm-none-eabi-newlib: "2.2.0"
  mips-mti-elf-newlib: missing
             avr-libc: missing (missing)

Installed development tools
---------------------------
                cmake: cmake version 3.0.2
             cppcheck: missing
              doxygen: 1.8.8
               flake8: missing
                  git: git version 2.1.4
           coccinelle: missing
```

aabadie, 2018-04-26:
I guess you are using the master version of RIOT so I think this problem is not related to RIOT but to the compiler version available on iotlab frontend (4.9).

We recently deployed unofficial gcc 7.2 toolchains on the frontends, they are installed in `/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/`, so using:
```
export PATH=/opt/gcc-arm-none-eabi-7-2017-q4-major/bin/:$PATH
```
fix build issues with recent versions of RIOT.

I just tested it and it works.

gebart, 2018-04-26:
We should add the binutils version to the print versions script.

miri64, 2018-04-26:
@aabadie thanks! But this information should be somewhere more prominent I think.

aabadie, 2018-04-26:
> But this information should be somewhere more prominent I think.

This is something relatively new (3 weeks) and we plan to announce it (and update riot tutorials, etc).

aabadie, 2018-04-26:
And we chose to keep the old toolchain as default to avoid unexpected problems with other OSes supported by iotlab (Contiki, OpenWSN, FreeRTOS, TinyOS).

miri64, 2018-04-26:
Ok, then this issue is resolved for me.

closed 2018-04-26 by miri64

aabadie, 2018-04-27:
I edited a [wiki page](https://github.com/iot-lab/iot-lab/wiki/Versions-of-the-GNU-toolchain-for-ARM) related to GNU Toolchains for ARM installed on the IoTLAB frontends.