RIOT-OS/RIOT #9049: xtimer mis-scaling with long sleep times

ZetaR60, 2018-04-29:
#### Description
On the ATmega platforms, xtimer is not scaling properly with XTIMER_HZ when sleeping for longer time periods. It is properly scaling for shorter sleep times. I don't know if this occurs on other platforms (as I have no non-ATmega boards). I discovered this while trying to track down problems with tests/posix_semaphore for #8904

#### Steps to reproduce the issue
Build and run something that uses longer sleep times, such as tests/xtimer_drift. Change the XTIMER_HZ define in board.h, then build and run again. Also try something with shorter sleep times, such as tests/xtimer_usleep.

#### Expected results
Both short and long sleep times (tests/xtimer_drift and tests/xtimer_usleep) vary only slightly with different XTIMER_HZ defines.

#### Actual results
Shorter sleep times (tests/xtimer_usleep) vary slightly, and longer sleep times (tests/xtimer_drift) vary wildly.

#### Versions
```
Installed compiler toolchains 
-----------------------------
             native gcc: missing
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (Fedora 7.2.0-1.fc27) 7.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: missing
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.0
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.14.2
             coccinelle: missing
```

kYc0o, 2018-05-18:
> Change the XTIMER_HZ define in board.h

Why would you do that? XTIMER_HZ is meant to be a "safe" timer initialisation, if you put "random" values there, you must be sure your hardware supports it.

What's the fastest timer speed in your atmega platform?

gebart, 2018-05-18:
What exact XTIMER_HZ values did you try? Does the atmega periph timer driver support the frequencies you tried?

ZetaR60, 2018-05-18:
The timers on ATmega support CLK/1, CLK/8, CLK/64, CLK/256, CLK/1024. I tried using 8MHz XTIMER_HZ, 1MHz XTIMER_HZ, and 125kHz XTIMER_HZ. All of these are supported frequencies by the underlying hardware, and the changes should all be easy for xtimer use (being powers of 2). The point is that xtimer does actually work properly on these frequencies, but only for short sleep times. Changing XTIMER_HZ reveals that xtimer is not scaling properly when using longer sleep times.

kYc0o, 2018-05-18:
Have you tried those speeds with tests/periph_timer?

gebart, 2018-05-18:
Try https://github.com/RIOT-OS/RIOT/pull/8531 for evaluating the timer driver. You can test both periph timer and xtimer with it. Try the default build settings first, then play around with TIM_TEST_FREQ to test different frequencies. If it hangs, you might need to increase TEST_MIN:
```
CFLAGS='-DTEST_MIN=10 -DTIM_TEST_FREQ=125000' make clean all flash
```
To test xtimer:
```
make clean test-xtimer flash
```
(That might also need tweaking the TEST_MIN parameter) 
Remember to make clean between parameter changes, I am not sure if the build system picks up on every change without cleaning

ZetaR60, 2018-05-23:
@gebart Here are (some) of the results. test-xtimer always seemed to hang, even when I set TEST_MIN very high. Also, I tried 8MHz test frequency, but it didn't like it and gave me a compile error.

Default settings:
```
[liveuser@localhost-live bench_timers]$ make BOARD=mega-xplained term
/home/liveuser/Desktop/RIOT-pr-periph-timer-stats/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "9600"
No handlers could be found for logger "root"
2018-05-22 20:13:26,871 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2018-05-22 20:13:27,876 - INFO # 

2018-05-22 20:13:28,942 - INFO # random: NO SEED AVAILABLE!
2018-05-22 20:13:29,043 - INFO # main(): This is RIOT! (Version: UNKNOWN (builddir: /home/liveuser/Desktop/RIOT-pr-periph-timer-stats))
2018-05-22 20:13:29,044 - INFO # 
2018-05-22 20:13:29,074 - INFO # Statistical benchmark for timers
2018-05-22 20:13:29,141 - INFO # Running timer test with seed 123 using Tiny Mersenne Twister PRNG.
2018-05-22 20:13:29,142 - INFO # TEST_MIN = 16
2018-05-22 20:13:29,174 - INFO # TEST_MAX = 128
2018-05-22 20:13:29,175 - INFO # TEST_MIN_REL = 0
2018-05-22 20:13:29,240 - INFO # TEST_MAX_REL = 112
2018-05-22 20:13:29,242 - INFO # TEST_NUM = 113
2018-05-22 20:13:29,244 - INFO # log2(TEST_NUM - 1) = 6
2018-05-22 20:13:29,275 - INFO # state vector elements per variant = 7
2018-05-22 20:13:29,340 - INFO # number of variants = 8
2018-05-22 20:13:29,343 - INFO # sizeof(state) = 32 bytes
2018-05-22 20:13:29,374 - INFO # state vector total memory usage = 1792 bytes
2018-05-22 20:13:29,444 - INFO # TIM_TEST_DEV = 0, TIM_TEST_FREQ = 1000000, TIM_TEST_CHAN = 0
2018-05-22 20:13:29,474 - INFO # TIM_REF_DEV  = 1, TIM_REF_FREQ  = 1000000
2018-05-22 20:13:29,542 - INFO # USE_REFERENCE = 1
2018-05-22 20:13:29,545 - INFO # TEST_PRINT_INTERVAL_TICKS = 30000000
2018-05-22 20:13:29,643 - INFO # Expected error variance due to truncation in tick conversion: 0cb: Warning! ref_state = NULL
2018-05-22 20:13:29,643 - INFO # 
2018-05-22 20:13:29,674 - INFO # Calibrating spin delay...
2018-05-22 20:13:29,676 - INFO # spin_max = 20
2018-05-22 20:13:29,774 - INFO # Estimating benchmark overhead...
2018-05-22 20:13:30,540 - INFO # overhead_target = 47 (s2 = 0)
2018-05-22 20:13:30,543 - INFO # overhead_read = 48 (s2 = 0)
2018-05-22 20:13:30,545 - INFO # exp_mask = 00000007
2018-05-22 20:13:30,575 - INFO # max interval = 255
2018-05-22 20:18:17,177 - INFO # Elapsed time:
2018-05-22 20:18:17,179 - INFO #         Reference: 4294
2018-05-22 20:18:17,243 - INFO #  Timer under test: 4294
2018-05-22 20:18:17,245 - INFO # ------------- BEGIN STATISTICS --------------
2018-05-22 20:18:17,277 - INFO # ===== Reference timer statistics =====
2018-05-22 20:18:17,345 - INFO # Limits: mean: [-10, 10], variance: [0, 16]
2018-05-22 20:18:17,443 - INFO # Target error (actual trigger time - expected trigger time), in reference timer ticks
2018-05-22 20:18:17,543 - INFO # positive: timer under test is late, negative: timer under test is early
2018-05-22 20:18:17,546 - INFO # === timer_set running ===
2018-05-22 20:18:17,643 - INFO #    interval     count       sum       sum_sq    min   max  mean  variance
2018-05-22 20:18:17,679 - INFO #    0 -    1:      493     -6830          419    -14   -13   -13      0  <=== SIC!
2018-05-22 20:18:17,777 - INFO #    2 -    3:      499     -6922          422    -14   -13   -13      0  <=== SIC!
2018-05-22 20:18:17,876 - INFO #    4 -    7:      494     -6509         1022    -14   -10   -13      2  <=== SIC!
2018-05-22 20:18:17,945 - INFO #    8 -   15:      451     -5934          514    -14   -11   -13      1  <=== SIC!
2018-05-22 20:18:18,044 - INFO #   16 -   31:      483      1807        11033     -4    12     3     22  <=== SIC!
2018-05-22 20:18:18,144 - INFO #   32 -   63:      421      5451       165106    -14    38    12    393  <=== SIC!
2018-05-22 20:18:18,243 - INFO #   64 -  127:      377      -904        48633    -14    19    -2    129  <=== SIC!
2018-05-22 20:18:18,342 - INFO #       TOTAL      3218    -19841       521279    -14    38    -6    162  <=== SIC!
2018-05-22 20:18:18,344 - INFO # === timer_set resched ===
2018-05-22 20:18:18,443 - INFO #    interval     count       sum       sum_sq    min   max  mean  variance
2018-05-22 20:18:18,477 - INFO #    0 -    1:      493     -6426       151511    -14   375   -13    307  <=== SIC!
2018-05-22 20:18:18,577 - INFO #    2 -    3:      488     -6775          429    -14   -13   -13      0  <=== SIC!
2018-05-22 20:18:18,644 - INFO #    4 -    7:      516     -7169          453    -14   -13   -13      0  <=== SIC!
2018-05-22 20:18:18,744 - INFO #    8 -   15:      478     -6628          408    -14   -13   -13      0  <=== SIC!
2018-05-22 20:18:18,845 - INFO #   16 -   31:      483      1817         9909     -4    12     3     20  <=== SIC!
2018-05-22 20:18:18,944 - INFO #   32 -   63:      470      6013       188300    -14    38    12    401  <=== SIC!
2018-05-22 20:18:19,044 - INFO #   64 -  127:      346     -1109        47185    -14    19    -3    136  <=== SIC!
2018-05-22 20:18:19,077 - INFO #       TOTAL      3274    -20277       708441    -14   375    -6    216  <=== SIC!
2018-05-22 20:18:19,144 - INFO # === timer_set stopped ===
2018-05-22 20:18:19,242 - INFO #    interval     count       sum       sum_sq    min   max  mean  variance
[... 354 lines cut ...]
2018-05-22 20:45:35,265 - INFO #   42 -   73:      453      2669        35346    -10    22     5     78  <=== SIC!
[... cut ...]
2018-05-22 20:45:36,565 - INFO #            resched     3248    -61227       640839    -31    21   -18    197  <=== SIC!
2018-05-22 20:45:36,665 - INFO #            stopped     3307    -47620       566493    -31    21   -14    171  <=== SIC!
2018-05-22 20:45:36,798 - INFO #   stopped, resched     3263    -47722       552092    -31    21   -14    169  <=== SIC!
2018-05-22 20:45:36,801 - INFO # -------------- END STATISTICS ---------------
/exit
2018-05-22 20:45:40,325 - INFO # Exiting Pyterm
[liveuser@localhost-live bench_timers]$
```

Josar, 2018-10-24:
@ZetaR60 

> longer sleep times (tests/xtimer_drift) vary wildly.

how did you inspect the varying times? did you rely on the values printed out by the Test?
Did you measure them with pin toggles?

Please see https://github.com/RIOT-OS/RIOT/pull/9625 .

The values printed out in xtimer_drift where not adjusted to the achieveable precision of XTIMER_HZ.

miri64, 2020-07-02:
Still an issue after https://github.com/RIOT-OS/RIOT/pull/9530? @pokgak @MichelRottleuthner @MrKevinWeiss

maribu, 2022-09-19:
`xtimer` is on its way out and no longer in used, as the xtimer APIs are provided by default as wrappers on top of `ztimer`. I don't expect any bugs in `xtimer` to be fixed anymore.

closed 2022-09-19 by maribu