RIOT-OS/RIOT #9052: misc issues with tests/trickle

ZetaR60, 2018-04-30:
#### Description
While running tests/trickle for #8904 on mega-xplained, I have run into a number of different issues. One issue is that prev_now is not being initialized properly in main (it assumes xtimer_now_usec is zero when main starts, which is not the case). This is easily fixed by adding ```prev_now = xtimer_now_usec();``` to the beginning of main. There are also a number of problems where it hangs at various points.

#### Steps to reproduce the issue
Run tests/trickle with the following variations:
- trunk on mega-xplained
- #8904 applied to trunk on mega-xplained
- trunk on mega-xplained with ```prev_now = xtimer_now_usec();```
- #8904 applied to trunk on mega-xplained with ```prev_now = xtimer_now_usec();```

#### Expected results
tests/trickle runs, printing 5 messages after ```[START]```, and then 7 messages after ```[TRICKLE_RESET]```, and then prints ```[SUCCESS]```.

#### Actual results
- Trunk hangs after ```[TRICKLE_RESET]```
- #8904 applied to trunk declares failure (due to not initializing prev_now properly)
- Trunk with prev_now initialized hangs after three messages (I think this is a #8896 related problem, and is fixed by #8904).
- #8904 applied to trunk and prev_now initialized hangs after ```[TRICKLE_RESET]```

#### Versions
Board: mega-xplained
```
Installed compiler toolchains 
-----------------------------
             native gcc: missing
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (Fedora 7.2.0-1.fc27) 7.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: missing
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.0
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.14.2
             coccinelle: missing
```

cladmi, 2018-05-11:
Thank for investigating this, I also had problem on `tests/trickle` with `arduino-zero`.
Adding the `prev_now = xtimer_now_usec();` in main fixes it. I will do a PR for it.

cladmi, 2018-05-11:
I implemented your `prev_now` fix in #9118.

However I misread and thought it also fixed your problem, I can amend my commit message depending on your feedback.

cladmi, 2018-05-15:
@ZetaR60 I am not seeing any hang on `arduino-zero` in the `trickle` test, with the initialization fix, tests are working.
However I am not really using the board for anything, just running the automated tests on IoT-LAB.

ZetaR60, 2018-05-15:
@cladmi Below is what I get on mega-xplained. Could you post the output you are getting as well?

```
[liveuser@localhost-live trickle]$ make BOARD=mega-xplained test
tests/01-run.py
/home/liveuser/Desktop/RIOT_master/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "9600"
No handlers could be found for logger "root"
2018-05-15 14:33:52,852 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2018-05-15 14:33:53,856 - INFO # 

2018-05-15 14:33:53,858 - INFO # random: NO SEED AVAILABLE!
2018-05-15 14:33:53,864 - INFO # main(): This is RIOT! (Version: 2018.07-devel-202-gd6c63-localhost-live)
2018-05-15 14:33:53,864 - INFO # [START]
2018-05-15 14:33:53,868 - INFO # now = 122792, prev_now = 75272, diff = 47520
2018-05-15 14:33:54,054 - INFO # now = 259928, prev_now = 122792, diff = 137136
2018-05-15 14:33:54,156 - INFO # now = 423144, prev_now = 259928, diff = 163216
2018-05-15 14:33:54,655 - INFO # now = 910360, prev_now = 423144, diff = 487216
2018-05-15 14:33:55,454 - INFO # now = 1640576, prev_now = 910360, diff = 730216
2018-05-15 14:33:55,456 - INFO # [TRICKLE_RESET]
Timeout in expect script at "child.expect(u"now = \d+, prev_now = \d+, diff = \d+")" (tests/trickle/tests/01-run.py:22)

make: *** [Makefile:10: test] Error 1
[liveuser@localhost-live trickle]$
```

```
Installed compiler toolchains 
-----------------------------
             native gcc: missing
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (Fedora 7.2.0-1.fc27) 7.2.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: missing
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.0
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.14.2
             coccinelle: missing
```

cladmi, 2018-05-16:
I am running from my branch to run on IoT-LAB:

```
IOTLAB_NODE=auto-ssh make test BOARD=arduino-zero
tests/01-run.py
ssh -t harter@saclay.iot-lab.info 'socat - tcp:arduino-zero-1.saclay.iot-lab.info:20000'
main(): This is RIOT! (Version: 2018.07-devel-237-g2e252-inetm02-dev/rebase)
[START]
now = 91529, prev_now = 72464, diff = 19065
now = 129538, prev_now = 91529, diff = 38009
now = 204642, prev_now = 129538, diff = 75104
now = 378837, prev_now = 204642, diff = 174195
now = 624123, prev_now = 378837, diff = 245286
[TRICKLE_RESET]
now = 639444, prev_now = 624123, diff = 15321
now = 667641, prev_now = 639444, diff = 28197
now = 735840, prev_now = 667641, diff = 68199
now = 819041, prev_now = 735840, diff = 83201
now = 1129242, prev_now = 819041, diff = 310201
now = 1509621, prev_now = 1129242, diff = 380379
now = 2494089, prev_now = 1509621, diff = 984468
[SUCCESS]
```
Toolchain
```
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
                  clang: clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: "2.1.0"
riscv-none-embed-newlib: missing
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.5.1
               cppcheck: Cppcheck 1.72
                doxygen: 1.8.11
                 flake8: 3.5.0 (mccabe: 0.6.1, pycodestyle: 2.3.1, pyflakes: 1.6.0) CPython 3.5.2 on Linux
                    git: git version 2.7.4
             coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

bergzand, 2018-05-16:
The test correctly runs if I increase `TR_IMIN` to 40 (`XTIMER_BACKOFF` for the arduino).

bergzand, 2018-05-16:
Not that those have anything to do with each other -_-

cladmi, 2018-08-13:
Note from the release tests, it is also broken on `stm32discovery` where the test detects the failure.

* arduino-mega2560: stuck after TRICKLE_RESET
* arduino-uno: stuck after TRICKLE_RESET
* stm32discovery: test failure detected by firmware