RIOT-OS/RIOT #9060: Unittests broken on iotlab-m3 iotlab-a8-m3

cladmi, 2018-05-02:
#### Description

`tests/unittests` fails on `iotlab-m3` and `iotlab-a8-m3` by a hardfault in `qDSA` test.
I tried debugging it with printfs and adding prints changes where it breaks… I do not have really ideas on why it breaks…

I even forced LARGE_STACK_TESTS and increased stack size.

#### Steps to reproduce the issue

    make -C tests/unittests BOARD=iotlab-m3 flash test

It fails after some time to a hardfault.

#### Expected results

Test OK

#### Actual results

```
CFLAGS=-DDEBUG_ASSERT_VERBOSE make BOARD=iotlab-m3 tests-qDSA DEVELHELP=1 clean   flash
main(): This is RIOT! (Version: 2018.07-devel-63-gc6ae40-inetm02-dev/rebase)
.
Context before hardfault:
   r0: 0x20002598
   r1: 0xfc6f4203
   r2: 0x20002588
   r3: 0x20002520
  r12: 0x20002480
   lr: 0x08002af5
   pc: 0x0800528c
  psr: 0x01000200

FSR/FAR:
 CFSR: 0x01000000
 HFSR: 0x40000000
 DFSR: 0x00000000
 AFSR: 0x00000000
Misc
EXC_RET: 0xfffffffd
Attempting to reconstruct state for debugging...
In GDB:
  set $pc=0x800528c
  frame 0
  bt

ISR stack overflowed by at least 16 bytes.

```
#### Versions

aabadie, 2018-05-02:
Have you tried using the following in GDB ?
```
set $pc=0x800528c
frame 0
bt
```

cladmi, 2018-05-02:
```

(gdb) set $pc=0x800528c
(gdb) frame 0
#0  bigint_mul ()
    at /home/harter/work/git/RIOT/tests/unittests/bin/pkg/iotlab-m3/qDSA/arm/asm/bigint_mul.S:18
18              ldm r1!, {r2,r3,r6,r7}
(gdb) bt
#0  bigint_mul ()
    at /home/harter/work/git/RIOT/tests/unittests/bin/pkg/iotlab-m3/qDSA/arm/asm/bigint_mul.S:18
#1  0x01010200 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

aabadie, 2018-05-02:
Not very useful...

cladmi, 2018-05-07:
Fixed by https://github.com/RIOT-OS/RIOT/pull/9091

closed 2018-05-07 by cladmi