RIOT-OS/RIOT #9064: examples: unable to build ccn-lite for cortex-m

haukepetersen, 2018-05-03:
#### Description
Just tried to build the `cc-lite-relay` example for ARM Cortex-M (`iotlab-m3` in my case). The build fails due to some libc issues as it seems to me:

```
hauke@Prag ~/dev/riot/RIOT/examples/ccn-lite-relay $ BOARD=iotlab-m3 make all
Building application "ccn-lite-relay" for "iotlab-m3" with MCU "stm32f1".

make[1]: Nothing to be done for 'prepare'.
make[1]: Nothing to be done for 'prepare'.
"make" -C /home/hauke/dev/riot/RIOT/pkg/ccn-lite
/home/hauke/dev/riot/RIOT/dist/tools/cmake/generate-xcompile-toolchain.sh > /home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/xcompile-toolchain.cmake
cd /home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src && \
cmake -DCMAKE_TOOLCHAIN_FILE=/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/xcompile-toolchain.cmake \
	  -DCCNL_RIOT=1 -DRIOT_CFLAGS="-Wall -Werror -Wextra -Wno-implicit-fallthrough -mno-thumb-interwork -mcpu=cortex-m3 -mlittle-endian -mthumb -mfloat-abi=soft -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -ggdb -g3 -Os -std=c99 -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -Werror=strict-prototypes -Werror=old-style-definition -fno-common -Wall -include '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/iotlab-m3/riotbuild/riotbuild.h'  -isystem /newlib-nano -I/home/hauke/dev/riot/RIOT/core/include -I/home/hauke/dev/riot/RIOT/drivers/include -I/home/hauke/dev/riot/RIOT/sys/include -I/home/hauke/dev/riot/RIOT/cpu/stm32f1/include -I/home/hauke/dev/riot/RIOT/boards/iotlab-m3/include -I/home/hauke/dev/riot/RIOT/boards/common/iotlab/include -I/home/hauke/dev/riot/RIOT/boards/iotlab-m3/include -I/home/hauke/dev/riot/RIOT/cpu/stm32_common/include -I/home/hauke/dev/riot/RIOT/cpu/cortexm_common/include -I/home/hauke/dev/riot/RIOT/cpu/cortexm_common/include/vendor -I/home/hauke/dev/riot/RIOT/sys/libc/include -I/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/ccnl-riot/include -I/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/ccnl-core/include -I/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/ccnl-pkt/include -I/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/ccnl-fwd/include -I/home/hauke/dev/riot/RIOT/sys/posix/include -I/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/tlsf/src -I/home/hauke/dev/riot/RIOT/drivers/at86rf2xx/include" -DBUILD_TESTING=OFF .
-- The CXX compiler identification is GNU 7.2.1
-- Check for working CXX compiler: /home/hauke/tools/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/hauke/tools/bin/arm-none-eabi-g++ -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCXXCompiler.cmake:54 (message):
  The C++ compiler "/home/hauke/tools/bin/arm-none-eabi-g++" is not able to
  compile a simple test program.

  It fails with the following output:

   Change Dir: /home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/bin/make" "cmTC_22036/fast"

  make[2]: Entering directory
  '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp'

  /usr/bin/make -f CMakeFiles/cmTC_22036.dir/build.make
  CMakeFiles/cmTC_22036.dir/build

  make[3]: Entering directory
  '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp'

  Building CXX object CMakeFiles/cmTC_22036.dir/testCXXCompiler.cxx.obj

  /home/hauke/tools/bin/arm-none-eabi-g++ -o
  CMakeFiles/cmTC_22036.dir/testCXXCompiler.cxx.obj -c
  /home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp/testCXXCompiler.cxx

  Linking CXX executable cmTC_22036

  /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_22036.dir/link.txt
  --verbose=1

  /home/hauke/tools/bin/arm-none-eabi-g++
  CMakeFiles/cmTC_22036.dir/testCXXCompiler.cxx.obj -o cmTC_22036

  
  /home/hauke/tools/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o):
  In function `exit':

  exit.c:(.text.exit+0x2c): undefined reference to `_exit'

  collect2: error: ld returned 1 exit status

  CMakeFiles/cmTC_22036.dir/build.make:97: recipe for target 'cmTC_22036'
  failed

  make[3]: *** [cmTC_22036] Error 1

  make[3]: Leaving directory
  '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp'

  Makefile:126: recipe for target 'cmTC_22036/fast' failed

  make[2]: *** [cmTC_22036/fast] Error 2

  make[2]: Leaving directory
  '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeTmp'

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  ccnl-core/CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeOutput.log".
See also "/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/iotlab-m3/ccn-lite/src/CMakeFiles/CMakeError.log".
Makefile:17: recipe for target '/src/Makefile' failed
make[1]: *** [/src/Makefile] Error 1
/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/../../Makefile.include:418: recipe for target '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/iotlab-m3/ccn-lite.a' failed
make: *** [/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/iotlab-m3/ccn-lite.a] Error 2
```

using 
```
arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
```

Is this something known (and I was to blind to find the issue), or did someone encounter this before?

#### Steps to reproduce the issue
build the `ccn-lite-relay` example for `iotlab-m3` using the toolchain as pointed to above.

#### Expected results
should build without errors...

#### Actual results
see above

#### Versions
see above

aabadie, 2018-06-01:
@haukepetersen, I just tried and it works for me.

```
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
```

Can you try again with the latest master ?

cgundogan, 2018-06-18:
@haukepetersen I am also able to build without problems. We had a version bump a couple of days ago. Could you give it another try?

cgundogan, 2018-06-21:
ping @haukepetersen 

Interesting though that your log says:
```
-- The CXX compiler identification is GNU 7.2.1
-- Check for working CXX compiler: /home/hauke/tools/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/hauke/tools/bin/arm-none-eabi-g++ -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCXXCompiler.cmake:54 (message):
  The C++ compiler "/home/hauke/tools/bin/arm-none-eabi-g++" is not able to
  compile a simple test program.
```
No idea why CMake only detects `arm-none-eabi-g++` on your side, and not `arm-none-eabi-gcc`.

cgundogan, 2018-06-21:
@haukepetersen Looking at your CMake log, is your CMake of version 3.5? This problem may occure in older CMake versions. Latest Ubuntus nowadays seem to have version 3.10.*, I have 3.11.4. Can you upgrade your CMake and check again?

cladmi, 2018-07-17:
I have the same issue here when testing the release. I installed `cmake 3.10.3` in `/usr/local` and it worked. I can try debugging this tomorrow.

cgundogan, 2018-07-24:
>I can try debugging this tomorrow.

what are your debugging results? If this issue is fixed with a recent `cmake` version that is included in current Ubuntu LTS versions, then I won't pursue this issue further.

miri64, 2018-08-14:
The issue is made more obvious, but not fixed IMHO.

haukepetersen, 2018-08-23:
ok, just checked again on my desktop at home (which is newly installed with Mint 19, cmake version 3.10.2, arm-none-eabi-gcc version 7.3.1):
- building for `iotlab-m3` works fine now
- but building for `native` fails:
```
"make" -C /home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/native/tlsf
/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/native/tlsf/tlsf.c: In function ‘default_walker’:
/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/pkg/native/tlsf/tlsf.c:901:29: error: format ‘%p’ expects argument of type ‘void *’, but argument 5 has type ‘block_header_t * {aka struct block_header_t *}’ [-Werror=format=]
  printf("\t%p %s size: %x (%p)\n", ptr, used ? "used" : "free", (unsigned int)size, block_from_ptr(ptr));
                            ~^                                                       ~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
/home/hauke/dev/riot/RIOT/Makefile.base:83: recipe for target '/home/hauke/dev/riot/RIOT/examples/ccn-lite-relay/bin/native/tlsf/tlsf.o' failed
```
Does this look familiar?

building with gcc 7.3.0

haukepetersen, 2018-08-23:
ok, no idea what just happened, but now I can build it just fine for native as well.

So for now, I consider this issue solved.

PR #9720 (2018-08-14): pkg/ccn-lite/relic: check for minimal cmake version
### Contribution description

ccn-lite/relic do not build with cmake < 3.6.0. This checks for a minimal version
instead of failing to compile.

    cmake version 3.5.2 is not >= to minimal required 3.6.0

    Makefile:25: recipe for target '..cmake_version_supported' failed

Tested with versions 3.5.2, 3.6.0-rc1, 3.6.0, 3.6.3, 3.7.0-rc1, 3.7.0.

Note: the check used does not consider '-rcX' as 'sort -V' does not handle them
properly.

### Testing

You can test this by compiling `examples/ccn-lite-relay` with different cmake versions:

Last 3. [... cut]
closed 2018-08-23 by haukepetersen