RIOT-OS/RIOT #9106: fatal error: stdio.h: No such file or directory - msp430

PedroAlfNeto, 2018-05-09:
#### Description

Cannot build example/default or any other application because of the error:
`/home/pedorcas/Documents/Code/RIOT_OS/RIOT/cpu/msp430_common/include/stdio.h:28:24: fatal error: stdio.h: No such file or directory`

#### Steps to reproduce the issue

Using the command:
`make all BOARD=telosb`

#### Expected results

To build the application x)

#### Actual results

Cannot build example/default or any other application because of the error:
/home/pedorcas/Documents/Code/RIOT_OS/RIOT/cpu/msp430_common/include/stdio.h:28:24: fatal error: stdio.h: No such file or directory

But the file is there with all the permissions, as you can see in the response of the cmd:
`cd /home/pedorcas/Documents/Code/RIOT_OS/RIOT/cpu/msp430_common/include/ && ls -l
`
output:
`(...)
-rw-rw-r-- 1 pedorcas pedorcas   915 May  4 10:13 stdio.h
(...)`

#### Versions

Linux Mint 18

Thank you for your time and effort.
Really appreciate the tool you provided (RIOT OS), and wish you all a fortune!.

rfuentess, 2018-05-09:
Hi and welcome to RIOT! 

Can you please run `./dist/tools/ci/print_toolchain_versions.sh` and copy/paste the output here ?
This will help us to identify the issue :)

Also, if you still don't read the [toolchains wiki](https://github.com/RIOT-OS/RIOT/wiki/Introduction#compiling-riot) page, this is a good moment

Finally, try to compile  `examples/hello-world` as I think the gnrc_*  examples are too big for the telosb boards.

kYc0o, 2018-05-09:
That error looks like a bad configured/installed toolchain. I recommend you to follow [this](https://github.com/contiki-os/contiki/wiki/MSP430X) guide and then try again.

rfuentess, 2018-05-09:
> > Finally, try to compile examples/hello-world as I think the **gnrc_* examples are too big** for the telosb boards.
>
>  I suspected it was too big, but i was trying to simulate builds on cooja, and didn't know a board that could simulate udp packages, or if i could/how use the native build.

A correction from my part,  `examples/gnrc_minimal` should be OK for telosb.

Also, native can send UDP packages (though will use standard IPv6 instead of 6LoWPAN).

PedroAlfNeto, 2018-05-09:
@rfuentess `examples/gnrc_minimal` gives the same error.
also, through the toolchains wiki's link i wasn't able to figure install the msp430 toolchain so i previously installed 
`sudo apt-get install gcc-msp430`.

PedroAlfNeto, 2018-05-09:
@kYc0o I'm giving up on the `mspgcc 4.7.0` script, after creating the folder running it, took about 20 min and failed in the end with various:
`as: unrecognized option '-mcpu=430'`
Thank you still.

PedroAlfNeto, 2018-05-10:
To update you and conclude this thread,

>  i ended up using VM RiotOS that comes with the toolchains pre installed.

As of the instructions, the became obsolete, most links are broken, this because the abandonment of the openSource gcc-msp430, so it is needed to download the compiler from the Texas instruments site. And that version i think it's not supported by the current Riot (suspect just needs a path update), or maybe just an update to the instructions will solve it :smile:. Thanks for the support.

closed 2018-05-14 by PedroAlfNeto