RIOT-OS/RIOT #9148: Hard fault after a while, 2 UART

ClementChotard, 2018-05-17:
Hi,

I have a problem when using 2 USART on STM32f407. I receive some data from both USART which triggers each its callback function. I tested two possibilities in the cb, either it's sending each received character by IPC to a Thread or it's saving the full message in an array before sending it to the Thread. Either way, a hard fault error triggers after a while. When sending each character individually, it's lasting longer before crashing. Here is the trace:
```
2018-05-17 15:15:06,852 - INFO # 
2018-05-17 15:15:06,852 - INFO # Context before hardfault:
2018-05-17 15:15:06,852 - INFO #    r0: 0x22d8ca84
2018-05-17 15:15:06,852 - INFO #    r1: 0x200016bc
2018-05-17 15:15:06,867 - INFO #    r2: 0x20000820
2018-05-17 15:15:06,867 - INFO #    r3: 0x0002c2c0
2018-05-17 15:15:06,867 - INFO #   r12: 0x200015b3
2018-05-17 15:15:06,867 - INFO #    lr: 0x08000247
2018-05-17 15:15:06,868 - INFO #    pc: 0x0800025c
2018-05-17 15:15:06,868 - INFO #   psr: 0x0100000e
2018-05-17 15:15:06,868 - INFO # 
2018-05-17 15:15:06,868 - INFO # FSR/FAR:
2018-05-17 15:15:06,868 - INFO #  CFSR: 0x00008200
2018-05-17 15:15:06,868 - INFO #  HFSR: 0x40000000
2018-05-17 15:15:06,869 - INFO #  DFSR: 0x00000008
2018-05-17 15:15:06,869 - INFO #  AFSR: 0x00000000
2018-05-17 15:15:06,883 - INFO #  BFAR: 0x2002c8d0
2018-05-17 15:15:06,883 - INFO # Misc
2018-05-17 15:15:06,883 - INFO # EXC_RET: 0xfffffff1
2018-05-17 15:15:06,884 - INFO # Attempting to reconstruct state for debugging...
2018-05-17 15:15:06,884 - INFO # In GDB:
2018-05-17 15:15:06,884 - INFO #   set $pc=0x800025c
2018-05-17 15:15:06,884 - INFO #   frame 0
2018-05-17 15:15:06,884 - INFO #   bt
```
And here is the debug result:
```
Remote debugging using :3333
hard_fault_handler (sp=0x2002c8d0, corrupted=536876580, exc_return=11, r4_to_r11_stack=0xa)
    at /home/clementc/Projects/Stage-2018/RIOT/cpu/cortexm_common/vectors_cortexm.c:330
330	    __BKPT(1);
(gdb) set $pc=0x800025c
(gdb) frame 0
#0  0x0800025c in sched_run () at /home/clementc/Projects/Stage-2018/RIOT/core/sched.c:120
120	        if (active_stat->laststart) {
(gdb) bt
#0  0x0800025c in sched_run () at /home/clementc/Projects/Stage-2018/RIOT/core/sched.c:120
#1  0x0800047c in isr_svc () at /home/clementc/Projects/Stage-2018/RIOT/cpu/cortexm_common/thread_arch.c:334
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

A similar issue was discussed here: #5001

Thank you for your help

smlng, 2018-05-28:
Hi @ClementChotard can you try compiling with `DEVELHELP=0 make` likely a timing issue with ISR and (too much) debug code

ClementChotard, 2018-05-28:
Hi @smlng thanks for replying. I started to test and it seems to work now! I will run more test and keep you updated.

ClementChotard, 2018-05-29:
Unfortunately, the problem remains the same... There is about 10 minutes before it crashes. I also increased the msgQ size that sends messages from the cb to the Thread.

kYc0o, 2018-05-29:
@ClementChotard can you also try to print a `ps();` from time to time? Or at the end of each of your tasks? Maybe you're running out of memory or have a stack overflow.

ClementChotard, 2018-05-29:
I figured out, the problem comes from too much print as suggested by @smlng. I was printing every message that it received and eventually it ends up by crashing the system. Thanks, problem solved.

closed 2018-05-29 by ClementChotard