RIOT-OS/RIOT #9149: examples: cannot build for nucleo-f091 / nucleo-f411

enzingerm, 2018-05-17:
#### Description

Cannot build hello_world and gnrc_networking (others untested) for nucleo boards

#### Steps to reproduce the issue
Checkout repo from github, `cd` into `examples/hello-world` and run `make BOARD=nucleo-f091`

#### Expected results

It should work

#### Actual results
```
/home/marinus/dev/riot/riot-upstream/sys/uart_stdio/uart_stdio.c: In function 'uart_stdio_init':
/home/marinus/dev/riot/riot-upstream/sys/uart_stdio/uart_stdio.c:89:52: error: cast between incompatible function types from 'int (*)(isrpipe_t *, char)' {aka 'int (*)(struct <anonymous> *, char)'} to 'void (*)(void *, uint8_t)' {aka 'void (*)(void *, unsigned char)'} [-Werror=cast-function-type]
     uart_init(UART_STDIO_DEV, UART_STDIO_BAUDRATE, (uart_rx_cb_t) isrpipe_write_one, &uart_stdio_isrpipe);
```

#### Versions
Installed compiler toolchains 
- native gcc: gcc (GCC) 8.1.0
- arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.1.0
- avr-gcc: avr-gcc (GCC) 8.1.0
- mips-mti-elf-gcc: missing
- msp430-gcc: missing
- riscv-none-embed-gcc: missing
- clang: missing

Installed compiler libs
- arm-none-eabi-newlib: "3.0.0"
- mips-mti-elf-newlib: missing
- riscv-none-embed-newlib: missing
- avr-libc: "2.0.0" ("20150208")

Installed development tools
- cmake: cmake version 3.11.1
- cppcheck: missing
- doxygen: missing
- flake8: missing
- git: git version 2.17.0
- coccinelle: missing

gebart, 2018-05-17:
This looks like #9121

cgundogan, 2018-05-17:
@enzingerm could you try again with the latest master?

enzingerm, 2018-05-17:
I can confirm that it works now, thanks.

closed 2018-05-17 by enzingerm