RIOT-OS/RIOT #9155: __libc_init_array () trigger HARD FAULT HANDLER on SAML21E18B arm core bug

nunux13, 2018-05-19:
#### Description
Cannot run helloworld or test/leds for saml21-custom board.

#### Steps to reproduce the issue
1) Clone board/saml21-xpro to board/saml21-jn
2) Tune boards/saml21-jn/include/periph_conf.h to match saml21e18b datasheet at best
2) update CPU name in board/Makefile.include to saml21e18b
3) Add saml21e18b in /cpu/sam0_common/Makefile.include otherwise linking will failed without rom and ram size.
4) compile with BOARD=saml21-jn make
5) flash the board with openocd + J-Link

#### Expected results
Hello world or blinking led

#### Actual results
#0  0xfffffffe in ?? ()
#1  0x00000648 in core_panic (crash_code=crash_code@entry=PANIC_HARD_FAULT, message=message@entry=0x3c8c "HARD FAULT HANDLER")
    at /home/parallels/RIOT-2018.04/core/panic.c:61
#2  0x000003da in hard_fault_default () at /home/parallels/RIOT-2018.04/cpu/cortexm_common/vectors_cortexm.c:347
#3  <signal handler called>
#4  0x7fffcf78 in ?? ()
#5  0x00002e26 in __libc_init_array ()
#6  0x0000041a in reset_handler_default () at /home/parallels/RIOT-2018.04/cpu/cortexm_common/vectors_cortexm.c:124

#### More infos
The custom board I try to make work has only ports PA8/PA9/PA10/PA11 as PIN OUT and SWD port for J-Link flashing.
 
PA8 TX Uart
PA9 RX Uart
PA10 LED0
PA11 LED1

I got also a SAML21 Xpro, that is full working (and a SAMR21 not tried)

#### Versions
parallels@parallels-vm:~/RIOT-2018.04$ ./dist/tools/ci/print_toolchain_versions.sh
Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.5.1
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.7.4
             coccinelle: missing

Thanks for helping me :)

nunux13, 2018-05-19:
I put a copy of my try here: https://github.com/nunux13/saml21-jn
Regards,

nunux13, 2018-05-19:
Also, i remove most of the features to keep it light at beginning. Hope it's ok ... 
 
https://github.com/nunux13/saml21-jn/blob/master/boards/saml21-jn/Makefile.features

nunux13, 2018-05-19:
[image]

nunux13, 2018-05-19:
[image]

nunux13, 2018-05-21:
Hello,
I started again from the last stable release. 
https://github.com/nunux13/saml21-jn-take2

Based on your remark, I setup everything including PIN for Feature not activated like SPI.

I also changed this array:

static const int8_t exti_config[2][32] = {
    { 0,  1,  2,  3,  4,  5,  6,  7, -1,  9, 10, 11, -1, -1, 14, 15,
      0,  1,  2,  3,  -1,  -1,  6,  7, 12, 13, -1, 15,  -1, -1, 10, 11},
    {-1, -1,  -1,  -1, -1, -1, -1, -1,  -1,  -1, -1, -1, -1, -1, -1, -1,
     -1, -1, -1, -1, -1, -1,  -1,  -1, -1, -1, -1, -1, -1, -1, -1, -1},
};

It'seems now possible to have some output ... between 2 crashs.

nunux13, 2018-05-21:
[image]

nunux13, 2018-05-21:
Reading symbols from bin/saml21-jn/hello-world.elf...done.
Remote debugging using localhost:3333
reset_handler_default () at /home/parallels/RIOT-2018.04/cpu/cortexm_common/vectors_cortexm.c:76
76	{
(gdb) continue
Continuing.
at91saml21e18b.cpu -- clearing lockup after double fault

Program received signal SIGINT, Interrupt.
0xfffffffe in ?? ()
(gdb) bt
#0  0xfffffffe in ?? ()
#1  0x000015f0 in _malloc_r ()
#2  0x00001326 in __sfmoreglue ()
#3  0x000013e4 in __sfp ()
#4  0x00001368 in __sinit ()
#5  0x00000ef8 in _puts_r ()
#6  0x00000fb0 in puts ()
#7  0x0000052a in hard_fault_handler (sp=0x20000188 <isr_stack+392>, corrupted=0, exc_return=4294967289,
    r4_to_r11_stack=0x20000168 <isr_stack+360>) at /home/parallels/RIOT-2018.04/cpu/cortexm_common/vectors_cortexm.c:262
#8  0x000003d2 in hard_fault_default () at /home/parallels/RIOT-2018.04/cpu/cortexm_common/vectors_cortexm.c:159
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

nunux13, 2018-05-21:
enabling timer get better startup

kYc0o, 2018-05-21:
Can you also share the output of `arm-none-eabi-readelf -S your-application.elf` ?

nunux13, 2018-05-21:
parallels@parallels-vm:~/RIOT-2018.04/examples/hello-world$ arm-none-eabi-readelf -S bin/saml21-jn/hello-world.elf
There are 19 section headers, starting at offset 0x1048f8:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .text             PROGBITS        00000000 010000 002008 00  AX  0   0  4
  [ 2] .stack            NOBITS          20000000 030000 000200 00  WA  0   0  1
readelf: Warning: [ 3]: Link field (0) should index a symtab section.
readelf: Warning: [ 3]: Info field (0) should index a relocatable section.
  [ 3] .relocate         REL             20000200 020200 00008c 08  WA  0   0  4
  [ 4] .bss              NOBITS          2000028c 02028c 0008ac 00  WA  0   0  4
  [ 5] .debug_info       PROGBITS        00000000 02028c 01e86f 00      0   0  1
  [ 6] .debug_abbrev     PROGBITS        00000000 03eafb 003c88 00      0   0  1
  [ 7] .debug_aranges    PROGBITS        00000000 042783 000578 00      0   0  1
  [ 8] .debug_ranges     PROGBITS        00000000 042cfb 000840 00      0   0  1
  [ 9] .debug_macro      PROGBITS        00000000 04353b 02b76d 00      0   0  1
  [10] .debug_line       PROGBITS        00000000 06eca8 00ee30 00      0   0  1
  [11] .debug_str        PROGBITS        00000000 07dad8 08107f 01  MS  0   0  1
  [12] .comment          PROGBITS        00000000 0feb57 00007e 01  MS  0   0  1
  [13] .ARM.attributes   ARM_ATTRIBUTES  00000000 0febd5 000032 00      0   0  1
  [14] .debug_frame      PROGBITS        00000000 0fec08 001078 00      0   0  4
  [15] .debug_loc        PROGBITS        00000000 0ffc80 0024bb 00      0   0  1
  [16] .symtab           SYMTAB          00000000 10213c 001c50 10     17 277  4
  [17] .strtab           STRTAB          00000000 103d8c 000aa9 00      0   0  1
  [18] .shstrtab         STRTAB          00000000 104835 0000c3 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings), I (info),
  L (link order), O (extra OS processing required), G (group), T (TLS),
  C (compressed), x (unknown), o (OS specific), E (exclude),
  y (purecode), p (processor specific)

kYc0o, 2018-05-21:
Can you format better the output? it's almost not readable... You can use \``` (github markdown) to format your output at the beginning and the end of the text, so it shows as if it was on the console.

kYc0o, 2018-05-21:
Anyways it looks good.

nunux13, 2018-05-21:
[image]

nunux13, 2018-05-21:
https://github.com/RIOT-OS/RIOT/issues/4470#issuecomment-164147588:

> Do you use the PLL? I've also seen this behavior before when you run from 8Mhz with the debugger connected, for some reason the debugger interfere with the MCU, disabling all breakpoints could solves the problem.

I remove the debugger and run from externe power supply, and that is looking great now.

nunux13, 2018-05-21:
After 5 days to try, I can confirme now RIOT OS is working on SAML21E18B.

The only fix you have to do regarding RIOT-2018.04 is:

```
diff -r RIOT-2018.04/cpu/sam0_common/Makefile.include RIOT-2018.04.diff/cpu/sam0_common/Makefile.include
5c5
< ifneq (,$(filter samd21g18a samd21j18a saml21j18b saml21j18a samr21g18a,$(CPU_MODEL)))
---
> ifneq (,$(filter samd21g18a samd21j18a saml21j18b saml21j18a samr21g18a saml21e18a saml21e18b, $(CPU_MODEL)))
```
**Cave-int and pits falls:**
When porting a new Board, you have to perfectly setup alls pins, even-if you disable the feature.
Don't run attached to the debugger. Stay halt after flashing, then move to a regular power source.
If using minicom for serial testing, press U to add carriage return. UART is mandatory for RIOT. Timer is mandatory for UART.

Thanks to everyone for patience and support.

Nunux

closed 2018-05-21 by nunux13

dylad, 2018-05-22:
Hi @nunux13.  
Glad to hear you solve your issue. 
I think you can also define ROM_LEN and RAM_LEN in the Makefile.include of your board if you don't want to modify any files under cpu/ folder
Cheers
Note: if you plan to use EXTI or RTC you will have some changes to do (like the array you modify) others periphs should run.

nunux13, 2018-05-22:
Yes, doing "ps" to show process on shell make crash. I still have to fix things...

Can you point me more "some changes" to check ?

I need to setup MRF24J40MA, done my first try, but need to try harder: I get crashs or no interface on "ipconfig" shell command ...

Regards,