RIOT-OS/RIOT #9157: (remote-reva) ccn-lite-relay: Only one content packet received. Returning Content not captured by RIOT sniffer tool

Burdaa, 2018-05-19:
Description:
Cannot receive more than one Content in ccn-lite on remote-reva over IEEE 802.15.4. All following Interests does not return Content. Also, only Interests are captured by RIOT sniffing tool (running on remote-reva). Return messages (Content) are not captured. 

Default ccn-lite-relay
node A: sudo BOARD=remote-reva PORT=/dev/ttyUSB0 make term
node B: sudo BOARD=remote-reva PORT=/dev/ttyUSB1 make term
(Alternatively using two different computers. Gives the same result either way)

node A: ccnl_cs /hei/ asdadadad
node B: ccnl_fib add /hei/ 00:12:4b:00:06:15:a0:b8

```
> ccnl_int /hei/
2018-05-19 13:32:55,848 - INFO #  ccnl_int /hei/
> 2018-05-19 13:32:55,867 - INFO #  PKTDUMP: data received:
2018-05-19 13:32:55,869 - INFO # ~~ SNIP  0 - size:   9 byte, type: GNRC_NETTYPE_CCN_CHUNK (2)
2018-05-19 13:32:55,870 - INFO # Content is: asdadadad
2018-05-19 13:32:55,871 - INFO # ~~ PKT    -  1 snips, total size:   9 byte
```

All following Interests are ignored. Tried to create Content under different Names and broadcasting the Interests.

Using the RIOT sniffing tool /applications/sniffer/ only the Interest is captured  (https://www.cloudshark.org/captures/f34257e469d0).

Any idea what can be wrong?

Same results on Ubuntu 16.04 and Kali

cgundogan, 2018-05-20:
@Burdaa thanks for reporting and your hints. Regarding the capturing problem: Do I understand correctly, that all subsequent Interests are received correctly, but they juts don't show up on the capturing node? So it's more or less a problem of the sniffing node?

Burdaa, 2018-05-20:
@cgundogan Yes, all subsequent Interests are received. I suppose it is starting to look like a problem with remote-reva working as a 802.15.4 sniffer together with the RIOT sniffer-script.  Would be interesting to see if both Interests and Content are captured with different boards, CC2538 and others.

cgundogan, 2018-05-21:
@Burdaa could be a problem with the sniffer script. I will try sniffing with an openlabs tomorrow in the office and capture from linux.

cgundogan, 2018-06-10:
@Burdaa  I did some testing last week with the openlabs and I was able to see all Interest and Data packets in wireshark. It's likely an issue with the sniffer script or the hardware. Nevertheless, the discussion moves in a separate direction. Should we close this issue and perhaps open a new one for the sniffer?

PeterKietzmann, 2018-06-13:
> Can this be related to #4822 ?

No because it is different hardware

>  suppose it is starting to look like a problem with remote-reva 

I couldn't find the issue right now but IIRC we had some issues with the cc2538 driver recently, unfortunately. Basically it seemed as if transmission stopped working with greater distances than 10 cm or so (@MrKevinWeiss can you give more insights or reference the issue?). @Burdaa it might be worth trying simple pinging between two motes. Use examples/gnrc_networking for this. 

> I updated the commit in RIOT/pkg/ccn-lite/Makefile

We updated the CCN-lite version recently (#9179), would you mind giving it an other try?

MrKevinWeiss, 2018-06-13:
Hmmm seems that I didn't create an issue or work on it...  I can pick it back up.  A brief description of what I noticed: Any board with a cc2538 cpu didn't seem to have range more than a few cms.  I tested on the remote and openmote-revb against each other and the samr21.  The only time I had range was samr against samr.  I also used a sniffer and the sniffer seemed to pick it up bit still not well (unless samr).  I tried using different tx powers and antennas and it did nothing.  Tomorrow I will confirm and write more info in an issue.

cgundogan, 2018-06-18:
The discussion seems to go in another direction now. We should open a new issue for the sniffer script if the problem persists. @Burdaa I will close this issue now, as we already updated the ccn-lite package on the current master. Feel free to re-open if you think otherwise.

closed 2018-06-18 by cgundogan