RIOT-OS/RIOT #9187: sys/newlib: gettimeofday() returns time since boot, not current wall time.

jcarrano, 2018-05-24:
#### Description

PR #4286 added support for `gettimeofday()` by implementing it on top of `xtimer_now_usec64()`.

[This definition](https://github.com/RIOT-OS/RIOT/blob/f49b8bbf496f28e4053de147b990a5c441c4ee94/sys/newlib_syscalls_default/syscalls.c#L501) is in conflict with the [specification](http://pubs.opengroup.org/onlinepubs/9699919799/functions/gettimeofday.html), which says:

> The gettimeofday() function shall obtain the current time, expressed as seconds and microseconds since the Epoch, and store it in the timeval structure pointed to by tp. The resolution of the system clock is unspecified.

#### Expected results

The current wall time is returned (if the board has a RTC). Otherwise, either the function should be made unavailable (this triggering a linker error if it is used), or it should always fail.

#### Actual results

`gettimeofday()` returns time since boot if xtimer was compiled in. If not, it fails with `ENOSYS`.

virajsahai, 2018-05-25:
@jcarrano if anyone isn't working on this..i think I can do it and create a pull request by tomorrow.

jcarrano, 2018-05-25:
@virajsahai As far as I know, [`xtimer_now()`](http://doc.riot-os.org/group__sys__xtimer.html#gaf3dbc76790c2beedce25bc1ba61faf33) returns a timestamp since an _unspecified time_ in the past (since last reset, maybe?). The docs are not 100% clear, but I would assume that, as measuring absolute time would require a RTC or some other time-setting mechanism.
`gettimeofday()`, on the other hand, requires _time since Epoch_.