RIOT-OS/RIOT #9191: Logic errors in nanocoap_get()

kb2ma, 2018-05-25:
#### Description
Three issues in nanocoap_get():

1. miscounts retries for a confirmable message
1. assumes the only valid success code is 2.05
1. includes unnecessary gotos

The easiest way to reproduce the retry problem is to turn disable the server you're testing with. You should see nanocoap_get() attempt only three retries, not the expected four.

The retry loop is problematic -- it counts up, but stops one retry short. The loop counts up, but also includes a test for remaining tries equal to zero. As @smlng pointed out in #9086, success should be tested against the 2.xx class, not the specific 2.05 response. Finally the gotos presently in the while loop are unnecessary; a break will suffice.

smlng, 2018-05-25:
someone working on this, otherwise I could?

kb2ma, 2018-05-25:
Go for it, @smlng! I'd be happy to review.

PR #10671 (2019-01-18): net/nanocoap: fix confirmable retry countdown
### Contribution description
The nanocoap sock module includes a `nanocoap_request()` function, which sends the request confirmably. As described in #9191, the function does not perform the expected count of retries when a send fails. This PR fixes the problem with a rework of the countdown logic.

### Testing procedure
This PR extends the `nanocoap_cli` test in #10640 with a server that provides an option (-i) to ignore a count of requests. So, you can setup a tap bridge with both endpoints running this test. One of them runs as a server and the o [... cut]
closed 2019-01-18 by smlng