RIOT-OS/RIOT #9201: telosb: radio initialization fails "again?"

PedroAlfNeto, 2018-05-25:
#### Description
Can't initialize cc2420.

#### Steps to reproduce the issue
Enabling debug messages in the cc2420 driver says cc2420: init: unable to communicate with device. And printing the reg value i get reg=0;

#### Expected results

I've been trying to build a simple comunication at the link-layer level (using 802.15.4) for a few weeks now, to start getting into RIOT, but have been failing for a while, and couldn't even find any similar example, that could compile to simpler boards like telosb or z1, in order to be able to simulate afterwards on cooja. Also, couldn't find the TSCH implementation, was it removed? Thanks.

#### Versions

Mint 18,

PedroAlfNeto, 2018-05-25:
Update: When simulating in cooja the `examples/default` compiled for z1 (i don't get anything at all with telosb, don't know), i get the DEBUG main()`main():cc2420: init: unable to communicate with device` and the reg value also prints 0;

smlng, 2018-05-28:
sounds like a configuration issue, i.e., check GPIO pins and SPI dev are all correctly set.

PedroAlfNeto, 2018-05-29:
I figured it was a configuration issue @smlng, but the problem is from Riot's implementation, and suposedly they had the same issue previously and thay had solved it, but since the error mantains with default example, i know it wasn't my mistake. Unfortunately i don't know how to configure the device & radio...

smlng, 2018-05-29:
Hi @PedroAlfNeto can you elaborate a bit more and also provide (a link to) your working configuration? Or if you like, provide a PR with a possible fix? Thx!!

PedroAlfNeto, 2018-05-30:
The previous reported issue was [telosb: radio initialization fails #5867]( https://github.com/RIOT-OS/RIOT/issues/5867), i think there is already a fix but i lack some background. I don't know what PR stands for, could you explain?

PedroAlfNeto, 2018-06-01:
I've concluded after testing with real hardware that actually the transmiter configuration is working, making this a cooja issue. Thank you for the suport. This way i'll close the issue.

closed 2018-06-01 by PedroAlfNeto