RIOT-OS/RIOT #9214: make: compiling is broken for any ARM platform on OS X.

kYc0o, 2018-05-28:
#### Description

As per #9080, the newlib-nano directory is not included anymore in my local setup for
ARM platforms. Apparently the reason is that the commands used to add the directory
don't work properly on OS X. 
#### Steps to reproduce the issue

Compile anything on current master for any ARM platform on OS X.

#### Expected results

Compiles correctly.
#### Actual results

```console
arm-none-eabi-gcc \
	-DRIOT_FILE_RELATIVE=\"boards/samr21-xpro/board.c\" \
	-DRIOT_FILE_NOPATH=\"board.c\" \
	-Wall -Werror -Wextra -Wno-implicit-fallthrough -mno-thumb-interwork -mcpu=cortex-m0plus -mlittle-endian -mthumb -mfloat-abi=soft -march=armv6s-m -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -ggdb -g3 -Os -std=c99 -fno-delete-null-pointer-checks -fdiagnostics-color -Wstrict-prototypes -Wold-style-definition -Werror=strict-prototypes -Werror=old-style-definition -fno-common -Wall -include '/Users/facosta/git/RIOT-OS/RIOT/examples/hello-world/bin/samr21-xpro/riotbuild/riotbuild.h'   -isystem  -I/Users/facosta/git/RIOT-OS/RIOT/core/include -I/Users/facosta/git/RIOT-OS/RIOT/drivers/include -I/Users/facosta/git/RIOT-OS/RIOT/sys/include -I/Users/facosta/git/RIOT-OS/RIOT/cpu/samd21/include -I/Users/facosta/git/RIOT-OS/RIOT/boards/samr21-xpro/include -I/Users/facosta/git/RIOT-OS/RIOT/cpu/sam0_common/include -I/Users/facosta/git/RIOT-OS/RIOT/cpu/cortexm_common/include -I/Users/facosta/git/RIOT-OS/RIOT/cpu/cortexm_common/include/vendor -I/Users/facosta/git/RIOT-OS/RIOT/sys/libc/include  -MD -MP -c -o /Users/facosta/git/RIOT-OS/RIOT/examples/hello-world/bin/samr21-xpro/board/board.o /Users/facosta/git/RIOT-OS/RIOT/boards/samr21-xpro/board.c
In file included from /Users/facosta/git/RIOT-OS/RIOT/boards/samr21-xpro/include/board.h:26:0,
                 from /Users/facosta/git/RIOT-OS/RIOT/boards/samr21-xpro/board.c:23:
/Users/facosta/git/RIOT-OS/RIOT/cpu/cortexm_common/include/cpu.h:35:10: fatal error: irq.h: No such file or directory
 #include "irq.h"
          ^~~~~~~
compilation terminated.
```
#### Versions

```
Installed toolchain versions
----------------------------
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
          native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
          msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
             avr-gcc: avr-gcc (GCC) 7.3.0
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: missing/error
               clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: "2.0.0" ("20150208")
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.17.0
```

jcarrano, 2018-05-28:
# ArchLinux

#9080 also breaks the build for ArchLinux. The reason is that Arch's package for newlib is defective and does not include separate headers for the `nano`.

## Steps to reproduce / expected results / actual results

Same as OSX.

## Versions

```
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (GCC) 8.1.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.1.0
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.11.2
               cppcheck: missing
                doxygen: 1.8.14
                 flake8: missing
                    git: git version 2.17.0
             coccinelle: missing
```

# Workaround

See https://github.com/jcarrano/RIOT/commit/7cd02ada9a7e51d7c42d545516d0169e8c2b8718

cladmi, 2018-06-01:
This should have been fixed by #9251 reverting the change
@kYc0o can you confirm ?
@jcarrano can you verify too ?

jcarrano, 2018-06-01:
@cladmi I can't verify it anymore because newlib has been fixed upstream for arch.

jcarrano, 2018-06-01:
On archlinux, it now compiles with latest RIOT and latest newlib.

cladmi, 2018-08-01:
I will close this as it has been solved, according to IRL discussions we had. Please re-open if it is not the case.

closed 2018-08-01 by cladmi