RIOT-OS/RIOT #9287: netif: use where appropriate

miri64, 2018-06-04:
#### Description
With #8844 merged now, we are able to use network interfaces stack independent now. There is however quite some work to do, still. This is the tracking issue for this progress.

### Stack adaptation
Currently only GNRC is supported. However, there are a few other stacks that could benefit from it:

- [ ] emb6 (@miri64)
- [ ] lwip (@miri64)
- [ ] openthread (@jia200x or @aabadie)

### Usage
We should also replace the usage of `gnrc_netif` in places, where otherwise no direct GNRC-interaction is happening (? mark unclear candidates):

- [ ] `gcoap` (incl. example, @kb2ma)
- [ ] `uhcpc` ? (@kaspar030)
- [ ] `ifconfig` shell command ? (@miri64)
- [ ] `sntp` shell command

aabadie, 2018-06-04:
I don't understand the link between this issue and #8844

miri64, 2018-06-04:
Sorry I meant #8841. Fixed in OP.