RIOT-OS/RIOT #9297: nib: crash after random time execution on `samr21-xpro`

cladmi, 2018-06-06:
#### Description

During suit hackathon, I get crash on `samr21-xpro` with `gnrc_networking`.

It is connected to an rpi+openlab border router and there are certainly other nodes non RIOT sending traffic

```
2018-06-06 10:54:16,653 - INFO # Context before hardfault:
2018-06-06 10:54:16,655 - INFO #    r0: 0x000119d0
2018-06-06 10:54:16,657 - INFO #    r1: 0x00000008
2018-06-06 10:54:16,658 - INFO #    r2: 0x00011a36
2018-06-06 10:54:16,660 - INFO #    r3: 0x00000021
2018-06-06 10:54:16,662 - INFO #   r12: 0x20001510
2018-06-06 10:54:16,663 - INFO #    lr: 0x000025ad
2018-06-06 10:54:16,665 - INFO #    pc: 0x00003fa0
2018-06-06 10:54:16,666 - INFO #   psr: 0x01000000
2018-06-06 10:54:16,667 - INFO # 
2018-06-06 10:54:16,667 - INFO # Misc
2018-06-06 10:54:16,669 - INFO # EXC_RET: 0xfffffffd
2018-06-06 10:54:16,673 - INFO # Attempting to reconstruct state for debugging...
2018-06-06 10:54:16,674 - INFO # In GDB:
2018-06-06 10:54:16,675 - INFO #   set $pc=0x3fa0
2018-06-06 10:54:16,676 - INFO #   frame 0
2018-06-06 10:54:16,677 - INFO #   bt
2018-06-06 10:54:16,677 - INFO # 
2018-06-06 10:54:16,680 - INFO # ISR stack overflowed by at least 8 bytes.
```
```
(gdb) frame 0
#0  0x00003fa0 in _nib_drl_remove (nib_dr=nib_dr@entry=0x200019b8 <_netifs>) at /home/harter/work/git/RIOT/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c:351
351             nib_dr->next_hop->mode &= ~(_DRL);
(gdb) bt
#0  0x00003fa0 in _nib_drl_remove (nib_dr=nib_dr@entry=0x200019b8 <_netifs>) at /home/harter/work/git/RIOT/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c:351
#1  0x000025ac in _handle_rtr_timeout (router=router@entry=0x200019b8 <_netifs>) at /home/harter/work/git/RIOT/sys/net/gnrc/network_layer/ipv6/nib/nib.c:1151
#2  0x0000334e in gnrc_ipv6_nib_handle_timer_event (ctx=0x200019b8 <_netifs>, type=<optimized out>) at /home/harter/work/git/RIOT/sys/net/gnrc/network_layer/ipv6/nib/nib.c:352
#3  0x000023ec in _event_loop (args=<optimized out>) at /home/harter/work/git/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:302
#4  0x00000258 in sched_switch (other_prio=<optimized out>) at /home/harter/work/git/RIOT/core/sched.c:209
#5  0x2000148c in _stack ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

#### Versions

Current master version 59e18df

miri64, 2018-06-06:
@cladmi can you be a little clearer how to reproduce this issue? Also, it would be great if you could tell, which values `nib_dr` or `nib_dr->next_hop` have.

miri64, 2018-06-06:
Ah `nib_dr=nib_dr@entry=0x200019b8 <_netifs>`.... That is definitely wrong.

miri64, 2018-06-06:
#9300 should fix this.

miri64, 2018-06-06:
Another question is though: why does the router timeout in the first place? Does the RPi not answer to router solicitations?

cladmi, 2018-06-07:
I do not really know why… but what I know is that with the fix, on samr21 after receiving a lot of traffic (30k-70k) the transfer could not finish because I was not able to request/receive all blocks. Which would make sense if the router timeouts.

Before it was crashing and with the fix I was just not able to finish the transfer.
I did not have any nib debug info unfortunately.

It was also a busy room with everybody on the same network with different hardware maybe interfering.

For the whole setup configuration @bergzand would know more.

PR #9300 (2018-06-07): gnrc_ipv6_nib: use router instead of netif for router timeout context
### Contribution description
`GNRC_IPV6_NIB_RTR_TIMEOUT` [expects a default router entry][doc] as
context, so using the `netif` here is clearly wrong. This causes the
hardfault described in #9297.

[doc]: http://doc.riot-os.org/group__net__gnrc__ipv6__nib.html#ga1c7e892ef6533f699c4e64737c2847c2

### Issues/PRs references
Fixes #9297
closed 2018-06-07 by cgundogan