RIOT-OS/RIOT #9309: CoAP roadmap proposal

kb2ma, 2018-06-08:
We would like to complete basic CoAP functionality for nanocoap and gcoap, including the Observe and Block extensions. This functionality provides a solid foundation on which to build higher level services, like pub-sub. This page proposes a sequence of PRs toward completion.

The CoAP Option functions below bring the nanocoap and gcoap APIs much closer together. In fact the gcoap API will become mostly convenience/sugar on the nanocoap API. See [API Options](https://github.com/RIOT-OS/RIOT/wiki/CoAP-API-Options-Update) for details.

For a summary of the goals for nanocoap, see the proposed [module documentation](https://kb2ma.github.io/riot-coap-api/nanocoap-api-doc.html), which itself will be a PR.

Any comments appreciated!

| Subject | Library | PRs | Description |
| ------- | ------- | --- | ----------- |
| coap_pkt_t message build | nanocoap | #9085 | Use a coap_pkt_t when build a message. |
| Simpler API for options | nanocoap | #9085 | Adds coap_opt_add_...() functions. |
| Simpler API for options | gcoap | #9156 | Use coap_opt_add_...() functions. |
| Isolate coap_put_option...() functions| nanocoap | #8920, ... | Use a compile-time macro to deprecate use of buffer-based application-level functions like coap_put_option_ct(). The preferred approach for that function is to use the coap_pkt_t-based coap_opt_add_uint(). |
| Isolate coap_get_option...() functions| nanocoap | #8920, ... | Rename these functions to coap_opt_get...(). Can inline the old functions on top of the new ones. However, we should deprecate the old function names. |
| Add module level documentation | nanocoap | | Documents the API created above and provides more direction on how to use it. See [doc page](https://kb2ma.github.io/riot-coap-api/nanocoap-api-doc.html) for WIP. |
| API update for 'finish' functions. | gcoap | | Deprecate use of `gcoap_finish()` after payload is written, in favor of `gcoap_opt_finish_format()` before payload is written. Simplifies implementation and removes gcoap-specific code. |
| Finish Block extension | nanocoap | #8932 | open PR for block2; block1 already complete |
| Finish Observe extension for server | gcoap | #7548 | open PR includes confirmable notifications |
| Finish confirmable messaging | gcoap | | Add option to pause calling thread rather than copying the sent buffer. Add immediate ACK with follow-on response. |

kYc0o, 2018-06-26:
@kb2ma would you mind to have "ticks" on the related PRs in the description? This way the progress can be seen in a more abstract view.

kb2ma, 2018-06-27:
I tried to include ticks in the table, but can't figure out how to do it, so used strikethrough instead to show completion. Which is more important, the ticks or the table?

kYc0o, 2018-06-27:
Ok I see, of course the table is more important! No doubt about it. I'll then link this issue to the project and add the ticks there.

Thanks for tackling this!

miri64, 2018-06-27:
How about a project to organize this and have an overview? I did the same for [6Lo refactoring](https://github.com/RIOT-OS/RIOT/projects/18)

kb2ma, 2018-07-02:
> How about a project to organize this and have an overview?

Thanks @kYc0o, and nice idea, @miri64! I'll keep this issue but focus it on Options and remove the PR references. I'll create a project for the Options work and include a link to this issue. I'll also remove the references to confirmable messaging, Observe, and Block. Those really are part of the Network Stack High Layers project.

corebase22, 2019-11-28:
> @kb2ma would you mind to have "ticks" on the related PRs in the description? This way the progress can be seen in a more abstract view.

kb2ma, 2019-11-28:
No thanks on the ticks. This issue is almost ready to be closed. We are of course always happy to receive feedback or improvements on the feature itself.

kb2ma, 2020-03-13:
Done! Thanks to everyone for helping with the migration. Once `gcoap_add_string()` and `gcoap_finish()` are actually removed, gcoap.h will contain only functions that are specific to gcoap, mostly how it sends and receives messages.

closed 2020-03-13 by kb2ma

leandrolanzieri, 2020-05-27:
@kb2ma can the related [GitHub project](https://github.com/RIOT-OS/RIOT/projects/21) be closed?

kb2ma, 2020-05-27:
Leaving the project open was an oversight; thanks for noticing. I just closed it.