RIOT-OS/RIOT #9324: gnrc_tcp: tests need work

miri64, 2018-06-11:
#### Description
The current tests for `gnrc_tcp` have some problems: 

* There are two tests (server and client) that are interdependent on each other
* They don't test against an independent implementation
* They only run on native

#### Proposal to makes things better
I propose the following changes to better tests to (a) help with automation (b) make them also run for other boards

- [ ] Use something like `ethos` or `slip` on boards instead of the radio for now (we can later add some larger scale integration tests for radios)
- [ ] Let the test(s) test against netcat via a scripting solution

brummer-simon, 2018-06-17:
The TCP tests could really need an improvement. About that netcat scripting. Are there any tests where 
this is done? It would be good to implement the tests in a similar way than is done for other test cases.

About the " They take ages to finish". The MSL - Timeout (that is the reason why the test takes ages to finish) can be reduced with via a define, however the timespan is demanded by the TCP RFC. So for a test its okay to reduce it but in general it behaves as it should.

About "They only run on native". Thats not true, I have tested it on samr-21xpro and it works. So that point can be crossed of the list i guess.

In two weeks I have a two week holiday. I think then I find the time to address this issue properly.

miri64, 2018-06-18:
> About the " They take ages to finish". The MSL - Timeout (that is the reason why the test takes ages to finish) can be reduced with via a define, however the timespan is demanded by the TCP RFC. So for a test its okay to reduce it but in general it behaves as it should.

I did not know you are testing timeouts as well. This should maybe made be clearer in the description in the README or the test output then (it wasn't certainly clear to me by just looking at the applications' code).

> About "They only run on native". Thats not true, I have tested it on samr-21xpro and it works. So that point can be crossed of the list i guess.

1. If I try to run them on `samr21-xpro` I first of all get 

```
serial.serialutil.SerialException: [Errno 2] could not open port tap0: [Errno 2] No such file or directory: 'tap0'
```
   (or `tap1` respectively for client). Yes, I know this is due to the app resetting the PORT environment variable, but this (in addition to the interdependence of the tests) might make them bad for automation (usually setting `SERIAL` for both flashing and `make term` is enough for the build system to determine the port). 
2. I ran the tests and they don't even send TCP packets (I sniffed it). This isn't surprising, since the apps are using link-local addresses not based on the link-layer addresses (this is [assumed to be the case by 6Lo-ND](https://tools.ietf.org/html/rfc6775#section-5.6) and was also the case before the big interface/forwarding rework). Only if I adapt the address (incl. interface, because it is different on `samr21-xpro`) with the environment variables expected by the app (again not ideal for automation + I expect some documentation for usage if this is necessary) I see traffic.
3. Only then, I get

```
2018-06-18 13:26:59,370 - INFO # Client running: TID=0
2018-06-18 13:28:59,556 - INFO # TID=0 : gnrc_tcp_send() : -ECONNABORTED
2018-06-18 13:28:59,559 - INFO # TID=0 : Payload verfication failed
2018-06-18 13:28:59,567 - INFO # TID=0 : 1 test cycles completed. 0 ok, 1 faulty, 1 failed payload verifications
```

miri64, 2018-06-18:
PS: is it wanted, that the 2048 bytes of payload are send over byte-by-byte?

brummer-simon, 2018-06-19:
Nope. And the last time I tested it was never sent byte-by-byte. Well I guess I have to look into that more closely. It would be great if you could give some input on changes in the lower network stack levels that occured recently. From my point of view it seems that much changed since last year.

miri64, 2018-06-19:
Indeed there is changed a lot since the last year, but in theory this shouldn't have much influence on TCP.

* Network interfaces were consolidated, so that there is only a single module to manage them: `gnrc_netif` (no more PID array in `gnrc_netif`, and various APIs in `gnrc_sixlowpan_netif`, `gnrc_ipv6_netif`, etc.), but the main API to access them is still `gnrc_netapi`. So if you used that to get e.g. your MSS based on the link-layers max packet size, this still should work with `gnrc_netapi_get(netif->pid, ...)`
* IPv6 neighbor discovery and the overall management of neighbors completely re-implemented and most issues with the old ND were fixed. This has on higher layers only influence on traffic with IPv6 link-local addresses, as you now have to provide the interface with them (which you already did).

miri64, 2019-08-10:
Addressed by #11930.

closed 2019-10-03 by miri64