RIOT-OS/RIOT #9342: OTA: tracking issue

bergzand, 2018-06-13:
# Tracking OTA progress

Tracking issue for the different parts required to enable over the air updates.

[image]

The goal is to get functionality required for OTA updates one by one instead of a single huge PR.

cladmi, 2018-06-28:
Linker offset supports depends on https://github.com/RIOT-OS/RIOT/pull/9451 to allow testing.

danpetry, 2018-08-07:
# State of development
Please update this list as required so that we all know what we're up to

## Development branches
Here's a list of active branches where people are developing. Please add a short description if you can get around to it.
https://github.com/kYc0o/RIOT/tree/hackathon_suit_2018  - Paco dev branch
https://github.com/kYc0o/riot-ota-server/tree/master/otaserver - OTA server (Paco fork)
https://github.com/aabadie/ota-server - OTA server (original)
https://github.com/danpetry/RIOT/tree/ota/sandbox - Dan dev branch

## Related issues not in the diagram above
Either because they are dependencies to the issues above, or some other reason. Basically here are some issues you might want to be aware of, and help along if you can.
https://github.com/RIOT-OS/RIOT/pull/8902 - Original OTA PR
https://github.com/RIOT-OS/RIOT/pull/9351 - PR to allow partial ROM flashing for cortex m cpus

## Current work being undertaken
Dan - working on RAPstore project on integration, so architecture can be informed by situational concerns
Anton (OYTIS) - implementing periph/flasher module for a board (? which one)

cladmi, 2018-08-13:
For *binflash* I would also have a dependency of `FLASHFILE`: https://github.com/RIOT-OS/RIOT/pull/8838

And I am thinking, why do we even need to do `binflash`. Flashing a binary means having both a file and an address. And this could be done by generating an `ihex` file instead.

kYc0o, 2018-08-13:
> And I am thinking, why do we even need to do binflash. Flashing a binary means having both a file and an address. And this could be done by generating an ihex file instead.

As far as my early tests went, it was very complicated to manage ihex files especially when you want to flash bootloader + image (slot 1 or 2). Then build an hex file with such information was very challenging at that time. To speed up things, and make something more reliable I used bin files flashed at the needed address.

Now we have all the information we need to know where to flash, so I'd say there's no problem using and flashing binary files while a bootloader takes the beginning of the flash.

bergzand, 2018-08-15:
For the bootloader, the mimimal set of required metadata is this:
```
/**                                                                              
 * @brief Structure to store firmware metadata                                   
 * @{                                                                            
 */                                                                              
typedef struct {                                                                 
    uint32_t magic_number;              /**< metadata magic_number (always "RIOT")  */
    uint32_t version;                   /**< Integer representing firmware version  */
    uint32_t start_addr;                /**< Start address in flash                 */
    uint32_t chksum;                    /**< checksum of metadata                   */
} firmware_metadata_t;                                                           
/** @} */                                                                        
```

 - **magic_number**: To indicate that a firmware metadata field is located here.
 - **version**: Version of this firmware. The bootloader will boot the firmware with the highest version number
 - **start_addr**: Start address of the firmware. This way the bootloader can easily work with variable metadata sizes.
 - **chksum**: Checksum of the metadata to check for a valid metadata (not firmware).

bergzand, 2018-08-15:
# Documentation todo

- dependencies
  - CPU peripherals required
  - metadata format and bootloader interaction

- metadata fields
- CPU initialization (happens twice)

cladmi, 2018-08-16:
Notes about `flashbin`:

### edbg

 `edbg` is already flashing binary files.
An offset from the start address can be given using `-o|--offset` the PR currently uses this as an absolute address which is wrong in general. So flashing a binary from start address works.

Some work is only required to disable erasing everything and supporting flashing from an `offset` of the start address. Which is only needed when flashing sub parts of the rom/invalidating a slot.

### Openocd

Openocd supports flashing from binary as is, except that the flashing address must be given.
It uses `offset` which is an offset from what is in the firmware, so offset from the link address for elf files/hex files, offset from `0` for binary files.

I plan to use the `ROM_START_ADDR` in the openocd script to make flashing a binary work by default without external configuration. And use `OFFSET` as an offset from this. 

This would give a common behavior between boards using edbg and openocd.

emmanuelsearch, 2018-08-17:
Minutes from this week's OTA meeting (15.08.2018) available [here](https://github.com/RIOT-OS/RIOT/wiki/minutes-from-OTA-Task-Force-virtual-meeting-15.08.2018)

waehlisch, 2018-08-18:
Which option is preferred to add comments/questions on the minutes? Should I comment via this issue, create a new issue, or ...?

emmanuelsearch, 2018-08-18:
I'd say: comment here?

kb2ma, 2018-08-20:
I agree that it makes sense to use the existing Block1 functionality for now. Conceptually, some host sends firmware to the (RIOT) device, although I guess use of PUT/POST or GET depends on the higher-level application protocol. At any rate, the device is on the receiving end.

Also, "roadblock" is not a word I like to see associated with my name! I'm reviewing the RFC and Block2 PR (#8932), and plan to comment in the next day or so on moving forward.

emmanuelsearch, 2018-08-20:
@kb2ma actually Block2 has advantages in terms of security characteristics (since it is pull-based).
If #8932 could be merged fast into master (before other CoAP reworks), it would improve the initial OTA solution we are going for, no doubt.
That said, we plan OTA enhancements/reworks anyways, after an initial (basic) OTA mechanism is merged in the master branch.
Hence, just let us know about #8932, indeed ;-)

cladmi, 2018-09-05:
Both `edbg` and `openocd` binflash have been merged. The branch will need to be rebased with the changes. @kYc0o we can spend some time doing this and testing this afternoon if you want.

It currently still requires setting two different variables to handle both openocd and edbg to flash a binary file but will be addressed by https://github.com/RIOT-OS/RIOT/pull/8838.

cladmi, 2018-09-05:
Another dependency I remember as I am rebasing the branch, is process dependencies for CPU and move `stm32_common` `periph_flashpage` dependencies to `Makefile.dep` I could easily provide a test for this one even if the test does not need to be merged.

kYc0o, 2018-09-14:
Finally I got to test the current state with a rebased branch with most recent changes in master related to OTA. The most updated branch is at [1].

Two problems:
  - Compile for slot 2 doesn't seem to work with the current verification for slot sizes:
    `Specified firmware size does not fit in ROM`. However, by bypassing the assert in cortexm.ld it compiles and actually is linked correctly, but:
  - For edbg this line is not working `EDBG_ARGS += $(addprefix --offset ,$(IMAGE_OFFSET))`:
  ```console
/Users/facosta/git/RIOT-OS/RIOT/dist/tools/edbg/edbg --offset $((0x1000 --offset + --offset 0x1f800)) -t atmel_cm0p -b -v -p -f /Users/facosta/git/RIOT-OS/RIOT/examples/suit_updater/bin/samr21-xpro/suit_updater-slot2.signed.bin
bash: 0x1000 --offset + --offset 0x1f800: syntax error in expression (error token is "--offset + --offset 0x1f800")
make: *** [/Users/facosta/git/RIOT-OS/RIOT/makefiles/riotboot.mk:112: riotboot/flash-slot2] Error 1
```

cladmi, 2018-09-17:
For `edbg` I did not thought about the value with spaces so 'addprefix' does not work in that case.
Two solutions:
* Replace `IMAGE_OFFSET=$((0x1000+0x1f800))` without spaces
* Replace the EDBG_ARGS line with `$(if $(IMAGE_OFFSET),--offset $(IMAGE_OFFSET)))`

The first one can be done directly on our side to make it work, and upstream the proper fix to RIOT after.

cladmi, 2018-09-17:
I found the issue for the offset on slot 2, the assert was write. In `makefiles/riotboot.mk` I wrote
`FW_ROM_LEN=$(RIOTBOOT_FW_SLOT_SIZE)`
when the slot size includes both the metadata and the firmware, so should be
`FW_ROM_LEN=$(RIOTBOOT_FW_SLOT_SIZE - $(RIOTBOOT_HDR_LEN))`

And then the ldscript changes can be reverted.

cladmi, 2018-09-17:
We may are not consistent with the names yet. If `slot` means metadata + firmware, the generated files should be adapted.

cladmi, 2018-09-17:
I also have some issues when building the bootloader on my laptop as `git am` which applies patches to packages tries to use `HOME` to find git configuration and extract email and author name but we do not pass it down for building.

For me it is an issue that 'git am` tries to use local configuration to do the commit. I want to fix this for some time but did not have real visible problems to show.

bergzand, 2018-09-17:
Just as a side note I made while playing around with @kYc0o branch, if the periph_gpio, periph_uart and uart_stdio are disabled (I had to comment out some initialization functions), the bootloader takes 1428B on the samr21-xpro. Original was 2920B. I didn't test whether it was still functioning, but I don't see a reason why the bootloader would need those peripherals. Chopping the flash requirement in half seems to be the more beneficial trade off here.

On a practical side, this requires making the initialization of the gpio and (stdio_)uart optional. While not impossible, this would add another todo on the already large list. I just want this to be noted down somewhere so I don't forget about it.

danpetry, 2018-09-18:
Would people possibly want the stdio uart for debugging?

> this would add another todo on the already large list.

I vote for getting the bootloader merged without this, and then optimizing more for size later. For this reason.

bergzand, 2018-09-18:
> Would people possibly want the stdio uart for debugging?

Probably, but if the bootloader is functioning, it should not be necessary. Furthermore, if your device is somewhere in the wall, the UART might not be accessible and kinda useless to include in the config.

> I vote for getting the bootloader merged without this, and then optimizing more for size later. For this reason.

That's exactly what I would also prefer here, just want to have it somewhere written down :smile:

cladmi, 2018-09-18:
Additional note, with the current implementation using two slots, saving space in the bootloader is only useful if you save two pages, as both slots have the same size.

For the `samr21-xpro` with a page size of `256` you save ROM, with the `iotlab-m3` the page size is `2048` so there would be nothing gained here, with the current implementation.

cladmi, 2018-09-18:
Proper fix for edbg issue with spaces in `IMAGE_OFFSET`: https://github.com/RIOT-OS/RIOT/pull/9956

cladmi, 2018-09-19:
@kYc0o I will do it and remove non needed commits, I will still keep the last fixups for the moment.

cladmi, 2018-09-19:
Rebased, I checked the rebased diff against the previous one.

Now I can do `make riotboot/flash-bootloader`, `make riotboot/flash-slot1` and `make riotboot/flash-slot2` for `samr21-xpro` without modifications.

cladmi, 2018-09-19:
I also included changes to make it work on `iotlab-m3` using `openocd.sh`. `IMAGE_OFFSET` and `IMAGE_FILE` should be exported to be visible in the script.

cladmi, 2018-09-19:
And fixed for `iotlab-m3` `flash-slot2` there are issues with `$(( ))` and `export`.

I also think that `flash-slot1` and `flash-slot2` should NOT flash the bootloader.

danpetry, 2018-09-19:
### Cpu_init idempotency test
It was found that re-running cpu_init() gave a false negative as a test for idempotency of the startup sequence. For the SAM R21, this is because the newlib initialization is run after cpu_init(), and it sets the state of the generic clock register which clocks the stdio USART. This means that no `[SUCCESS]` assert printout is received by the testing machine.

In general, the startup sequences are, currently, quite heterogeneous. This means that an integration test for the whole bootloader can't really be broken down. (Such a test would be to load a firmware with the bootloader and have the firmware print `[SUCCESS]`) Although this test is important to help people to isolate bugs more quickly - we have already had issues in this area - it would require refactoring of the startup code for all different boards to a common format. This work is also considered non-blocking to the implementation of a bootloader.

I've raised an issue (https://github.com/RIOT-OS/RIOT/issues/9964) for this work, which can continue in parallel with the initial implementation of a bootloader.

kYc0o, 2018-09-19:
> if the periph_gpio, periph_uart and uart_stdio are disabled (I had to comment out some initialization functions)

Can you share the diff if you have it? I'd like to know how easy/hard is to achieve this.

bergzand, 2018-09-19:
> Can you share the diff if you have it? I'd like to know how easy/hard is to achieve this.

I'd had to comment out a few things such as initialization of stdio and buttons/leds
```patch
diff --git a/sys/newlib_syscalls_default/syscalls.c b/sys/newlib_syscalls_default/syscalls.c
index 91543fcd1..86a995ece 100644
--- a/sys/newlib_syscalls_default/syscalls.c
+++ b/sys/newlib_syscalls_default/syscalls.c
@@ -65,7 +65,9 @@ char *heap_top = &_sheap + 4;
  */
 void _init(void)
 {
+#ifdef MODULE_UART_STDIO
     uart_stdio_init();
+#endif
 }
 
 /**
```

This should be an ifdef instead of a cpp comment :)
```patch
diff --git a/boards/samr21-xpro/board.c b/boards/samr21-xpro/board.c
index 9c63ee27b..df2fd57dc 100644
--- a/boards/samr21-xpro/board.c
+++ b/boards/samr21-xpro/board.c
@@ -26,7 +26,7 @@
 void board_init(void)
 {
     /* initialize the on-board LED */
-    gpio_init(LED0_PIN, GPIO_OUT);
+    //gpio_init(LED0_PIN, GPIO_OUT);
 
     /* initialize the CPU */
     cpu_init();
```

And blacklisting the modules in the makefile

```patch
diff --git a/dist/riotboot/Makefile b/dist/riotboot/Makefile
index 70d6ac8c3..daa7a6f1a 100644
--- a/dist/riotboot/Makefile
+++ b/dist/riotboot/Makefile
@@ -5,7 +5,7 @@ BOARD ?= samr21-xpro
 USEMODULE += firmware
 
 CFLAGS += -DNDEBUG -DLOG_LEVEL=LOG_NONE
-DISABLE_MODULE += auto_init
+DISABLE_MODULE += auto_init uart_stdio
 
 RIOTBASE ?= $(CURDIR)/../..
```

danpetry, 2018-09-20:
Hi all. Would it make sense to re-name the task force and this issue from "OTA" to "OTA and bootloader"? Or maybe "Bootloader and software updates"? Although the bootloader work we're doing will lead into OTA, there's no over-the-air updating happening... I've had feedback that it's misleading for someone wanting to know at-a-glance what we're doing... what do you think.

waehlisch, 2018-09-20:
I gave @danpetry the feedback: the current title doesn't reflect the work properly.

danpetry, 2018-09-21:
I do believe that it is important to use clear and readily understood semantics. In this case, I agree with @waehlisch that "OTA" doesn't accurately reflect the work package which is described here and could be misinterpreted by people looking at the task force for the first time. I think his viewpoint acts as evidence for this.

emmanuelsearch, 2018-09-21:
the prototype we are aiming for, PRed per component here, amounts to what is reported in slide 3 of SUIT hackathon [recap](https://github.com/suit-wg/Hackathon-Interim-Berlin/blob/master/SUIT-Berlin-Hackathon-recap.pdf).
This is over-the-air firmware udpate.
So let's stop quibbling about words and get to work ;)

emmanuelsearch, 2018-09-21:
To disambiguate, I added an overview paragraph at the top of this issue.
Feel free to improve!

waehlisch, 2018-09-21:
@emmanuelsearch it's not about quibbling but clarity.
As we don't have a separate issue that tracks progress of this bootloader work, a basic requirement for OTA but not core OTA, it would be helpful to update the issue name.

If there is no explicit reason against it, I will update over the weekend.

emmanuelsearch, 2018-09-21:
@waehlisch so far this issue tracks the technical progress merging parts of the SUIT hackathon OTA prototype.
If some consider opening a separate issue for the bootloader, which relates to #9342 as overarching issue, perhaps that might be useful?
What would be misleading is renaming this issue with something less "overarching" than OTA firmware update (which is about more than only the bootloader).
Makes sense?

cladmi, 2018-09-21:
> I gave @danpetry the feedback: the current title doesn't reflect the work properly.

The previous dependency graph was doing the job (except a bit outdated) and the project tracker https://github.com/RIOT-OS/RIOT/projects/11 also gives some updates on the currently active work being done.

If you want the title to reflect the work properly, it should also include everything that is required for OTA. Like fixing the dependency definition order in RIOT, or having a common variable for the file to flash and fixing all the flashers, or the developments done by bergzand and kYc0o already integrated in RIOT.

This issue is not a management tool, it's a way for people developing to synchronize the work they are doing to split and update the task force branch. Currently I preferred using the comment sections as a more chatty way of giving updates over modifying a bullet list.

If nothing was mentioned about the bootloader, it is because no one was working on splitting it.
Our reference branch is only working with the last master since two days ago and is now in a dedicated PR with some testing documentation.

waehlisch, 2018-09-21:
@cladmi the previous dependency graph highlighted two building blocks (1) Linker/Bootloader and (2) OTA, and each building showed the same amount of details. Furthermore, roughly half of the discussions in this issue relate to the bootloader. I really don't see why this issue is exclusively related to OTA.

danpetry, 2018-09-25:
Would "OTA support" be a title that would satisfy the requirements discussed here? It seems to me that all parties are, essentially, arguing for something overarching which doesn't exclude bootloader implementation (or, indeed, anything else we need to implement). Would this definition therefore be appropriate?

waehlisch, 2018-10-02:
@cladmi I'm confused. A boot loader basically calls the OS into memory. Boot loaders may have different capabilities (e.g., think about first-stage and second-stage boot loaders). That a boot loader decides between two different OS versions (or "firmwares") is not a conflict.  And, you can use this piece of software outside of OTA

I think what confused me is that a link to #10065 (or any other issue that tracks the state ot the boot loader) was missed, and that some discussions in #9342 related more to a boot loader instead of core OTA.

waehlisch, 2018-10-04:
@kYc0o 

> #10065 is pretty new and in the diagram is clearly the next step (it was also clear in the previous diagram) so it was missed for obvious reasons.

I don't know why it was missed for obvious reasons but this missing piece of information was one reason why I was originally arguing that the caption is misleading. And, I don't care if you discuss bootloader and OTA in a single or in separate issues/PRs ;).

> Thus,  to be clear about my position: OTA stands for the whole process of updating the firmware of a running RIOT node, which involves a  bootloader, wired/wireless protocols to transfer the firmware and the tools to secure it, if it's wanted. IMHO no renaming is needed.

I don't agree here: OTA is one functionality, boot loader is another. OTA needs a boot loader, but a boot loader doesn't need OTA.

For me, OTA includes everything that (securely) delivers the OS wirelessly to the target.

The boot loader ensures to execute the stored OS, maybe also securely.

Side comment: There are other options to ship the OS. Think about an USB stick.

emmanuelsearch, 2018-10-15:
@bergzand @cladmi @kYc0o @danpetry for the record, to summarize based on the last TF discussions, the immediate next steps are simplified such that we go for:
(please modify/complement if I missed something)

- polishing/finishing #10065 adding multislot;
- initiate a separate PR providing a SUIT manifest parser;
- initiate a separate PR generic Firmware_Flash_Write (with init, putbytes, similar to [this](https://github.com/bergzand/RIOT/blob/app/suit-ota/sys/firmware/firmware_manifest.c#L201), and finish writing an image header compatible with #10065).

After that, a simple follow-up PR providing CoAP handlers for OTA compliant with SUIT ([draft-01](https://tools.ietf.org/html/draft-moran-suit-manifest-01) version) would be close-at-hand, now that CoAP block2 support has been merged.

emmanuelsearch, 2019-08-15:
#11818

cladmi, 2019-08-15:
Issue with `riotboot`: "bootloaders|tests/riotboot: broken with BUILD_IN_DOCKER and wrong flashfile" #12003

miri64, 2020-07-02:
I don't see any progress tracking in this tracking issue. So what is the progress?

maribu, 2022-09-19:
Ping?

maribu, 2023-05-22:
Let's see if anyone screams when I close this.

closed 2023-05-22 by maribu

kYc0o, 2023-05-22:
😱