RIOT-OS/RIOT #9359: thread: possibility to add architecture specific data to thread control block required

gschorcht, 2018-06-14:
During the ports of RIOT for ESP8266 and ESP32 MCUs, I had to learn that it would be really helpful if it would be possible to extend the thread control block by architecture specific data that belong to the thread context, .e.g., the saved interrupt level for nested interrupts with task switching.

At the moment I'm doing that by managing additional structures in additional tables for all threads in parallel to the thread table.

From my point of view it would be very easy to realize it as following:

The definition of ```struct thread``` is by a simple preprocessor definition, for example ```THREAD_ARCH_EXT```. Since ```thread.h``` includes ```cpu_conf.h```, additional members for specific data could be added by the architecture:

```
struct _thread {
    char *sp;                       /**< thread's stack pointer         */
    uint8_t status;                 /**< thread's status                */
    uint8_t priority;               /**< thread's priority              */
    ...
    THREAD_ARCH_EXT                 /**< architecture specific extensions */
    ...
}
```
If the definition of ```THREAD_ARCH_EXT``` in ```cpu_conf.h``` would be
```
#define THREAD_ARCH_EXT   uint32_t saved_int_level;
```
```saved_int_level``` would become member of the thread control block.

kaspar030, 2018-06-15:
Yeah, we've had this issue before (#1234), but dismissed it as there was no need anymore.
Feel free to pick this up again!

I'd suggest using a struct, e.g.,

```
struct _thread {
    char *sp;                       /**< thread's stack pointer         */
    uint8_t status;                 /**< thread's status                */
    uint8_t priority;               /**< thread's priority              */
    ...
#ifdef CPU_HAS_THREAD_ARCH_T
  thread_arch_t arch;     /** architecture dependent thread state */
#endif
    ...
}
```

In ```cpu/$arch/include/cpu_conf.h```:
```
typedef struct {
  uint32_t saved_int_level;
} thread_arch_t;
#define CPU_HAS_THREAD_ARCH_T
```

kYc0o, 2018-08-27:
Can we close this since #9364 adds this functionality?

kaspar030, 2018-08-27:
Yes!

closed 2018-08-27 by kaspar030