RIOT-OS/RIOT #9376: net/sock_udp: assign ephemeral source port on `sock_udp_create()`

haukepetersen, 2018-06-20:
#### Description
In the current sock/udp implementation for GNRC (and probably others as well), an ephemeral source port is assigned only when calling `sock_udp_send()` for the first time. This casts IMO unnecessary restrictions on the interface, and this seems to be counter intuitive to me. 

Take the following example, using a dedicated listener thread to handle incoming data: 
- thread_1 is started during system init as pure listener thread
- thread_1: calls `sock_udp_create()` with `local := NULL`
- thread_1: enters its main loop and calls `sock_udp_recv()`
- main_thread: the user app calls (most likely indirectly) `sock_udp_send()` at some later point in time

This example will not work -> the call to `sock_udp_recv()` will return `-EADDRNOTAVAIL` as no local port has been assigned.

**Solution: I suggest to move the assignment of the ephemeral source port into `sock_udp_create()`.**

This has two advantages:
1. the sock interface is more flexible (see example above)
2. we can assert the `sock.local` field to be configured in the `sock_udp_send` function and thus save a tiny amount of run time overhead

TODOs the follow from this proposal:
- [ ] fix sock_udp documentation: state more specifically when, where, and how ephemeral source ports are assigned
- [ ] adapt existing sock implementations
  - [ ] GNRC
  - [ ] lwip
  - [ ] emb6

#### Steps to reproduce the issue
n/a

#### Expected results
Ephemeral source port is assigned to socket when `sock_udp_create()` is called with `local := NULL`

#### Actual results
Ephemeral source port is assigned on the first call of `sock_udp_send()

#### Versions
n/a

haukepetersen, 2018-06-20:
Thinking and talking about it, I might have found a better solution: how about we change the semantics for `sock_udp_create()` to the following:
- `local == NULL` -> ephemeral source port is assigned when calling `sock_udp_send()` [no change]
- `local != NULL && local.port != 0` -> use given ep as local ep [no change]
- `local != NULL && local.port == 0` -> assign generate ephemeral port and use as source port [change]

haukepetersen, 2018-06-20:
also local.family needs to be set accordingly, of course

miri64, 2018-06-20:
:+1: This would keep the current API in tact (maybe doc needs some update though) and would be similar how such a case is handled with sockets.

haukepetersen, 2018-06-20:
perfect. Can I get you to implement the changes? :-)

miri64, 2018-06-20:
I can get do that.

miri64, 2018-06-20:
@haukepetersen see https://github.com/RIOT-OS/RIOT/pull/9380

miri64, 2018-06-20:
No, see https://github.com/RIOT-OS/RIOT/pull/9382. This time from the right branch and with tests + adaptations that came out of these tests

PR #9382 (2018-06-21): sock_udp: allow creation with ephemeral ports
### Contribution description
This change allows the port for local endpoint to be zero 0. If this is the case the `sock_udp_create()` function binds the object to an ephemeral port.

This also provides the port for GNRC, lwIP, and emb6 (including an update to the tests of GNRC and lwIP). Since GNRC doesn't support ephemeral ports natively (something like a UDP end-point isn't modeled in `netapi`) the port for this is a little bit more involved, but still a 5 liner. lwIP and emb6 handle this situation already internally, so only the assert for `loca [... cut]
closed 2018-06-21 by cgundogan