RIOT-OS/RIOT #9379: Priority inheritance - enabled by default?

danpetry, 2018-06-20:
### Description

https://github.com/RIOT-OS/RIOT/pull/7445 adds priority inheritance to RIOT. However, there is some open debate around whether the feature should be enabled or disabled by default. This issue aims to clarify the issue.

### Desired outcome

As well as making a decision on this particular issue, it would be helpful if the discussion can serve as a platform to clarify our approach, in a wider sense, to the tradeoffs involved in the decision. Also, it would be helpful if this was then captured in a project document, for example in https://github.com/RIOT-OS/RIOT/wiki/Best-Practice-for-RIOT-Programming, or elsewhere as appropriate. This would assist a shared understanding throughout the community with regards to design priorities.

Whatever the case, the outcome should be clearly documented where appropriate, perhaps in the API reference and/or starting some "real-time programming guidelines" or similar.

### Summary of previous discussion

Rough consensus on the issue pointed towards enabling by default. However, there was also strong opinion towards disabling by default.

#### Arguments for enabling by default

- Any other industrial RTOS I've noticed that brags about this, doesn't offer it as an option, instead it just claims it does it. @travisgriggs 
- The memory argument below could just as well refer to core_msg which is built-in per default. @OlegHahm 
- This makes the module easier to use. It's easier to disable when the user knows they don't need it rather than debug scheduling issues. It gives a stabler platform for users who may not know/care about the issue at all. @danpetry
- Disabling features that are not always needed (e.g., IPC) seems legit since a developer will easily figure out that he needs this feature. Disabling certain properties of the kernel may become much harder to detect. @OlegHahm 

#### Arguments for disabling by default

- Priority inheritance adds noticeable memory overheads when used. For example, the RAM requirement for a mutex rises from 4 to 8 bytes, and there was a lot of work historically in achieving such a low overhead. Enabling this per default in RIOT, where many applications don't even have need for real-time features, and what's more noone has shown a real world application where priority inheritance would have saved the day, but cannot be trivially fixed through other means, seems like simply a waste @haukepetersen @kaspar030 
- Having priority inheritance as an option gives the same marketing benefits as having it enabled by default @kaspar030 
- It can be bound to the "prioritized threads" module as it's an inherent issue with this @travisgriggs 

### Broader design questions

- Do we per default want many features in that can be disabled when optimizing for memory usage, or do we want a very slim default system and users enable the features they want to have?
- Ease of use and low memory requirements are two of our stated design goals. When they come into conflict, which takes priority?
- Is it possible to satisfy both at once, for example having "low resource" APIs/modules and "easy to use" APIs/modules alongside them which use more memory?

gdoffe, 2018-06-22:
Hi everyone,

Some good args from both sides. ;)

Except memories overhead, does it break actual RIOT use cases ?
If not, I'm in favor of its default activation, joining the point of view of @OlegHahm about disabling features.

I'm trying to put myself in the shoes of someone starting a new project with RIOT (which was my case not so long ago ;) ). When you start a project you always have few code, few threads, ... But you want to have it immediatly functionnal.
With few threads and few mutexes, you could have already some priority inversion problems which indeed could be tricky to debug for someone not really used to RIOT scheduling.

About overhead, yes it includes somes, but if this overhead is minimalist and thus optimized, regarding the importance of this new feature, it seems good to me.

About robotics, my team and I did not have time to go deeply into RIOT scheduling.
So we used it 'as is' with a sequential threads execution, knowing the risks and limitations. It works good but it can not stay like that.
As we are now turning RIOT into a ticks based OS (PR https://github.com/RIOT-OS/RIOT/pull/9332), we need to protect some critic sections of code and we have now to consider priority inversion problems.

Turning it on by default also give access by default to new use cases for RIOT which seems a good point too.

Whatever it is enabled or not by default, one important thing to me is that it should be "easy to configure". Some Kconfig-like menus would help to configure RIOT for beginners. But it is an other subject. ;)

Hope it helps

danpetry, 2018-07-03:
Hi @gdoffe 
Thank you for your input and sorry for the late reply.
Anybody else with any comments?
Dan

bergzand, 2018-08-04:
@jcarrano I'm not sure what you're trying to say with your arguments. I hope you do realize that priority inheritance protocol is not supposed to fix deadlocking issues, but is to improve the WCET of high priority threads.

> I think that an inheritance protocol amounts to moving to run-time a problem that should be solved at design time. Additionally, it does not provide guarantees: deadlocks are still possible though maybe less likely and that creates a false sense of safety. Some problems are better avoided than solved.

To me, your statement that priority inheritance protocol solves a design time problem at run-time sounds a bit naive. Do you have something to back this claim?

If you like I can give an extended explanation about PIP and some drawings of thread schedules to explain how PIP helps.

jia200x, 2020-04-08:
I think this should be revived

jia200x, 2020-04-08:
I agree with @bergzand. Without this it's not possible to guarantee an upper bound for responsiveness. If we claim to be a Real Time OS, I think this should be mandatory.

Also it shouldn't break any current use cases and it's always possible to disable it if there are concerns in memory consumption (same as `core_msg`).

kaspar030, 2020-04-08:
> I agree with @bergzand. Without this it's not possible to guarantee an upper bound for responsiveness. 

With it, it is not possible to guarantee an upper bound for responsiveness.

> If we claim to be a Real Time OS, I think this should be mandatory.

Why?

I'm assuming you want to revive this because the current isr event handler thread proposal has mutable access shared across differently prioritized threads, where the logic breaks if intermediate threads jump in between. I suggest fixing that first. Then you don't need the scheduler to fix priority issues.

> Also it shouldn't break any current use cases and it's always possible to disable it if there are concerns in memory consumption (same as `core_msg`).

This is not at all the same. `core_msg` is used 1000 times throughout the code. It has *no* runtime cost elsewhere apart from the extra thread_t fields.
Priority inheritance is not used anywhere (but in rare cases, you just found one), but the runtime cost is *everywhere*, *all the time*.

jia200x, 2020-04-08:
> With it, it is not possible to guarantee an upper bound for responsiveness.
... with a limited set of resources. Of course it's possible to design an architecture that has a defined upper bound, but you will require to add more threads than needed.

> Why?

I'm assuming you want to revive this because the current isr event handler thread proposal has mutable access shared across differently prioritized threads, where the logic breaks if intermediate threads jump in between. I suggest fixing that first. Then you don't need the scheduler to fix priority issues.

How would you solve this, without adding e.g extra threads or centralize driver access in the same thread?

> This is not at all the same. core_msg is used 1000 times throughout the code. It has no runtime cost elsewhere apart from the extra thread_t fields.
Priority inheritance is not used anywhere (but in rare cases, you just found one), but the runtime cost is everywhere, all the time.

In practice, how big is this runtime cost? How bad is to add some bytes of RAM considering we could
simplifyseveral RIOT components to have less thread (e.g `gnrc_netif`)

And if these points were a problem, can't a user simply deactivate the feature? What speaks against that?

kaspar030, 2020-04-08:
> And if these points were a problem, can't a user simply deactivate the feature? What speaks against that?

I'm fine with a user *activating a feature* if it is *needed*. Pay for what you use. Nothing against that.

kaspar030, 2020-04-08:
> In practice, how big is this runtime cost? How bad is to add some bytes of RAM considering we could
> reduce the number of threads in several RIOT components? (e.g `gnrc_netif`)

Yes, please, mesurements! RAM is probably not the issue, but the extra cycles.

closed 2020-11-11 by stale[bot]