RIOT-OS/RIOT #9381: newlib-nano: NEWLIB_INCLUDE_DIR set incorrectly

haukepetersen, 2018-06-20:
#### Description
With #9243 merged, I can not build anything for Cortex-M anymore, as apparently the `NEWLIB_INCLUDE_DIR` is not set properly for my environment:
```
cc1: error: /newlib-nano: No such file or directory [-Werror=missing-include-dirs]
cc1: all warnings being treated as errors
```

I might have a more or less custom setup (though not too fancy I would guess):
- toolchain at custom location
- custom folder containing symbolic links to binaries added to global PATH var

Now the definition of the `NEWLIB_INCLUDE_DIR` variable in `makefiles/libc/newlib.mk` does not cover this case, and thus leading to my setup not be able to build anything with the newly enabled warnings set.

Quick solution is for me to export the `NEWLIB_INCLUDE_DIR` variable manually for my environment, but this seems quite hacky to me. So any suggestion on how to fix this in a more nice way?

This is one of the extremely annoying things, forcing me to debug our make system for things that break from one day to the other...

aabadie, 2018-06-20:
I reported the same problem [yesterday](https://github.com/RIOT-OS/RIOT/pull/9243#issuecomment-398390544) and this is a known problem (and already reported elsewhere). But from what I understand, your local toolchain might not be setup correctly.

I could fix it by uninstalling newlib with my package manager and use the [ARM Embedded toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm) in a custom location (/opt) with my system PATH updated accordingly.

cgundogan, 2018-06-20:
did you try #9216 ? Otherwise, does it help to add a new `NEWLIB_INCLUDE_PATTERNS` for your newlib location?

cladmi, 2018-06-20:
The reason I did not personally revert the `warnings` pr, is that cortexm, it is expected to be compiled with `newlib-nano` and for me, I expect the build to produce an error if it cannot be found. Which is my way of interpreting it.

The question that is really asked is, is it ok to include newlib-nano and compile without the headers because the toolchain is not installed in a supported directory ?

Which I asked here: https://github.com/RIOT-OS/RIOT/pull/9216/files#r196359454

----

```
makefiles/arch/cortexm.inc.mk:USEMODULE += newlib_nano
```

With your current setup, even if requested to build with `newlib_nano` the headers are actually not used as the`NEWLIB_INCLUDE_DIR` is set to the wrong directory. So I see it as broken in master, but just silently hiding the error. Adding the warning did make this unsupported setup explicit.

Which is why I try to help fixing the newlib_nano detection more than hiding the problem.

----

If it is "acceptable" to use the wrong header, a solution could be to just not set this include directory if it does not exist.

miri64, 2018-06-21:
I have a similar problem on  Ubuntu 17.10 now:

```
BOARD=iotlab-m3 IOTLAB_SITE=paris IOTLAB_PHY_NODES=1-11 IOTLAB_DURATION=20 make -C examples/gnrc_networking
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking'
Building application "gnrc_networking" for "iotlab-m3" with MCU "stm32f1".

cc1: error: /etc/alternatives/gcc-arm-none-eabi-include/newlib-nano: No such file or directory [-Werror=missing-include-dirs]
cc1: error: unrecognized command line option '-Wno-implicit-fallthrough' [-Werror]
cc1: all warnings being treated as errors
/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:83: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/iotlab-m3/application_gnrc_networking/main.o' failed
make[1]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/iotlab-m3/application_gnrc_networking/main.o] Error 1
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/../../Makefile.include:359: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/iotlab-m3/application_gnrc_networking.a' failed
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/iotlab-m3/application_gnrc_networking.a] Error 2
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking'
```

```
[mlenders@mk RIOT]<3 lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful
[mlenders@mk RIOT]<3 uname -a
Linux mk 4.13.0-45-generic #50-Ubuntu SMP Wed May 30 08:23:18 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
[mlenders@mk RIOT]<3 ./dist/tools/ci/print_toolchain_versions.sh 
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
                  clang: clang version 4.0.1-6 (tags/RELEASE_401/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.4.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.9.1
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: 3.2.1 (pyflakes: 1.5.0, mccabe: 0.6.1, pycodestyle: 2.3.1) CPython 3.6.3 on Linux
                    git: git version 2.14.1
             coccinelle: missing
```

cgundogan, 2018-06-21:
@haukepetersen would #9394 help you out with your custom setup?

cgundogan, 2018-06-21:
if not, then I would suggest that you symlink your newlib folder to `/usr/include/newlib`

smlng, 2018-06-22:
on macOS I currently get: 
```
BOARD=samr21-xpro make -C examples/gnrc_networking
Warning: no PORT set!
Building application "gnrc_networking" for "samr21-xpro" with MCU "samd21".

cc1: error: -I/RIOT/core/include: No such file or directory [-Werror=missing-include-dirs]
In file included from /RIOT/examples/gnrc_networking/main.c:23:0:
/RIOT/sys/include/shell.h:25:10: fatal error: kernel_defines.h: No such file or directory
 #include "kernel_defines.h"
          ^~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
compilation terminated.
```

when building anything for cortex based boards.

closed 2018-06-26 by smlng

cladmi, 2018-06-28:
@smlng having `-I/RIOT/core/include ` in your include directories is really really wrong. Could you please open a dedicated issue for this with output of `make info-build` and any debug information.

PR #9394 (2018-06-26): newlib: include fixes for newlib-nano
### Contribution description
At least for Ubuntu newlib-nano seems to be in a different location
(`/usr/include/newlib/nano/newlib.h`).

This PR fixes #9381 by adding `/usr/include` to the newlib includes and searching for both `newlib-nano` (Arch et al.) and `newlib/nano` (Ubuntu et al.) explicitly.

Tested on Arch and Ubuntu.

---

**EDIT**: The major change in this PR is now to autodetect newlib in the compiler's default search paths, using:
```bash
$(PREFIX)gcc -v -x c -E /dev/null 2>&1 | grep '^\s' | tr -d '\n'
```

### Issues/PRs references
#9381