RIOT-OS/RIOT #9387: shell `ping6` doesn't display duplicate packages

dingens, 2018-06-21:
#### Description
This is rather a question. Maybe there is a good reason for this behavior.

When pinging and duplicates arrive, the riot shell `ping6` command just says
```
12 bytes from fe80::xxx: id=84 seq=1 hop limit=64 time = 0.280 ms
dropping additional response packet (probably caused by duplicates)
```

IPutils ping does it like this:
```
64 bytes from fe80::yyy%wlp3s0: icmp_seq=1 ttl=64 time=0.051 ms
64 bytes from fe80::zzz%wlp3s0: icmp_seq=1 ttl=64 time=166 ms (DUP!)
```

I wonder why you chose to do it like this. My primary reason to ask is that I use `ping ff02::1` etc for debugging the network, for which the latter output is way more useable. Also, in general, when you have duplicates, I suppose you'd want to know where they are from. (Plus, it is quite arbitrary which of the packets are "hidden" by not displaying their information.)

#### Steps to reproduce the issue
Set up a node in a way that there's at least two other hosts listening on the same network.
Run `ping6 ff02::1`

#### Versions
linux
             native gcc: gcc (GCC) 8.1.1 20180531
      arm-none-eabi-gcc: arm-none-eabi-gcc (Arch Repository) 8.1.0

miri64, 2018-06-21:
I think there is no specific reason, other than a lazy approach to handling of packets... @OlegHahm according to `git blame` that error message is yours. Do you remember?

miri64, 2018-09-12:
> […] it would require a rather big rewrite of the ping command handler […] 

You mean something like #9523? 🙃

PR #9523 (2019-02-05): shell_commands: adapt ping6 to original-ping-like implementation
### Contribution description
Loosely based on [the original ping] and [netutil]'s ping

New features (compared to old RIOT version):
 - Non-positional parameters
 - Configurable hop-limit
 - Better duplicate detection (addresses #9387)
 - Better asynchronous behavior
 - Potential for future move to `sock_ip`
 - (Optional) DNS-support
 - Multithreading-safe (in case shell-command handler gets called
   from multiple threads)

[the original ping]: http://ftp.arl.army.mil/~mike/ping.html
[netutil]: https://www.gnu.org/software/inetu [... cut]
closed 2019-02-05 by cgundogan